openwrt/packages #18135: python3: host build compilation error

feckert, 2022-03-23:
Maintainer: @jefferyto 
Environment: x86_64, APU3, latest reboot-19180-g73c6d8fd04

Description:
I get the follwing compilation error during the latest update. I get this erro only for x86_64 build.
The other build lantiq_xrx200(MIPS) and x86_geode(x86_32) works as expected.
I suspect that the error message here is to link against the wrong or non-existent library libintl?
I have not changed anything in my build env.

```
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:736: undefined reference to `g_libintl_bindtextdomain'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_dcgettext_impl':
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:681: undefined reference to `g_libintl_dcgettext'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_dgettext_impl':
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:662: undefined reference to `g_libintl_dgettext'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_bind_textdomain_codeset_impl':
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:764: undefined reference to `g_libintl_bind_textdomain_codeset'
/usr/bin/ld: /home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:764: undefined reference to `g_libintl_bind_textdomain_codeset'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_textdomain_impl':
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:697: undefined reference to `g_libintl_textdomain'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_gettext_impl':
/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/hostpkg/Python-3.10.3/./Modules/_localemodule.c:643: undefined reference to `g_libintl_gettext'

```

jefferyto, 2022-03-24:
I'm a bit puzzled why this host Python build failed while others (I assume on the same machine) succeeded. Target arch should have no impact on host Python.

I assume you have cleaned your buildroot / checked out a new buildroot and got the same error?

feckert, 2022-03-24:
@jefferyto Thanks  for your reply
I have my development machine debian 10 and a buildsystem also debain 10 where the firmware is built.
It runs through on my development machine.
But on the buildsystem it fails with this error.

With both, the whole OpenWrt is always rebuild from scratch so I do not use old envs.

I suspect that did libintl lib is missing here, isnt it?
Perhaps it is not yet built at this point?
```
From: Florian Eckert <fe@dev.tdt.de>
Date: Thu, 24 Mar 2022 12:08:07 +0100
Subject: [PATCH] python3: fix host compilation

Redmine-patch-id: 6871
During compilation for x86_64 the following linker error arises:

undefined reference to `g_libintl_bindtextdomain'
undefined reference to `g_libintl_dcgettext'
undefined reference to `g_libintl_dgettext'
undefined reference to `g_libintl_bind_textdomain_codeset'
undefined reference to `g_libintl_textdomain'
undefined reference to `g_libintl_gettext'i

By adding the -lintl to HOST_LDFLAGS fixes this issue.

---
 lang/python/python3/Makefile | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/lang/python/python3/Makefile b/lang/python/python3/Makefile
index ece5faaa4bb199a0a6cead3c2edde172fc3e1286..93f931a4f2e04615a12e4bf3d32cdd70c6037e47 100644
--- a/lang/python/python3/Makefile
+++ b/lang/python/python3/Makefile
@@ -313,7 +313,7 @@ HOST_LDFLAGS += \

 ifeq ($(HOST_OS),Linux)
 HOST_LDFLAGS += \
-       -Wl,--no-as-needed -lrt
+       -Wl,--no-as-needed -lrt -lintl
 endif

 # Would be nice to be able to do this, but hosts are very fiddly
--
``

jefferyto, 2022-03-25:
I'd like to figure out the source of the error before jumping into a fix - since you were able to build host Python for other targets, I have built x86_64 many times, and buildbots have no issues, I still suspect there is a build host issue.

Can you post the complete configure/build log for host Python?

feckert, 2022-03-25:
@jefferyto
The logging output is quit long, so I have pasted it to [pastebin](https://pastebin.com/GBsDbq5m).

I am doing a rebuild on my development system which it is not failing.
I also send this to pastbin and send you the link when this has build.

feckert, 2022-03-25:
That is the log from a success python host-compile build.
Unfortunately I am not able to upload this to pastebin because of the size.
So I past this file hiere. [host-compile.txt](https://github.com/openwrt/packages/files/8349290/host-compile.txt)

feckert, 2022-03-29:
@jefferyto
Now I have also included openwrt-22.03 in my build environment, which runs on the same buildhost (debian10) in my buildsystem env, and I now have the same error on this branch too.
I have also updated my development system to debian 11 and now I see the same error before this I do not see this.

I also noticed that if I change the openwrt Makefile of python3 and want to try out a fix, for example, give -lintl to HOST_LDFLGAS for linux see (https://github.com/openwrt/packages/issues/18135#issuecomment-1077646091), then it builds.
After removing this change and to a package clean, compile, I can no longer produce the error.

To get the error again. I then have to purge the build and start again.
The whole thing is very strange.

feckert, 2022-03-30:
@jefferyto I have now investigated this further.

I not only have a problem with python3 but also with the core package [elfutil](https://github.com/openwrt/openwrt/blob/master/package/libs/elfutils/Makefile). Something is not right here with libintl.

When I look at both python3 and elfutil packages to see what functions are missing, then they come from the glib2 lib.

**Package elfutil:**
Host-Build fail:
```gcc -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Werror -Wunused -Wextra -Wstack-usage=262144    -D_FORTIFY_SOURCE=2 -O2 -I/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/host/include -I/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/hostpkg/include -I/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/target-x86_64_musl/host/include  -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/host/lib -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/hostpkg/lib -L/home/feckert/workspace/openwrt/LDM-master-lantiq_xrx200/build/openwrt/staging_dir/target-x86_64_musl/host/lib -o libelf.so -shared -Wl,-z,defs -Wl,-z,relro -o libelf.so \
        -Wl,--soname,libelf.so.1 \
        -Wl,--version-script,libelf.map,--no-undefined \
        -Wl,--whole-archive libelf_pic.a -Wl,--no-whole-archive \
        ../lib/libeu.a -lz
/usr/bin/ld: libelf_pic.a(elf_error.os): in function `elf_errmsg':
elf_error.c:(.text+0xd6): undefined reference to `g_libintl_dgettext'
/usr/bin/ld: elf_error.c:(.text+0xf0): undefined reference to `g_libintl_dgettext'
collect2: error: ld returned 1 exit status
```
Search for function `g_libintl_dgettext` in build tree:
```
feckert@feckert01 ~/openwrt/staging_dir/hostpkg/lib:
grep -R 'g_libintl_dgettext'
grep: libintl.a: binary file matches
grep: libglib-2.0.a: binary file matches
```
**Package python3:**
Host-Build fail:
```
rm -f libpython3.10.a
ar rcs libpython3.10.a Modules/getbuildinfo.o Parser/token.o  Parser/pegen.o Parser/parser.o Parser/string_parser.o Parser/peg_api.o Parser/myreadline.o Parser/tokenizer.o Objects/abstract.o Objects/accu.o Objects/boolobject.o Objects/bytes_methods.o Objects/bytearrayobject.o Objects/bytesobject.o Objects/call.o Objects/capsule.o Objects/cellobject.o Objects/classobject.o Objects/codeobject.o Objects/complexobject.o Objects/descrobject.o Objects/enumobject.o Objects/exceptions.o Objects/genericaliasobject.o Objects/genobject.o Objects/fileobject.o Objects/floatobject.o Objects/frameobject.o Objects/funcobject.o Objects/interpreteridobject.o Objects/iterobject.o Objects/listobject.o Objects/longobject.o Objects/dictobject.o Objects/odictobject.o Objects/memoryobject.o Objects/methodobject.o Objects/moduleobject.o Objects/namespaceobject.o Objects/object.o Objects/obmalloc.o Objects/picklebufobject.o Objects/rangeobject.o Objects/setobject.o Objects/sliceobject.o Objects/structseq.o Objects/tupleobject.o Objects/typeobject.o Objects/unicodeobject.o Objects/unicodectype.o Objects/unionobject.o Objects/weakrefobject.o Python/_warnings.o Python/Python-ast.o Python/asdl.o Python/ast.o Python/ast_opt.o Python/ast_unparse.o Python/bltinmodule.o Python/ceval.o Python/codecs.o Python/compile.o Python/context.o Python/dynamic_annotations.o Python/errors.o Python/frozenmain.o Python/future.o Python/getargs.o Python/getcompiler.o Python/getcopyright.o Python/getplatform.o Python/getversion.o Python/hamt.o Python/hashtable.o Python/import.o Python/importdl.o Python/initconfig.o Python/marshal.o Python/modsupport.o Python/mysnprintf.o Python/mystrtoul.o Python/pathconfig.o Python/preconfig.o Python/pyarena.o Python/pyctype.o Python/pyfpe.o Python/pyhash.o Python/pylifecycle.o Python/pymath.o Python/pystate.o Python/pythonrun.o Python/pytime.o Python/bootstrap_hash.o Python/structmember.o Python/symtable.o Python/sysmodule.o Python/thread.o Python/traceback.o Python/getopt.o Python/pystrcmp.o Python/pystrtod.o Python/pystrhex.o Python/dtoa.o Python/formatter_unicode.o Python/fileutils.o Python/suggestions.o Python/dynload_shlib.o    Modules/config.o Modules/getpath.o Modules/main.o Modules/gcmodule.o Modules/posixmodule.o  Modules/errnomodule.o  Modules/pwdmodule.o  Modules/_sre.o  Modules/_codecsmodule.o  Modules/_weakref.o  Modules/_functoolsmodule.o  Modules/_operator.o  Modules/_collectionsmodule.o  Modules/_abc.o  Modules/itertoolsmodule.o  Modules/atexitmodule.o  Modules/signalmodule.o  Modules/_stat.o  Modules/timemodule.o  Modules/_threadmodule.o  Modules/_localemodule.o  Modules/_iomodule.o Modules/iobase.o Modules/fileio.o Modules/bytesio.o Modules/bufferedio.o Modules/textio.o Modules/stringio.o  Modules/faulthandler.o  Modules/_tracemalloc.o  Modules/symtablemodule.o  Modules/zlibmodule.o  Modules/xxsubtype.o Python/frozen.o
gcc -pthread -L/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/host/lib -L/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib -L/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/host/lib -L/home/bbworker/bbworker/owrt_mas [... cut ...]
feckert@feckert01 ~/openwrt/staging_dir/hostpkg/lib:
grep -R g_libintl_bindtextdomain
grep: libintl.a: binary file matches
grep: libglib-2.0.a: binary file matches
```

This function are provided by libintl.a `/openwrt/staging_dir/hostpkg/lib` and this lib will be generated by glib2 and copied this this location.
```
feckert@feckert01  ~/openwrt: 
find ./* -name libintl.a
./openwrt/build_dir/hostpkg/glib-2.70.4/openwrt-build/subprojects/proxy-libintl/libintl.a
./openwrt/staging_dir/hostpkg/lib/libintl.a
```

jefferyto, 2022-03-31:
So something is causing glib2 to build its own version of libintl?

jefferyto, 2022-03-31:
I can reproduce the error with elfutils by (manually/separately) building host glib2 then trying to build host elfutils. With host Python though, it detected libintl during configure and added `-lintl` to the linker flags; perhaps the previous error was a race condition, where glib2 and Python were being built at the same time.

feckert, 2022-03-31:
That definitely a race condition.
Because with buildsystem I always get the error.
With my development machine only sometimes.

1. The question now is why does glib2 build this lib at all?
2. A dependnecy on glib2 from elfutils is not possible, because glib2 is in the package feed. Therefore, we must somehow switch off the elfutils so that it looks for -lintl. This only happens during the host build.

neheb, 2022-03-31:
So uhh I found a solution. In the python Makefile somewhere there's a Darwin check that makes it pass a configure car. Move it down and problem solved. Might work for elfutils too.

This stuff is failing as a result of me trying to fix compilation under Alpine Linux and having host libraries be mostly static libraries.

jefferyto, 2022-03-31:
My theory is because `gettext-full/host` was removed from `HOST_BUILD_DEPENDS` in d2062486e87a1115d915492b460f16480ebc28a4, glib2 (not finding libintl) [builds](https://gitlab.gnome.org/GNOME/glib/-/blob/2.70.4/meson.build#L2075) [proxy-libintl](https://github.com/frida/proxy-libintl) (a [subproject](https://gitlab.gnome.org/GNOME/glib/-/blob/2.70.4/subprojects/proxy-libintl.wrap)), and this replacement libintl is causing the linker errors for other packages.

feckert, 2022-03-31:
@jefferyto Thanks for your advice, I have now added again the host build dependency for gettext-full/host to glib2/host build, that was removed by the posted commit https://github.com/openwrt/packages/commit/d2062486e87a1115d915492b460f16480ebc28a4

My expectation would have been that libintl.h and libintl.a would be provided by gettext-full/host and so installed in `./staging_dir/hostpkg/include/` and `./staging_dir/lib/libintl.a`
But this is not the case! For some reason glib2 still use its own subproject.

But somehow these files are not copied and the link from msgmt points to nothing. This was added by @stintel with this [commit](https://github.com/openwrt/openwrt/commit
/636cb00ecc8d693c36e48952f6d154f91e0e569e).
Very strange. I removed the commit and the host files from gettext-full are copied to the staging_dir.

I'm a bit confused now. I'm starting a new build with the following changes.
- gettext-full: Remove last commit from gettext-full https://github.com/openwrt/openwrt/commit
/636cb00ecc8d693c36e48952f6d154f91e0e569e
- glib2: Add again build dependency for glib2/host to gettext-full/host
- elfutil: Add to [HOST_CONFIGURE_VARS](https://github.com/openwrt/openwrt/blob/master/package/libs/elfutils/Makefile#L82) ` ac_cv_header_libintl_h=no` as suggest by @neheb

jefferyto, 2022-03-31:
I was wrong about gettext-full/host - removing [patches-host/001-musl-libintl.patch](https://github.com/openwrt/packages/blob/master/libs/glib2/patches-host/001-musl-libintl.patch) causes glib2 to not build the subproject. Perhaps there is another way to fix building on Alpine.

neheb, 2022-03-31:
https://github.com/neheb/openwrt/commit/ea7e653f0535640825a6c3e5067f55821cb76252

sort of fixed elfutils. It actually fails with

```
        ../lib/libeu.a ../libelf/libelf.so -ldl -lz -largp -lfts  -lbz2  -pthread -lintl
/usr/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib/libargp.a(argp-parse.o): in function `argp_usage':
argp-parse.c:(.text+0x1ae0): multiple definition of `argp_usage'; ../libdwfl/libdwfl_pic.a(argp-std.os):argp-std.c:(.text+0x0): first defined here
/usr/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib/libargp.a(argp-parse.o): in function `_option_is_short':
argp-parse.c:(.text+0x1b00): multiple definition of `_option_is_short'; ../libdwfl/libdwfl_pic.a(argp-std.os):argp-std.c:(.text+0x20): first defined here
/usr/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib/libargp.a(argp-parse.o): in function `_option_is_end':
argp-parse.c:(.text+0x1b30): multiple definition of `_option_is_end'; ../libdwfl/libdwfl_pic.a(argp-std.os):argp-std.c:(.text+0x50): first defined here
```

Conflict between libc's argp and argp_standalone. No real idea how to fix.

jefferyto, 2022-04-01:
The issue isn't with elfutils, changing elfutils isn't going to fix the Python build error. The issue is with glib2 and the musl libintl patch.

neheb, 2022-04-01:
python build error fix:

```
--- a/lang/python/python3/Makefile
+++ b/lang/python/python3/Makefile
@@ -320,9 +320,9 @@ endif
 # HOST_MAKE_VARS += \
 #      PYTHONSTRICTEXTENSIONBUILD=1

-ifeq ($(HOST_OS),Darwin)
 HOST_CONFIGURE_VARS += \
        ac_cv_header_libintl_h=no
+ifeq ($(HOST_OS),Darwin)
 HOST_MAKE_VARS += \
        USE_PYTHON_CONFIG_PY=1
 endif

```

jefferyto, 2022-04-01:
> python build error fix:

So every time there is a conflict between glib2 and another package, your plan is to patch the other package?

neheb, 2022-04-01:
let's see. meson-log.txt says

```
Running compile:
Working directory:  /home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.70.4/openwrt-build/meson-private/tmpm2jjmpeq
Command line:  ccache clang -L/home/mangix/devstuff/openwrt/staging_dir/host/lib -L/home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib -I/home/mangix/devstuff/openwrt/staging_dir/host/include -I/home/mangix/devstuff/openwrt/staging_dir/hostpkg/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips64_mips64r2_64_musl/host/include /home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.70.4/openwrt-build/meson-private/tmpm2jjmpeq/testfile.c -o /home/mangix/devstuff/openwrt/build_dir/hostpkg/glib-2.70.4/openwrt-build/meson-private/tmpm2jjmpeq/output.exe -O2 -D_FILE_OFFSET_BITS=64 -O0 -Werror=implicit-function-declaration -Wl,--start-group -lintl -Wl,--end-group -Wl,--allow-shlib-undefined

Code:
 int main(void) { return 0; }

Compiler stdout:

Compiler stderr:
 /usr/bin/ld: cannot find -lintl
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
```

libintl.a is definitely present in hostpkg/lib. Wonder why it's failing.

jefferyto, 2022-04-01:
@neheb I built glib2/host (armvirt-32) without 001-musl-libintl.patch on Alpine Linux 3.15.3 (x86-64, with gcc and musl-libintl) and it succeeded. What error message do you see when you try to compile glib2/host without that patch?

neheb, 2022-04-01:
I don't on Fedora. IIRC it was a linker error on Alpine.

petrov-adg, 2022-04-01:
Same problem
OS: Ubuntu 22.04 via Docker
Version: openwrt-22.03 (commit b664bb617da44ba40dc5f18b72375ee2bb583d6e)
Target: x86_64
```
make[4]: Entering directory '/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2'
ccache gcc -L/usr/src/openwrt/staging_dir/host/lib -L/usr/src/openwrt/staging_dir/hostpkg/lib -L/usr/src/openwrt/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/usr/src/openwrt/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/usr/src/openwrt/staging_dir/host/lib -L/usr/src/openwrt/staging_dir/hostpkg/lib -L/usr/src/openwrt/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/usr/src/openwrt/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -fno-semantic-interposition  -Xlinker -export-dynamic -o python Programs/python.o libpython3.10.a -lcrypt -ldl  -lm -lz   -lm 
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_bind_textdomain_codeset_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:764: undefined reference to `g_libintl_bind_textdomain_codeset'
/usr/bin/ld: /usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:764: undefined reference to `g_libintl_bind_textdomain_codeset'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_textdomain_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:697: undefined reference to `g_libintl_textdomain'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_dcgettext_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:681: undefined reference to `g_libintl_dcgettext'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_dgettext_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:662: undefined reference to `g_libintl_dgettext'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_gettext_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:643: undefined reference to `g_libintl_gettext'
/usr/bin/ld: libpython3.10.a(_localemodule.o): in function `_locale_bindtextdomain_impl':
/usr/src/openwrt/build_dir/hostpkg/Python-3.10.2/./Modules/_localemodule.c:736: undefined reference to `g_libintl_bindtextdomain'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:601: python] Error 1
```

feckert, 2022-04-01:
@jefferyto I can confirm that if I delete the said patch for glib2 then the build works for me again :+1:

jefferyto, 2022-04-07:
I think https://github.com/openwrt/packages/commit/2784618ab57a9533a615eb139e87b920f733c002 fixes this issue.

feckert, 2022-04-08:
Thanks for your helps :+1:

closed 2022-04-08 by feckert