openwrt/packages #18148: usbutils broken....

CanadianJeff, 2022-03-25:
continued from issue #14488 

this seems to be broken in later versions of openwrt??? that it does not even read the `usb.ids` file

```
root@APU2:/usr/share/hwdata# opkg files usbutils
Package usbutils (013-2) is installed on root and has the following files:
/usr/bin/lsusb
/usr/bin/usbreset
root@APU2:/usr/share/hwdata# opkg files usbids
Package usbids (0.354-1) is installed on root and has the following files:
/usr/share/hwdata/usb.ids
root@APU2:/usr/share/hwdata# lsusb
Bus 002 Device 002: ID 2357:0120 Realtek  802.11ac WLAN Adapter
Bus 002 Device 001: ID 1d6b:0002 Linux 5.4.179 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 0438:7900
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.179 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0003 Linux 5.4.179 xhci-hcd xHCI Host Controller
root@APU2:/usr/share/hwdata# strace lsusb 2>&1 | grep "usb.ids"
root@APU2:/usr/share/hwdata#
```
```
root@APU2:/usr/share/hwdata# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.2, r16495-bf0c965af0
 -----------------------------------------------------
```

kuanyili, 2022-03-27:
Well... this was actually discussed here: https://github.com/openwrt/packages/pull/16458#issuecomment-905579787

CanadianJeff, 2022-03-27:
the older and BETTER version

```
root@APU2:/tmp# opkg remove usbutils
Removing package usbutils from root...
root@APU2:/tmp# opkg files usbutils
Collected errors:
 * opkg_files_cmd: Package usbutils not installed.
root@APU2:/tmp# wget https://downloads.openwrt.org/releases/19.07.7/packages/x86_64/base/usbutils_007-10_x86_64.ipk
Downloading 'https://downloads.openwrt.org/releases/19.07.7/packages/x86_64/base/usbutils_007-10_x86_64.ipk'
Connecting to 2a01:4f8:251:321::2:443
Writing to 'usbutils_007-10_x86_64.ipk'
usbutils_007-10_x86_ 100% |*******************************|   240k  0:00:00 ETA
Download completed (246616 bytes)
root@APU2:/tmp# opkg install usbutils_007-10_x86_64.ipk
Installing usbutils (007-10) to root...
Configuring usbutils.
root@APU2:/tmp# opkg files usbutils
Package usbutils (007-10) is installed on root and has the following files:
/usr/share/usb.ids
/usr/bin/lsusb
root@APU2:/tmp# rm -fv /usr/share/usb.ids
root@APU2:/tmp# ln -s /usr/share/hwdata/usb.ids /usr/share/usb.ids
root@APU2:/tmp# cd /usr/share/hwdata/
root@APU2:/usr/share/hwdata# lsusb
Bus 002 Device 005: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

neheb, 2022-04-14:
https://github.com/illiliti/libudev-zero mentions this under Cons. Maybe poke the developer to implement hwdb support.

acoul, 2026-05-03:
it will be nice (useful) for current lsusb to have the option to also read device info from /usr/share/hwdata/usb.ids (if available)

I've opened a [relative request upstream](https://github.com/gregkh/usbutils/issues/177#issuecomment-4365658605)