openwrt/packages #18155: mwan3: setname 'mwan3_sticky_ipv4_default_rule_v4' is longer than 31 characters

RasimKalimullin, 2022-03-26:
Maintainer: @feckert
Environment: mediatek, Belkin rt3200, snapshot)

Description:
Fw4, empty mwan3_rules chain
```
Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
```
because
```
Sat Mar 26 11:35:04 2022 user.err mwan3-init[15552]: set_sticky_ipset (default_rule_v4): ipset v7.15: Error in line 2: Syntax error: setname 'mwan3_sticky_ipv4_default_rule_v4' is longer than 31 characters
Sat Mar 26 11:35:04 2022 user.err mwan3-init[15552]: set_user_rules (ipv4): iptables-restore v1.8.7 (nf_tables): setname `mwan3_sticky_ipv4_default_rule_v4' too long, max 31 characters. Error occurred at line: 36 Try `iptables-restore -h' or 'iptables-restore --help' for more information.
```
mwan3 version: 2.11.0

jamesmacwhite, 2022-03-27:
By default, the default_rule_v4 doesn't have sticky enabled, so I assume this had been enabled in your own configuration?

You are correct though the chain names do exceed 31 characters. I believe this is due to the different chain names created in 2.11, before they were prefixed as `mwan3_sticky_v4` `mwand_sticky_v6` but now it's `mwan3_sticky_ipv4` and `mwan3_sticky_ipv6`.

Combined with the default rule, `mwan3_sticky_ipv4_` (18 characters) and the rule name `default_rule_v4` (15), exceeds the limit by 2 characters.

I believe rule names remain fixed at 15 characters, so it means the mwan3 rule chain prefix can not exceed 16 characters to stay within the limit, relative to a rule name.

feckert, 2022-03-28:
@jamesmacwhite Thanks for this spotting and sorry, I didn't think of that at all and it slipped through my fingers!

There are 2 characters too many now because of **ip** :-(
I would suggest that we have to check if this restriction still exist in iptables?
That would be the simpleset change to change the max char from 31 to 33 

I don't want to revert the change, because the source code it is now generic.
Everywhere ipv4 or ipv6 is used now.

jamesmacwhite, 2022-03-28:
@feckert I think the limit set by iptables is configurable via compile option XT_FUNCTION_MAXNAMELEN, but this might be tricky as even if you could update it in OpenWrt. Any version before 22.03 would be under a different limit, given it's a compile option unless it was backported down to other branches.

How about just shortening the name prefix so rather than `mwan3_sticky`, it could be `mw3_sticky`? That saves two characters without having to change the ipv4 and ipv6 naming in 2.11 and we don't have to worry about the upstream iptables max length. To be consistent, this would need to be changed everywhere though.

Any mwan3 chain name prefix can't ever exceed 16 characters, to ensure the maximum size user created rule can use 15 characters, unless the limit is increased, but I think that then touches on the OpenWrt core and the compile options with the kernel.

feckert, 2022-03-28:
fixes in [master](https://github.com/openwrt/packages/commit/87c611abf8facb3b401006cece0c722676c63860) and [openwrt-22.03](https://github.com/openwrt/packages/commit/1534c958eb1e43df38b590698839b031a135b9a5)

closed 2022-03-28 by feckert