openwrt/packages #18165: ffmpeg patented

neheb, 2022-03-26:
Fedora 36 will include ffmpeg with patented stuff disabled. We should probably mirror their decisions here. I don't know if it's currently the same. https://src.fedoraproject.org/rpms/ffmpeg/blob/rawhide/f/ffmpeg.spec

ping @diizzyy @BKPepe @antonlacon

diizzyy, 2022-03-27:
Depending on your view (I haven't check what they deem as "patented") you might as well remove the complete package since pretty much anything useful is patented in some way.

neheb, 2022-03-29:
ping @yousong

antonlacon, 2022-03-29:
@thess took steps to have features believed by contributors to be patented removed from OpenWRT provided packages. See https://github.com/openwrt/packages/blob/f7cbd445408756b7a44764080063a9a4158988ed/multimedia/ffmpeg/Makefile#L469-L473 and the associated variables for libffmpeg-full. libffmpeg-mini has similar. If you would like to go through the fedora .spec file to add onto this, that may be reasonable as maybe Fedora has legal counsel to review it. My guess is that it's line 633 in their spec file from the listed codecs, but I'm not familiar with Fedora's package files to be certain that's all of it.

An addition was made to the all the ffmpeg package descriptions reading:

>  FFmpeg licensing / patent issues are complex. It is the reponsibility of the
>  user to understand any requirements in this regard with its usage. See:
>  https://ffmpeg.org/legal.html for further information.

FFmpeg themselves do not provide any guidance on what is or is not patented (see the legal FAQ). I don't know what's prompting the question to weigh in on whether the appropriate response is to remove the package instead.

neheb, 2022-04-13:
Hmm this seems more complicated. Fedora creates its own ffmpeg tarball.