openwrt/packages #18166: gensio: error: recursive dependency

Alexey71, 2022-03-27:
OpenWrt Snapshot

```
feeds/packages/net/gensio/Config.in:11:error: recursive dependency detected!
feeds/packages/net/gensio/Config.in:11: symbol GENSIO_SCTP depends on PACKAGE_libgensio
tmp/.config-package.in:53955:   symbol PACKAGE_libgensio depends on GENSIO_SCTP
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

Alexey71, 2022-03-27:
@WereCatf

WereCatf, 2022-03-27:
I do not know how to fix that. It seems to only happen with the GENSIO_SCTP, none of the other configuration-options, even though there is literally zero difference between them. Renaming it to something else doesn't work and, as far as I know, it has to depend on PACKAGE_libgensio or else the configuration-options won't show up in the correct place.

Given how none of the other options trigger this error, it seems to be a bug with the build-system instead of gensio.

mhei, 2022-03-27:
I tend to agree that this is an issue with the build system when it tries to resolve or to pull in the conditional deps here:

```
    ...
    menuconfig PACKAGE_libgensio
        tristate "libgensio............................ Abstract stream I/O (runtime files)"
        default y if DEFAULT_libgensio
        default m if ALL
        select PACKAGE_librt if USE_GLIBC
        select PACKAGE_libsctp if GENSIO_SCTP
        depends on !(GENSIO_SCTP && LINUX_5_15) || IPV6
        select PACKAGE_glib2 if GENSIO_GLIB
        select PACKAGE_libpthread if GENSIO_PTHREADS
       ...
```

My feeling is that the line `depends on !(GENSIO_SCTP && LINUX_5_15) || IPV6` should not be there since it's the business of libscpt to deal with its internal deps?

mhei, 2022-03-28:
I still don't get there difference to other dependencies which apparently work out-of-the-box. And there is a comment in [package-metadata.pl](https://github.com/openwrt/openwrt/blob/1c7e146f7e0f42e5e414c6cff0c73f75b405e0dd/scripts/package-metadata.pl#L200) which I don't understand. Maybe @nbd168, @jow- or @NeoRaider could have a look here? Help would be really appreciated.

WereCatf, 2022-03-28:
@mhei The difference seems to be libsctp depending directly on a kernel-module, whereas the others don't.

Adding `@IPV6` to libgensio's dependencies, ie. `DEPENDS:=@IPV6 +GENSIO_SSL:libopenssl +GENSIO_WRAP:libwrap +GENSIO_SCTP:libsctp +GENSIO_PTHREADS:libpthread +GENSIO_AVAHI:libavahi-client +GENSIO_TCL:tcl +GENSIO_GLIB:glib2 +GENSIO_SSHD:libpam`, fixes the circular dependency error, but I don't know enough about the build-system to tell, whether that is the correct way of fixing this or not.

mhei, 2022-03-29:
I understand that going the dependencies down to `kmod-udptunnel6` which depends on `@IPV6` is the "culprit".

Adding `@IPV6` as global dep does not seem to be the correct solution, I'd suggest `... +(IPV6&&GENSIO_SCTP):libsctp ... `.

This also fixes the circular dep error, the generated syntax in `tmp/.config-package.in` is now
`depends on !((IPV6&&GENSIO_SCTP) && LINUX_5_15) || IPV6`. I'm not sure why this makes the circular dep disappear, but can you confirm it is gone?

In my eyes, this is still not the correct fix - however, I could live with it... let's wait for other comments.

WereCatf, 2022-03-29:
Well, yes, I can confirm that works for making the circular dependency error disappear, though shouldn't I also change the CONFIGURE_ARGs accordingly? Ie. `--$(if $(and $(CONFIG_GENSIO_SCTP), $(CONFIG_IPV6)),with,without)-sctp \` -- I am not sure I am using `and` correctly here, but it seems to work....

That said, I'm not going to rush into making any pull-requests until others have chimed in.

mhei, 2022-03-29:
Good point.I think, I'd not change the configure but rather add the @IPV6 depend on GENSIO_SCTP in menuconfig so that it is hidden until IPV6 is enabled. But on the other hand, carrying a "workaround" in multiple places does not make things better...

NeoRaider, 2022-03-29:
I don't have a better solution either... my personal opinion is that OpenWrt's requirements have grown out of Kconfig's capabilities years ago and we need something completely different, but people like menuconfig, so I assume it's going to stay for now.

I can explain what the mentioned [package-metadata.pl](https://github.com/openwrt/openwrt/blob/1c7e146f7e0f42e5e414c6cff0c73f75b405e0dd/scripts/package-metadata.pl#L200) comment means though: This refers to the fact that "select" dependencies are enabled regardless of their requirements, meaning for
```
config A

config B
  depends on A

config C
  select B
```
setting `C` will enable `B` as well, even when `A` is not set. OpenWrt avoids this issue by implicitly adding `depends on A` to `C` for generated package dependencies, which works great for simple cases, and results in a nightmare of circular dependencies for more complex cases...

Alexey71, 2022-03-31:
Fixed https://github.com/openwrt/packages/commit/ea3ad6b0909b2f5d8a8dcbc4e866c9ed22f3fb10

closed 2022-03-31 by Alexey71