openwrt/packages #18192: strongswan: chacha20poly1305 not working

ru67tm, 2022-03-30:
Maintainer: @[stintel](https://github.com/stintel)
Environment: x86_64 and ramips, openwrt master branch

Description:
Strongswan on Openwrt was not able to set up a SA with chacha20poly1305 or chacha20poly1305-x25519, and the log said:
`daemon.info ipsec: 10[KNL] received netlink error: Function not implemented`
while strongswan on debian works well and wireguard on openwrt which uses chacha20poly1305-x25519 as well with no problem.

ru67tm, 2022-03-31:
The problem resolved by adding `CONFIG_CRYPTO_CHACHA20POLY1305=y`. Maybe it should be added to depends of strongswan-mod-chapoly?

xwang1498, 2022-05-10:
Thank you @ru67tm for figuring this out!

Is there a way to get this working that avoids recompiling the kernel?

`opkg install kmod-crypto-lib-chacha20poly1305` doesn't work for me- I get the same netlink error after doing this. I think that's because this package is for the [8-byte nonce library version of chacha20poly1305](https://openwrt.org/packages/pkgdata/kmod-crypto-lib-chacha20poly1305) that's used by wireguard.

Is there a similar kmod I could build or install to get strongswan working? (Forgive my very amateur knowledge about this topic)

ru67tm, 2022-05-11:
@xwang1498 `kmod-crypto-lib-chacha20poly1305` actually corresponds to `CONFIG_CRYPTO_LIB_CHACHA20POLY1305`, not `CONFIG_CRYPTO_CHACHA20POLY1305`. I'm not familiar with kernel modules of OpenWrt though. Maybe you could try to add an extra kmod in /package/kernel/linux/modules/crypto.mk.

pprindeville, 2022-05-29:
> The problem resolved by adding `CONFIG_CRYPTO_CHACHA20POLY1305=y`. Maybe it should be added to depends of strongswan-mod-chapoly?

Yes, that would be here: https://github.com/openwrt/packages/blob/master/net/strongswan/Makefile#L673

xwang1498, 2022-05-30:
> > The problem resolved by adding `CONFIG_CRYPTO_CHACHA20POLY1305=y`. Maybe it should be added to depends of strongswan-mod-chapoly?
> 
> Yes, that would be here: https://github.com/openwrt/packages/blob/master/net/strongswan/Makefile#L673

Hi @pprindeville , my knowledge here is shaky, so please correct if I'm wrong:

Setting this kernel config flag from the package makefile would only work for the user who is building their own image from source, right? I don't see how `opkg install strongswan-mod-chapoly` would work for a normal user if their kernel wasn't originally compiled with that flag.

I've submitted [PR#9951](https://github.com/openwrt/openwrt/pull/9951) to include the cipher as a kmod package (`kmod-crypto-chacha20poly1305`) that could be depended on by `strongswan-mod-chapoly`.

pprindeville, 2022-05-31:
> Hi @pprindeville , my knowledge here is shaky, so please correct if I'm wrong:
>
> Setting this kernel config flag from the package makefile would only work for the user who is building their own image [...]

Go ahead and create a PR adding the dependency on `kmod-crypto-chacha20poly1305`.

pprindeville, 2022-06-01:
> Go ahead and create a PR adding the dependency on `kmod-crypto-chacha20poly1305`.

Actually, already done as https://github.com/openwrt/packages/pull/18654.

pprindeville, 2022-06-02:
Merged as #18654

closed 2022-06-02 by pprindeville