openwrt/packages #18214: ksmbd-tools needs to be downgraded

4IceG, 2022-03-31:
Hi,
I updated the system/compiled from new sources (**_openwrt-22.03 branch and Master openwrt-21.02_**) today and my network shares disappeared (Win7/10/11) after everything. I didn't know what was going on until I found an error in the log:

`
Thu Mar 31 18:24:38 2022 kern.err kernel: [  356.384727] ksmbd: No IPC daemon response for 20s
`

Then I remembered it was a known bug after the version upgrade.

So I went back to 3.4.2.

```
PKG_NAME:=ksmbd-tools
PKG_VERSION:=3.4.2
PKG_RELEASE:=$(AUTORELEASE)

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://codeload.github.com/cifsd-team/ksmbd-tools/tar.gz/$(PKG_VERSION)?
PKG_HASH:=fb79bf9201321adb33f8a5f0a12bd205a86b2d4bb057a9f98e0e9521664c2bcd
```
And everything works beautifully again.

neheb, 2022-04-01:
Done.

closed 2022-04-01 by neheb

erdoukki, 2022-04-21:
Why ksmbd was downgraded ?
Look to https://github.com/openwrt/packages/issues/18303
3.4.4 was working fine and now, no more, because of this downgrade

4IceG, 2022-10-11:
I will say .. if it works, I would not be interested in what the version is
.. but I on

PKG_NAME: = ksmbd-tools PKG_VERSION: = 3.4.4

I lost access to network resources and as far as I know it is not only me,
so .. which version is real?