openwrt/packages #18216: xtables-addons: x86_64 kernel 5.15 compilation error

trippleflux, 2022-04-01:
Maintainer: @jow- 
Environment: OpenWRT x86-64 5959c4645605ae759a331b6b700f2d9bf8547705 SNAPSHOT

Description:
Getting the following when compiling 
```
make[4]: Entering directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make  all-recursive
make[5]: Entering directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
Making all in extensions
make[6]: Entering directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions'
Xtables-addons 3.13 - Linux 5.15.31
if [ -n "/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31" ]; then make -C /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31 M=/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions modules; fi;
make[7]: Entering directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31'
  CC [M]  /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.o
In file included from ./include/linux/bitops.h:7,
                 from ./include/linux/kernel.h:12,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:12,
                 from /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c:10:
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c: In function 'pknock_seq_show':
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c:251:41: error: implicit declaration of function 'get_seconds' [-Werror=implicit-function-declaration]
  251 |                         if (time_before(get_seconds(), peer->login_sec +
      |                                         ^~~~~~~~~~~
./include/linux/typecheck.h:11:16: note: in definition of macro 'typecheck'
   11 |         typeof(x) __dummy2; \
      |                ^
./include/linux/jiffies.h:108:33: note: in expansion of macro 'time_after'
  108 | #define time_before(a,b)        time_after(b,a)
      |                                 ^~~~~~~~~~
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c:251:29: note: in expansion of macro 'time_before'
  251 |                         if (time_before(get_seconds(), peer->login_sec +
      |                             ^~~~~~~~~~~
./include/linux/typecheck.h:12:25: warning: comparison of distinct pointer types lacks a cast
   12 |         (void)(&__dummy == &__dummy2); \
      |                         ^~
./include/linux/jiffies.h:106:10: note: in expansion of macro 'typecheck'
  106 |          typecheck(unsigned long, b) && \
      |          ^~~~~~~~~
./include/linux/jiffies.h:108:33: note: in expansion of macro 'time_after'
  108 | #define time_before(a,b)        time_after(b,a)
      |                                 ^~~~~~~~~~
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c:251:29: note: in expansion of macro 'time_before'
  251 |                         if (time_before(get_seconds(), peer->login_sec +
      |                             ^~~~~~~~~~~
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c: In function 'autoclose_time_passed':
./include/linux/typecheck.h:12:25: warning: comparison of distinct pointer types lacks a cast
   12 |         (void)(&__dummy == &__dummy2); \
      |                         ^~
./include/linux/jiffies.h:105:10: note: in expansion of macro 'typecheck'
  105 |         (typecheck(unsigned long, a) && \
      |          ^~~~~~~~~
/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.c:315:55: note: in expansion of macro 'time_after'
  315 |         return peer != NULL && autoclose_time != 0 && time_after(get_seconds(),
      |                                                       ^~~~~~~~~~
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:277: /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.o] Error 1
make[8]: *** [scripts/Makefile.build:540: /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock] Error 2
make[7]: *** [Makefile:1868: /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions] Error 2
make[7]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.31'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[3]: *** [Makefile:179: /home/username/works/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.built] Error 2
make[3]: Leaving directory '/home/username/works/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#2.10#0.77#2.73
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/username/works/openwrt'
make[1]: *** [package/Makefile:110: /home/username/works/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:230: world] Error 2
```

trippleflux, 2022-04-02:
How to define -Wno-error for xtables-addons?, I have tried to define it in xtables-addons Makefile but it seems the build process doesn't picked it up : 
```
# uses GNU configure

EXTRA_CFLAGS += -Wno-error -Wno-error=implicit-function-declaration
TARGET_CFLAGS += -Wno-error -Wno-error=implicit-function-declaration
```

neheb, 2022-04-03:
https://github.com/openwrt/packages/pull/18226

trippleflux, 2022-04-05:
Closed as fixed by @neheb