openwrt/packages #18219: ooniprobe: v3.5.1 (21.02) does not start on x86

ValdikSS, 2022-04-01:
Maintainer: @ja-pa, @neheb
Environment: 21.02 x86 generic

Description:
ooniprobe main executable from ooniprobe package does not start, at least on x86 platform (have not tested other platforms).

```
root@OpenWrt:~# ooniprobe 
panic: qtls.ClientHelloInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-15.init.0()
        github.com/marten-seemann/qtls-go1-15@v0.1.1/unsafe.go:20 +0x191

root@OpenWrt:~# apitool
panic: qtls.ClientHelloInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-15.init.0()
        github.com/marten-seemann/qtls-go1-15@v0.1.1/unsafe.go:20 +0x191

root@OpenWrt:~# miniooni
panic: qtls.ClientHelloInfo doesn't match

goroutine 1 [running]:
github.com/marten-seemann/qtls-go1-15.init.0()
        github.com/marten-seemann/qtls-go1-15@v0.1.1/unsafe.go:20 +0x191
```

Any argument result in the panic above.

jamesmacwhite, 2022-04-05:
At a guess, I would suspect a Go version incompatibility. Similar errors have been seen before with AdGuardHome.

The fact the stacktrace mentions Go 1.15 suggests this, but I'm no expert. OpenWrt 21.02 builds with Go 1.17 for reference.

I've never used the package in question, but the latest release is v3.14.1 and it seems to be built against Go 1.17, so bumping the package version might just fix the issue.

ValdikSS, 2022-04-07:
>However, you'd have to confirm the latest version can build on Go 1.18, as master is using that version now, before backporting.

I have close-to-zero experience in Golang.

jamesmacwhite, 2022-04-07:
@ValdikSS Good new. Building the latest version for 21.02.2 it 

```
root@linksys-wrt3200acm:~# ooniprobe version
3.14.1

root@linksys-wrt3200acm:~# apitool --help
Usage of apitool:
  -input string
        Input of the measurement
  -mode string
        One of: check, meta, raw
  -report-id string
        Report ID of the measurement
  -v    Enable verbose mode

root@linksys-wrt3200acm:~# miniooni --version
3.14.1
```

So we should be able to backport the latest version down to 21.02 to fix it. It builds for both Go 1.17 and 1.18

jamesmacwhite, 2022-04-24:
This should be available in 21.02 now.

ValdikSS, 2022-04-24:
It works, thanks a lot!

PR #18298 (2022-04-22): [21.02] ooniprobe v3.14.2
Maintainer: @ja-pa @jefferyto 
Compile tested: armv7/mvebu
Run tested: OpenWrt 21.02 mvebu Linksys WRT3200ACM

Description:

The ooniprobe package in 21.02 is currently broken due to the version 3.5.1 using a much older version of Go (1.15) which leads to panics when trying to run commands. 

I have had to cherry-pick the previous updates to apply the latest 3.14.2 commit.

This will fix #18219, the various tools now run without errors:

```
root@linksys-wrt3200acm:~# ooniprobe version
3.14.2

root@linksys-wrt3200acm:~# apitool --help
Usage of apitool:
  -input string [... cut]
closed 2022-04-24 by ValdikSS