openwrt/packages #18265: 'obfs4proxy' new release?

rezad1393, 2022-04-07:
Maintainer: @jefferyto 

the version in opnewrt is 3 years old.
can this be updated to 0.13?

jefferyto, 2022-04-07:
Which version of OpenWrt are you using?

rezad1393, 2022-04-07:
> The stable series normally only receives bug fixes.

even for programs that are standalone ? nano gets updates. for software like tor and obfs ,technically every update is kinda security update.
though that is my opinion only.

AndersTrier, 2022-12-03:
@jefferyto 
I think this kind of software is a good candidate for ignoring normal practices - this program is used to circumvent government censorship, and not keeping it up to date might put people at risk. Version 0.0.14 comes with an important security fix:
https://forum.torproject.net/t/tor-relays-security-update-for-obfs4proxy/5152

AndersTrier, 2022-12-10:
@jefferyto has been inactive on github since Juli. I hope he's alright.

jefferyto, 2023-04-24:
In hindsight, I think obfs4proxy should have been updated in 21.02 at the time. Now I suggest upgrading to 22.03 (the current stable release) - the update to 0.0.14 was merged recently (#20843).

> @jefferyto has been inactive on github since Juli. I hope he's alright.

Thanks for the concern :slightly_smiling_face:

closed 2023-04-24 by jefferyto

rezad1393, 2023-06-02:
hi.
thank you for the update.
I hope you can track software that you are in charge of maintaining here on openwrt and update them more.

jefferyto, 2023-06-02:
>  I hope you can track software that you are in charge of maintaining here on openwrt and update them more.

This is open source, you can contribute updates if you want.

rezad1393, 2023-06-02:
> > I hope you can track software that you are in charge of maintaining here on openwrt and update them more.
> 
> This is open source, you can contribute updates if you want.

my first language is not english. sorry. I didn't mean it as a order. 

but if you meant what you said, I am not a programmer just a noob user.
my experience with compiling is mostly copy/paste from internet and typing them in terminal.
so if I try to become maintainer of this package ( I have no idea how I get access to be that) and the update requires more than just upping a number, then people that depend of that tool would not get the update.
sometimes ago I tried to learn cross compiling (just the parameters not the programming part) but I am not good enough so that I can take this responsibility on my own.

and than you again for update.