openwrt/packages #18267: Syncthing failed to build under GO1.18！！！

zxlhhyccc, 2022-04-07:
[image]

```
github.com/syncthing/syncthing/lib/stats
github.com/syncthing/syncthing/lib/weakhash
# github.com/lucas-clemente/quic-go/internal/qtls
../../vendor/github.com/lucas-clemente/quic-go/internal/qtls/go118.go:5:13: cannot use "quic-go doesn't build on Go 1.18 yet." (untyped string constant) as int value in variable declaration
github.com/syncthing/syncthing/lib/tlsutil
golang.org/x/crypto/blowfish
golang.org/x/text/runes
golang.org/x/crypto/bcrypt
`````

jefferyto, 2022-04-07:
Ping @aparcar.

(Please use the default issue template and at-mention the relevant maintainer(s).)

zxlhhyccc, 2022-04-07:
> Several Go packages were broken when Go 1.18 was updated on master/snapshot.
> 
> quic-go does support Go 1.18, but it's likely the version that syncthing is including is an older version that doesn't support it. Same issue occurred with AdGuardHome and others.
> 
> The easiest solution is seeing if SyncThing have bumped the dependency in their code and released a new version since, which it should be straight forward to bump the package version in OpenWrt, if not, a patch will be needed to bump the go.mod/go.sum file before compiling. That's been my experience anyway.

My level is limited, make patches for a few days can not solve!

jamesmacwhite, 2022-04-07:
I tried building 1.19.2 and it still failed so I'd suggest it's a potential upstream problem. I can see you opened an issue on the upstream repo but they closed it. I added some details to the issue for context.

aparcar, 2022-04-07:
Maybe the suncthing version is simply to old and I should upgrade it?

jamesmacwhite, 2022-04-07:
I tried compiling the latest release earlier and it failed still on go 1.18 errors. Reported upstream for further comments.

cotequeiroz, 2022-04-07:
I'm noob with go.  If I remove the vendor directory, then 1.19.2 builds.  I don't even know what's the purpose of `vendor/`, but I noticed that the git repo does not carry it, so I imagine it is generated for the releases.

jamesmacwhite, 2022-04-07:
@cotequeiroz That's an interesting find!

This seems to have been discovered by others as well: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=262808#c6

cotequeiroz, 2022-04-07:
Try https://github.com/cotequeiroz/packages/commit/e9f217dd545e692223e3d5a319cece0d0a52e436.

jamesmacwhite, 2022-04-07:
@cotequeiroz Yep, removing the vendor folder resolves the latest errors on 1.19.2, nice work, seems like a suitable fix. Looking at the FreeBSD conversation that seemed to be the conclusion as well, so this patch seems suitable.

cotequeiroz, 2022-04-07:
@jefferyto, what do you think about this?   
If we are indeed going to remove `vendor/` as a hack, I would hardcode the version number, to 👊  encourage a review; something like:
``` make
# When bumping versions, try to remove this Build/Prepare definition, to check if it is still needed
define Build/Prepare
	$(call Build/Prepare/Default)
	cd $(BUILD_DIR)/$(PKG_NAME)-1.19.2/$(PKG_NAME) && rm -rf vendor
endef
```

jamesmacwhite, 2022-04-07:
It's been flagged upstream for further review in the meantime: https://github.com/syncthing/syncthing/issues/8258#issuecomment-1092146807

aparcar, 2022-04-07:
Anyone interested to take over the maintainership of synching?

zxlhhyccc, 2022-04-09:
> Anyone interested to take over the maintainership of synching?

The problem has been resolved and the source code has been compiled using vebdor under GO1.18. The existing patches are compiled using their own go.mod

closed 2022-04-10 by aparcar