openwrt/packages #18303: ksmbd-tools-3.4.2 not working on OpenWrt-21.02.2

erdoukki, 2022-04-14:
I opened by mistake an issue at OpenWrt instead of Packages, so here is the next step from issue [smbd-server-3.4.2-2 no access on OpenWrt-21.02.2 #9701](https://github.com/openwrt/openwrt/issues/9701#)

ksmbd components was rebuilt and the three packages were installed on a fesh reinstall...
The all was broken by the downgrade to 3.4.2 of some components...
All was working before this downgrade and the mistake was to try "updated" packages (downgraded) because of some network latency issues tests !

Maintainer: @Andy2244 
Environment: (ARCH64, LS421-DE, OpenWrt 21.02-2)

Description:
I get a local compilation of ksmbd-tools to 3.4.4-2 from the master for the LS421DE on 21.02.2 !
Installed the 3 packages:
- ksmbd-avahi-service_3.4.4-2_arm_cortex-a9_vfpv3-d16.ipk
- ksmbd-server_3.4.4-2_arm_cortex-a9_vfpv3-d16.ipk
- ksmbd-utils_3.4.4-2_arm_cortex-a9_vfpv3-d16.ipk

My kernel is (still):
`Linux LS421DE 5.4.179 #0 SMP Wed Feb 16 20:29:10 2022 armv7l GNU/Linux`

And now, all works, simply fine and normally, again !

I can browse again by SMB from my Gnu-Linux/Mint, from another OpenWrt MVEBU, and I will test more later !

Have you any idea on this and why some parts were downgraded ?
The correct version may be partly from https://github.com/openwrt/packages/commit/3ffd540b049dd521dd62f43427f61f264396ad97 just applied on openwrt-21.02 branch of packages..
Hope it will help to get back on release branch again, I have still some others boxes broken by downgrade.

Thank in advance!

bosim, 2022-04-17:
I am also experiencing this on a fresh install of 21.02.2 on a archer c7 v4.

Noticed `kmod-fs-ksmbd` wasn't downgraded to 3.4.2.. wonder if this is causing trouble..

```
kmod-fs-ksmbd - 5.4.179+3.4.4-1
ksmbd-server - 3.4.2-2
ksmbd-utils - 3.4.2-2
```

erdoukki, 2022-04-17:
The problem *IS* the downgrade !
I have no more error or issue until I have reset the correct latest version...
But it needed a local build!

KristerLundmark, 2022-04-18:
Same here on Netgear R6220 (to small /overlay for samba, therefore I use ksmbd).

Upgraded from 21.02.1 to 21.02.2 and ksmbd stopped working, same versions as mentioned above
(kernel module 5.4.179+3.4.4-1, server + utils 3.4.2-2)

I downgraded to 21.02.1, and ksmbd works again.

erdoukki, 2022-04-19:
Thank you.
Very interesting to understand the problem!
I suppose the revert was because of some issue report that missed this point...

erdoukki, 2022-04-21:
Still some more users who complains on the forum

erdoukki, 2022-04-22:
@neheb @Andy2244 @rg4github

> Look’at https://forum.openwrt.org/t/ksmbd-samba3-4-alternative-ex-cifsd-smbd-package-support-thread/51695/231
> Why the issue asking for downgrade was accepted ?

https://forum.openwrt.org/t/ksmbd-samba3-4-alternative-ex-cifsd-smbd-package-support-thread/51695/241?u=erdoukki
> Thanks to you and everyone who answered me, but I got tired of waiting and returned to the 19.07.8 version, apparently it's too early to update.

I am very surprised of this ?!
How can this happen ???

Andy2244, 2022-04-22:
oh i'm sorry, but i did not use ksmbd for quite a while personally and are busy with RL stuff. So i'm not really in the ksmbd loop, did kinda rely on neheb there !
I will drop maintainership for ksmbd and the other packages i don't use anymore, since i cant really test those reliably.

closed 2022-05-02 by erdoukki