openwrt/packages #18314: Adblock - adblock.mail script

ruralroots, 2022-04-16:
Maintainer: @didbot
Environment: arm_cortex-a9_vfpv3-d16. Stable: 21.02.2

Line 80 in `/etc/adblock/adblock.mail` Piped variables `"${adb_mail}" "${adb_mailprofile}" "${adb_mailreceiver}" ` should not be quoted.
```
printf "%b" "${adb_mailhead}${adb_mailtext}" 2>/dev/null | "${adb_mail}" ${debug} -a "${adb_mailprofile}" "${adb_mailreceiver}" >/dev/null 2>&1
```
Fixed with:
`printf "%b" "${adb_mailhead}${adb_mailtext}" 2>/dev/null | ${adb_mail} ${debug} -a ${adb_mailprofile} ${adb_mailreceiver} >/dev/null 2>&1 `

dibdot, 2022-04-30:
What does this suppose to fix?

ruralroots, 2022-05-01:
I added email notification to Adblock. 

- Configured `/etc/config/adblock` appropriately.
- Added `adb_notify` account to my existing `/root/.msmtprc` config
-  `cd /etc/adblock` and run `./adblock.mail`
-  Adblock `Log View` Returns:
`Sat April 16 10:40:35 2022 user.info adblock-[7785]: mail sent to 'my mail@gmail.com' with rc '0' ` but no email is received at my gmail account.

I confirmed all variables used by the `# send mail ` routine are correct.
I confirmed using CLI that `adb_notify` account settings allowed receipt to my gmail account.
I added debug to msmtp and re-ran `adblock.mail` - nothing returned in stdout or logread relating to `msmtp`.

On a whim, I removed quotes from the variables as noted in my issue above and email started working. Restoring `/rom/etc/adblock/adblock.mail` returned the original issue.

I see absolutely no issues with your script, or my settings, but I couldn’t get mail to work as original. With the changes, daily cron job runs successfully.

dibdot, 2022-05-05:
Not reproducible with my gmail account - works as expected. Maybe your adblock/mail config includes leading or trailing whitespaces!? No more ideas ..., sorry.

ruralroots, 2022-05-08:
I’ll put it down to some unknown local anomalies then.  

No leading/trailing white space in `adblock or .msmtprc` configs. `adblock.mail` generates a proper log entry at script close, just doesn’t even try to call the sendmail process. 

My odd workaround resolves the issue locally at least. Thanks.

closed 2022-05-08 by ruralroots