openwrt/packages #18338: i2pd: update to 2.41 [entware ]

Regloom, 2022-04-21:
The latest available "Entware" repo package 2.39 for "mipsel" architecture [crashes](https://github.com/PurpleI2P/i2pd/issues/1738#issuecomment-1069413649). I see that packages are being updated but [i2pd version is outdated](http://bin.entware.net/mipselsf-k3.4/). 

Maintainer: @yangfl (David Yang <mmyangfl@gmail.com> )
Environment: arch = mipsel, cpu model=MIPS 34Kc V5.5 (mips32r2)

Description:
Entware repo [maintainer](https://github.com/Entware/Entware/issues/783#issuecomment-1099987934) referenced to your community for this request.

ryzhovau, 2022-04-21:
I'm afraid it's @Entware specific for now. OpenWrt guys can't do anything with that.

BKPepe, 2022-04-21:
Versions of i2pd in each OpenWrt branch for now:
master: 2.41.0
openwrt-23.02: 2.41.0
openwrt-21.02: 2.41.0

and OpenWrt 19.07 has version 2.24.0. 

We don't have 2.38.0 version in our repository, closing.

closed 2022-04-21 by BKPepe