openwrt/packages #18340: etherwake-nfqueue: DEPENDS issue

rwalli, 2022-04-21:
Maintainer: @mister-benjamin
Environment: x86_64, 22.03.0-rc1

Description:

etherwake-nfqueue not working on 22.03.0-rc1 because of missing "kmod-nft-queue" pakage.
switching "iptables-mod-nfqueue" with  "kmod-nft-queue" solves the issue.

mister-benjamin, 2022-04-22:
Thank you for reporting this issue.

Are your experiencing a runtime issue or does your build fail? The build server successfully compiled the package.

Also, I didn't see that a package `kmod-nft-queue` exists. The dependency `iptables-mod-nfqueue` indirectly pulls in `kmod-nfnetlink-queue` and `kmod-ipt-nfqueue`, which at least in 21.02. is working fine.

rwalli, 2022-04-22:
I'm sorry for being unclear!
There are no build fails  in 21.02 and 22.03.0-rc1.
The only issue is, that you can't insert a rule in 22.03.0-rc1 (eg: "nft add rule inet fw4  input counter queue") because of the missing "kmod-nft-queue" module.

> Also, I didn't see that a package kmod-nft-queue exists. The dependency iptables-mod-nfqueue indirectly pulls in kmod-nfnetlink-queue and kmod-ipt-nfqueue, which at least in 21.02. is working fine.

But it does not pull "kmod-nft-queue".

I did a fresh install of 22.03.0-rc1:

```
root@OpenWrt:~# opkg install etherwake-nfqueue
<snip>
root@OpenWrt:~# nft add rule inet fw4  input counter queue
Error: Could not process rule: No such file or directory
add rule inet fw4 input counter queue
                                ^^^^^
root@OpenWrt:~# opkg install kmod-nft-queue
<snip>
root@OpenWrt:~# opkg install kmod-nft-queue
Installing kmod-nft-queue (5.10.111-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/targets/x86/generic/packages/kmod-nft-queue_5.10.111-1_i386_pentium4.ipk
Configuring kmod-nft-queue.
root@OpenWrt:~# nft add rule inet fw4  input counter queue

```

mister-benjamin, 2022-05-03:
The related PRs are merged, so this issue can be closed.

closed 2022-05-03 by rwalli