openwrt/packages #18341: prometheus-node-exporter-lua: check for clients_call method is missing

Noki, 2022-04-21:
Maintainer: @champtar
Environment: not model specific, OpenWrt 21.02

Description:

In [hostapd_ubus_stations.lua](https://github.com/openwrt/packages/blob/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/hostapd_ubus_stations.lua#L58-L65) the method `get_clients` gets called for interfaces that might not be in the hostapd namespace and therefore the call fails:

```
Thu Apr 21 21:08:03 2022 daemon.err prometheus-node-exporter-lua[20: Failed to connect to hostapd - wpa_ctrl_open: No such file or directory
Thu Apr 21 21:08:04 2022 daemon.err prometheus-node-exporter-lua[20: Failed to connect to hostapd - wpa_ctrl_open: No such file or directory
Thu Apr 21 21:08:04 2022 daemon.info prometheus-node-exporter-lua[20: .../lua/prometheus-collectors/hostapd_ubus_stations.lua:62: attempt to index local 'clients_call' (a nil value)
```

The hostapd namespace needs to be checked for the availability of the get_clients method or the hostapd namespace should be used to get the interfaces instead of using the [network.wireless status call](https://github.com/openwrt/packages/blob/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/hostapd_ubus_stations.lua#L6).

Here is an example of the network.wireless status call (`ubus -v call network.wireless status`):

```
{
        "radio0": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "retry_setup_failed": false,
                "config": {
                        "hwmode": "11a",
                        "path": "pci0000:00/0000:00:00.0",
                        "htmode": "VHT40",
                        "channel": "36",
                        "txpower": 19,
                        "country": "DE",
                        "legacy_rates": false
                },
                "interfaces": [
                        {
                                "section": "radio0_if0",
                                "ifname": "wlan5-ff",
                                "config": {
                                        "ifname": "wlan5-ff",
                                        "mode": "ap",
                                        "encryption": "none",
                                        "ssid": "berlin.freifunk.net",
                                        "isolate": true,
                                        "network": [
                                                "dhcp"
                                        ],
                                        "mode": "ap",
                                        "isolate": true
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        },
                        {
                                "section": "radio0_if1",
                                "ifname": "wlan5-ffowe",
                                "config": {
                                        "ifname": "wlan5-ffowe",
                                        "mode": "ap",
                                        "encryption": "owe",
                                        "ssid": "berlin.freifunk.net Encrypted",
                                        "ieee80211w": 1,
                                        "isolate": true,
                                        "network": [
                                                "dhcp"
                                        ],
[... 186 lines cut ...]
        "rrm_beacon_req":{"addr":"String","mode":"Integer","op_class":"Integer","channel":"Integer","duration":"Integer","bssid":"String","ssid":"String"}
        "wnm_disassoc_imminent":{"addr":"String","duration":"Integer","neighbors":"Array","abridged":"Boolean"}
        "bss_transition_request":{"addr":"String","disassociation_imminent":"Boolean","disassociation_timer":"Integer","validity_period":"Integer","neighbors":"Array","abridged":"Boolean"}
'hostapd.wlan5-ffowe' @bbacf20d
        "reload":{}
        "get_clients":{}
        "get_status":{}
        "del_client":{"addr":"String","reason":"Integer","deauth":"Boolean","ban_time":"Integer"}
        "list_bans":{}
        "wps_start":{}
        "wps_status":{}
        "wps_cancel":{}
        "update_beacon":{}
        "get_features":{}
        "switch_chan":{"freq":"Integer","bcn_count":"Integer","center_freq1":"Integer","center_freq2":"Integer","bandwidth":"Integer","sec_channel_offset":"Integer","ht":"Boolean","vht":"Boolean","block_tx":"Boolean"}
        "set_vendor_elements":{"vendor_elements":"String"}
        "notify_response":{"notify_response":"Integer"}
        "bss_mgmt_enable":{"neighbor_report":"Boolean","beacon_report":"Boolean","bss_transition":"Boolean"}
        "rrm_nr_get_own":{}
        "rrm_nr_list":{}
        "rrm_nr_set":{"list":"Array"}
        "rrm_beacon_req":{"addr":"String","mode":"Integer","op_class":"Integer","channel":"Integer","duration":"Integer","bssid":"String","ssid":"String"}
        "wnm_disassoc_imminent":{"addr":"String","duration":"Integer","neighbors":"Array","abridged":"Boolean"}
        "bss_transition_request":{"addr":"String","disassociation_imminent":"Boolean","disassociation_timer":"Integer","validity_period":"Integer","neighbors":"Array","abridged":"Boolean"}
```

pktpls, 2022-04-22:
I've looked into this a bit since I've started to see the same errors for a mesh no-encryption interface. I'm using the regular hostapd-wolfssl package, and it seems to ignore the mesh interface and list it as unmanaged in `cat /var/state/wireless`. I'm assuming that's because hostapd might not have anything to do for no-encryption interfaces.

The first error `Failed to connect to hostapd - wpa_ctrl_open` is from hostapd_stations.lua, the second error `attempt to index local 'clients_call'` is from hostapd_ubus_stations.lua. 

Proposed fix: both collectors should do `ubus list hostapd.*` instead of `ubus call network.wireless status`, so that they only look at interfaces that hostapd can deal with.

champtar, 2022-04-25:
@PolynomialDivision ping as this is your collector

PolynomialDivision, 2022-04-26:
Fixes:
- https://github.com/openwrt/packages/pull/18384
- https://github.com/openwrt/packages/pull/18385

Noki, 2022-04-29:
The fixes look fine to me. 

@champtar can you please review and merge / cherrypick for 21.02?

champtar, 2022-04-29:
Added a small review, I prefer @pktpls to use the equivalent of `ubus list hostapd.*`

Noki, 2022-05-01:
The changes to the pull requests look good to me. I tested them and they work.

champtar, 2022-05-01:
@PolynomialDivision By "the equivalent of `ubus list hostapd.*`" I meant the lua equivalent ;)

Noki, 2022-05-18:
@champtar please merge and cherrypick the fixes for 21.02 and 22.03 if you have no further objections.

fakuivan, 2025-04-24:
As for the last line:
```
Thu Apr 21 21:08:04 2022 daemon.info prometheus-node-exporter-lua[20: .../lua/prometheus-collectors/hostapd_ubus_stations.lua:62: attempt to index local 'clients_call' (a nil value)
```

This issue is an oversight on the hostapd interface finder function. The method `find` treats the dot as a special character meaning _any character_, similarly to how grep and regex works. To match a literal dot you need to escape it using a percent sign so the result is `%.`. Here's a demonstration:

```lua
pattern = "ab-cd ab.cd"
print(pattern:find("ab."))
-- prints: 3, it matched on "ab-"
print(pattern:find("ab%."))
-- prints: 9, it matched on the "ab." at the end
```

This means the match also grabs stuff like `hostapd-auth`, which does not implement a `get_clients` method, thus the `nil` return value. Here's that seen from the cli:

```
root@cuatro-router:~# # We match on hostapd-auth
root@cuatro-router:~# ubus list | grep '^hostapd.'
hostapd-auth
hostapd.wlan-5ghz-casac
hostapd.wlan-casac
hostapd.wlan-guests
hostapd.wlan-iofft
root@cuatro-router:~# # Here's the correct command
root@cuatro-router:~# ubus list | grep '^hostapd\.'
hostapd.wlan-5ghz-casac
hostapd.wlan-casac
hostapd.wlan-guests
hostapd.wlan-iofft
root@cuatro-router:~# # Propper call
root@cuatro-router:~# ubus call hostapd.wlan-guests get_clients '{}'
{
        "freq": 2452,
        "clients": {

        }
}
root@cuatro-router:~# # Failure
root@cuatro-router:~# ubus call hostapd-auth get_clients '{}'
Command failed: Method not found
```

The fix is basically changing line 9 here to `net:find("hostapd%.")` (bonus tip, `net.find(net, ...)` is equivalent to just `net::find(...)`):

https://github.com/openwrt/packages/blob/c6daef227b2faa3388b5a300269d263ba9cdaad2/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/hostapd_ubus_stations.lua#L4-L15