openwrt/packages #18345: DHCP static leases not save once reboot router [Openwrt adguardhome 0.107.6-1 package]

sunindersingh, 2022-04-22:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (ramips/mt7621, D-Link DIR-2660-A1, OpenWrt SNAPSHOT r19481-a5ac8ad0ba)

Description:

Hello,

Once i install adguardhome 0.107.6-1 from Openwrt there two issues i had notice.

a) It not autostart the service, so i manually change the /etc/init.d/adguardhome. "Start from 21 to 99 and Stop 89 to 120" and it fix.
b) Now second issue is after reboot it lost all DHCP static leases [Mac / IP / Hostname] entries.

In my router already overlay configure on USB stick so just wondering how to preserve the settings? Because all other settings are saved except "DHCP static leases'

```
root@OpenWrt:/# ./overlay/upper/usr/bin/AdGuardHome -v --version
AdGuard Home
Version: v0.107.6
Channel: release
Go version: go1.18.1
Build time: 2022-04-15T05:11:18Z+0000
GOOS: linux
GOARCH: mipsle
Race: false
Dependencies:
github.com/AdguardTeam/dnsproxy@v0.41.4 (sum: h1:zA8BJmWBkSL5kp4b8CblQRgIrLGzJ4IUGQ7tA1255Cw=)
github.com/AdguardTeam/golibs@v0.10.8 (sum: h1:diU9gP9qG1qeLbAkzIwfUerpHSqzR6zaBgzvRMR/m6Q=)
github.com/AdguardTeam/urlfilter@v0.15.2 (sum: h1:LZGgrm4l4Ys9eAqB+UUmZfiC6vHlDlYFhx0WXqo6LtQ=)
github.com/NYTimes/gziphandler@v1.1.1 (sum: h1:ZUDjpQae29j0ryrS0u/B8HZfJBtBQHjqw2rQ2cqUQ3I=)
github.com/aead/chacha20@v0.0.0-20180709150244-8b13a72661da (sum: h1:KjTM2ks9d14ZYCvmHS9iAKVt9AyzRSqNU1qabPih5BY=)
github.com/aead/poly1305@v0.0.0-20180717145839-3fee0db0b635 (sum: h1:52m0LGchQBBVqJRyYYufQuIbVqRawmubW3OFGqK1ekw=)
github.com/ameshkov/dnscrypt/v2@v2.2.3 (sum: h1:X9UP5AHtwp46Ji+sGFfF/1Is6OPI/SjxLqhKpx0P5UI=)
github.com/ameshkov/dnsstamps@v1.0.3 (sum: h1:Srzik+J9mivH1alRACTbys2xOxs0lRH9qnTA7Y1OYVo=)
github.com/beefsack/go-rate@v0.0.0-20200827232406-6cde80facd47 (sum: h1:M57m0xQqZIhx7CEJgeLSvRFKEK1RjzRuIXiA3HfYU7g=)
github.com/cheekybits/genny@v1.0.0 (sum: h1:uGGa4nei+j20rOSeDeP5Of12XVm7TGUd4dJA9RDitfE=)
github.com/digineo/go-ipset/v2@v2.2.1 (sum: h1:k6skY+0fMqeUjjeWO/m5OuWPSZUAn7AucHMnQ1MX77g=)
github.com/fsnotify/fsnotify@v1.5.1 (sum: h1:mZcQUHVQUQWoPXXtuf9yuEXKudkV2sx1E06UadKWpgI=)
github.com/go-ping/ping@v0.0.0-20211130115550-779d1e919534 (sum: h1:dhy9OQKGBh4zVXbjwbxxHjRxMJtLXj3zfgpBYQaR4Q4=)
github.com/google/go-cmp@v0.5.6 (sum: h1:BKbKCqvP6I+rmFHt06ZmyQtvB8xAkWdhFyr0ZUNZcxQ=)
github.com/google/gopacket@v1.1.19 (sum: h1:ves8RnFZPGiFnTS0uPQStjwru6uO6h+nlr9j6fL7kF8=)
github.com/google/renameio@v1.0.1 (sum: h1:Lh/jXZmvZxb0BBeSY5VKEfidcbcbenKjZFzM/q0fSeU=)
github.com/google/uuid@v1.3.0 (sum: h1:t6JiXgmwXMjEs8VusXIJk2BXHsn+wx8BZdTaoZ5fu7I=)
github.com/insomniacslk/dhcp@v0.0.0-20211214070828-5297eed8f489 (sum: h1:jhdHqd7DxBrzfuFSoPxjD6nUVaV/1RIn9aHA0WCf/as=)
github.com/josharian/native@v0.0.0-20200817173448-b6b71def0850 (sum: h1:uhL5Gw7BINiiPAo24A2sxkcDI0Jt/sqp1v5xQCniEFA=)
github.com/kardianos/service@v1.2.0 (sum: h1:bGuZ/epo3vrt8IPC7mnKQolqFeYJb7Cs8Rk4PSOBB/g=)
github.com/lucas-clemente/quic-go@v0.25.0 (sum: h1:K+X9Gvd7JXsOHtU0N2icZ2Nw3rx82uBej3mP4CLgibc=)
github.com/marten-seemann/qtls-go1-18@v0.1.0-beta.1 (sum: h1:EnzzN9fPUkUck/1CuY1FlzBaIYMoiBsdwTNmNGkwUUM=)
github.com/mdlayher/ethernet@v0.0.0-20190606142754-0394541c37b7 (sum: h1:lez6TS6aAau+8wXUP3G9I3TGlmPFEq2CTxBaRqY6AGE=)
github.com/mdlayher/netlink@v1.5.0 (sum: h1:r4fa439+SsMarM0rMONU3iSshSV3ArVqJl6H/zjrhh4=)
github.com/mdlayher/raw@v0.0.0-20211126142749-4eae47f3d54b (sum: h1:MHcTarUMC4sFA7eiyR8IEJ6j2PgmgXR+B9X2IIMjh7A=)
github.com/mdlayher/socket@v0.1.1 (sum: h1:q3uOGirUPfAV2MUoaC7BavjQ154J7+JOkTWyiV+intI=)
github.com/ainar-g/dns@v1.1.49-0.20220411125901-8a162bbc18d8 (sum: h1:Hp2waLwK989ui3bDkFpedlIHfyWdZ77gynvd+GPEqXY=)
github.com/patrickmn/go-cache@v2.1.0+incompatible (sum: h1:HRMgzkcYKYpi3C8ajMPV8OFXaaRUnok+kx1WdO15EQc=)
github.com/pkg/errors@v0.9.1 (sum: h1:FEBLx1zS214owpjy7qsBeixbURkuhQAwrK5UwLGTwt4=)
github.com/satori/go.uuid@v1.2.0 (sum: h1:0uYX9dsZ2yD7q2RtLRtPSdGDWzjeM3TbMJP9utgA0ww=)
github.com/ti-mo/netfilter@v0.4.0 (sum: h1:rTN1nBYULDmMfDeBHZpKuNKX/bWEXQUhe02a/10orzg=)
github.com/u-root/uio@v0.0.0-20210528151154-e40b768296a7 (sum: h1:XMAtQHwKjWHIRwg+8Nj/rzUomQY1q6cM3ncA0wP8GU4=)
go.etcd.io/bbolt@v1.3.6 (sum: h1:/ecaJf0sk1l4l6V4awd65v2C3ILy7MSj+s/x1ADCIMU=)
golang.org/x/crypto@v0.0.0-20211215153901-e495a2d5b3d3 (sum: h1:0es+/5331RGQPcXlMfP+WrnIIS6dNnNRe0WB02W0F4M=)
golang.org/x/net@v0.0.0-20220403103023-749bd193bc2b (sum: h1:vI32FkLJNAWtGD4BwkThwEy6XS7ZLLMHkSkYfF8M0W0=)
golang.org/x/sync@v0.0.0-20210220032951-036812b2e83c (sum: h1:5KslGYwFpkhGh+Q16bwMP3cOontH8FOep7tGV86Y7SQ=)
golang.org/x/sys@v0.0.0-20220406163625-3f8b81556e12 (sum: h1:QyVthZKMsyaQwBTJE04jdNN0Pp5Fn9Qga0mrgxyERQM=)
golang.org/x/text@v0.3.7 (sum: h1:olpwvP2KacW1ZWvsR7uQhoyTYvKAupfQrRGBFM352Gk=)
gopkg.in/natefinch/lumberjack.v2@v2.0.0 (sum: h1:1Lc07Kr7qY4U2YPouBjpCLxpiyxIVoxqXgkXLknAOE8=)
gopkg.in/yaml.v2@v2.4.0 (sum: h1:D8xgwECY7CYvx+Y2n4sBz93Jn9JRvxdiyyo8CTfuKaY=)
howett.net/plist@v1.0.0 (sum: h1:7CrbWYbPPO/PyNy38b2EB/+gYbjCe2DXBxgtOOZbSQM=)
```

jamesmacwhite, 2022-04-22:
a)

It is possible there is a race condition on startup for AdGuard Home. There are a few potential network related conditions that will possibly be required but not available at the time AdGuard Home started, these mainly include:

* Binding to an interface i.e. interface not available
* ipset
* NTP

If for example, a bind address is not available at the time of startup this will cause AdGuardHome to error and not start.

Changing the startup value to such a high value is not recommended. You are leaving your router without DNS functionality (if AdGuardHome is the primary DNS resolver) for much longer than it should be. The better solution is to investigate the race condition you are experiencing and potentially mitigate it or create a custom init script that runs before AdGuard Home attempts to start, to prevent it.

You should be able to understand why AdGuard Home did not start by looking at syslog, it should tell you what triggered the failure.

```
logread -e adguardhome
```

b)

By default, adguardhome uses `/var/adguardhome` to store it's data. Under a default OpenWrt build `/var` is a symlink to the `/tmp` directory. You therefore might want to change this in order to preserve data, this could be a path such as /opt or /mnt if you have USB storage. You can adjust this by editing `/etc/config/adguardhome` and changing the `workdir` path then restart AdGuard Home.

sunindersingh, 2022-04-22:
> logread -e adguardhome

### Current Log after fixing startup script. 

a) service adguardhome disable
b) vi /etc/init.d/adguardhome. Change Start from 21 to 99 and Stop 89 to 120.
c) service adguardhome enable
d) reboot

logread -e adguardhome
Thu Apr 21 23:34:12 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:12.527610 [info] Initializing auth module: /tmp/adguardhome/data/sessions.db
Thu Apr 21 23:34:14 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:14.581935 [info] saving filter 1 contents to: /tmp/adguardhome/data/filters/1.txt
Thu Apr 21 23:34:15 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:15.347152 [info] saving filter 2 contents to: /tmp/adguardhome/data/filters/2.txt
Thu Apr 21 23:34:16 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:16.147069 [info] saving filter 1650206064 contents to: /tmp/adguardhome/data/filters/1650206064.txt
Thu Apr 21 23:34:16 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:16.345929 [info] saving filter 1650206065 contents to: /tmp/adguardhome/data/filters/1650206065.txt
Thu Apr 21 23:34:20 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:20.503423 [info] saving filter 1650206066 contents to: /tmp/adguardhome/data/filters/1650206066.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.189991 [info] saving filter 1650206067 contents to: /tmp/adguardhome/data/filters/1650206067.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.309499 [info] saving filter 1650206068 contents to: /tmp/adguardhome/data/filters/1650206068.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.338724 [info] saving filter 1650206069 contents to: /tmp/adguardhome/data/filters/1650206069.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.367674 [info] saving filter 1650206070 contents to: /tmp/adguardhome/data/filters/1650206070.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.394533 [info] saving filter 1650206071 contents to: /tmp/adguardhome/data/filters/1650206071.txt
Thu Apr 21 23:34:22 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:22.538810 [info] saving filter 1650206072 contents to: /tmp/adguardhome/data/filters/1650206072.txt
Thu Apr 21 23:34:24 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:24.074773 [info] saving filter 1650206073 contents to: /tmp/adguardhome/data/filters/1650206073.txt
Thu Apr 21 23:34:24 2022 daemon.err AdGuardHome[2379]: 2022/04/21 18:04:24.099996 [info] saving filter 1650206074 contents to: /tmp/adguardhome/data/filters/1650206074.txt

### Previous Log, it stop autostart after reboot

logread | grep -i AdGuard
Thu Apr 21 22:30:46 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:46.721464 [info] AdGuard Home, version v0.107.6
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.007858 [info] Initializing auth module: /tmp/adguardhome/data/sessions.db
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.009930 [info] auth: initialized.  users:1  sessions:0
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.010222 [info] Initialize web module
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.074235 [error] starting dhcp server: dhcpv4: finding interface br-lan by name: route ip+net: no such network interface
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.081412 [info] AdGuard Home is available at the following addresses:
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.082612 [info] Go to http://192.168.1.1:8080
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.089143 [info] Stopped
Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.090084 [fatal] listen tcp 192.168.1.1:8080: bind: cannot assign requested address
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.806111 [info] AdGuard Home, version v0.107.6
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.898028 [info] Initializing auth module: /tmp/adguardhome/data/sessions.db
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.898793 [info] auth: initialized.  users:1  sessions:0
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.899049 [info] Initialize web module
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.909383 [info] dhcpv4: listening
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.912810 [info] AdGuard Home is available at the following addresses:
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.913027 [info] Go to http://192.168.1.1:8080
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: [dhcpv4] 2022/04/21 17:08:01 Server listening on 0.0.0.0:67
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: [dhcpv4] 2022/04/21 17:08:01 Ready to handle requests
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.965192 [info] Starting the DNS proxy server
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.965414 [info] Ratelimit is enabled and set to 20 rps
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.965527 [info] The server is configured to refuse ANY requests
Thu Apr 21 22:38:01 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:08:01.965632 [info] DNS cache is enabled
[... cut ...]
Thu Apr 21 22:56:31 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:26:31.282466 [info] dhcp: stored 4 leases in db
Thu Apr 21 22:56:31 2022 daemon.err AdGuardHome[4222]: [dhcpv4] 2022/04/21 17:26:31 Handling request from 0.0.0.0:68
Thu Apr 21 22:56:31 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:26:31.332789 [info] dhcp: stored 4 leases in db
Thu Apr 21 22:56:31 2022 daemon.err AdGuardHome[4222]: [dhcpv4] 2022/04/21 17:26:31 Handling request from 0.0.0.0:68
Thu Apr 21 22:56:31 2022 daemon.err AdGuardHome[4222]: 2022/04/21 17:26:31.390094 [info] dhcp: stored 4 leases in db

sunindersingh, 2022-04-22:
> By default, adguardhome uses `/var/adguardhome` to store it's data. Under a default OpenWrt build `/var` is a symlink to the `/tmp` directory. You therefore might want to change this in order to [...]

cat /etc/config/adguardhome
config adguardhome config
        # Where to store persistent data by AdGuard Home
        option workdir /var/adguardhome

And this folder orders, My USB disk already mounted on overlay and any settings changes save even after reboot except "DHCP static leases"

find / -iname adg*
/etc/config/adguardhome
/etc/init.d/adguardhome
/etc/adguardhome.yaml
/lib/upgrade/keep.d/adguardhome
/overlay/upper/etc/config/adguardhome
/overlay/upper/etc/adguardhome.yaml
/overlay/upper/etc/init.d/adguardhome
/overlay/upper/lib/upgrade/keep.d/adguardhome
/overlay/upper/usr/bin/AdGuardHome
/overlay/upper/usr/lib/opkg/info/adguardhome.conffiles
/overlay/upper/usr/lib/opkg/info/adguardhome.control
/overlay/upper/usr/lib/opkg/info/adguardhome.prerm
/overlay/upper/usr/lib/opkg/info/adguardhome.postinst
/overlay/upper/usr/lib/opkg/info/adguardhome.list
/sys/fs/cgroup/services/adguardhome
/tmp/adguardhome
/usr/bin/AdGuardHome
/usr/lib/opkg/info/adguardhome.conffiles
/usr/lib/opkg/info/adguardhome.control
/usr/lib/opkg/info/adguardhome.prerm
/usr/lib/opkg/info/adguardhome.postinst
/usr/lib/opkg/info/adguardhome.list

jamesmacwhite, 2022-04-22:
> Thu Apr 21 22:30:47 2022 daemon.err AdGuardHome[1461]: 2022/04/21 17:00:47.074235 [error] starting dhcp server: dhcpv4: finding interface br-lan by name: route ip+net: no such network interface

This looks specific to the DHCP service, it would appear when AdGuard Home tried to start, it could not find the interface to bind to.

sunindersingh, 2022-04-22:
Yes this one is fixed by adding delay from 21 to 99 in startup script and its fixed as of now.

sunindersingh, 2022-04-22:
So my only issue is right now i am not able to trace where it storing "DHCP static leases". If I made any changes into Adguard it save all settings except "DHCP static leases". Once I reboot all setting change preserve expect "DHCP static leases"

jamesmacwhite, 2022-04-22:
Change the workdir value to a path that's not `/tmp/adguardhome`. Anything in /tmp is going to be lost on a reboot. If you've got an exroot/USB storage configuration already.

Stop adguardhome
Move the current /tmp/adguardhome directory to somewhere like /mnt or /opt
Update the workdir value in `/etc/config/adguardhome`
Restart adguardhome.

You will also notice that your DNS stats are also lost for the same reason.

sunindersingh, 2022-04-22:
Ok let me try. Wait

jamesmacwhite, 2022-04-22:
Within `/etc/config/adguardhome` is the workdir value. You need to change the path to the adguardhome data folder there. Where it exists on your router is entirely up to you, I would use /opt/ or /mnt so /opt/adguardhome or /mnt/adguardhome, providing that this is backed USB storage and isn't being written to your router flash, as this will cause heavy write cycles, so external storage recommended.

sunindersingh, 2022-04-22:
> Within `/etc/config/adguardhome`

find / -iname adguardhome
/etc/config/adguardhome
/etc/init.d/adguardhome
/lib/upgrade/keep.d/adguardhome
/overlay/upper/etc/config/adguardhome
/overlay/upper/etc/init.d/adguardhome
/overlay/upper/lib/upgrade/keep.d/adguardhome
/overlay/upper/usr/bin/AdGuardHome
/sys/fs/cgroup/services/adguardhome
/tmp/adguardhome
/usr/bin/AdGuardHome

I have so many adguardhome folder name with default opkg install adguardhome. I am not sure which one is mapped to "option workdir /var/adguardhome" right now. Any idea how to check /var link to USB stick by default by Openwrt??

jamesmacwhite, 2022-04-22:
It's currently mapped to /tmp/adguardhome. The default value of /var/adguardhome, is actually /tmp/adguardhome because /var is a symlink to the /tmp directory.

sunindersingh, 2022-04-22:
> It's currently mapped to /tmp/adguardhome. The default value of /var/adguardhome, is actually /tmp/adguardhome because /var is a symlink to the /tmp directory.

So in that case should i change "option workdir /var/adguardhome" to "option workdir /tmp/adguardhome" or "option workdir /overlay/upper/etc/config/adguardhome" ??

jamesmacwhite, 2022-04-22:
You need to have USB backed storage first. What's the output of:

```
df -h
```

I need to see what your current storage mount points are, before recommending a new location.

sunindersingh, 2022-04-22:
root@OpenWrt:~# grep -e /overlay /etc/mtab
/dev/sda1 /overlay ext4 rw,relatime 0 0
overlayfs:/overlay / overlay rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work 0 0
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.8M      3.8M         0 100% /rom
tmpfs                   122.1M      6.1M    116.0M   5% /tmp
/dev/sda1                 7.3G      3.9G      3.0G  57% /overlay
overlayfs:/overlay        7.3G      3.9G      3.0G  57% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/ubi0_1              24.3M      4.5M     18.6M  19% /rwm

Even in this article /var mapping is not mention anywhere. I used same instruction to configure USB stick. 

https://openwrt.org/docs/guide-user/additional-software/extroot_configuration

sunindersingh, 2022-04-22:
> df -h

root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.8M      3.8M         0 100% /rom
tmpfs                   122.1M      6.1M    116.0M   5% /tmp
/dev/sda1                 7.3G      3.9G      3.0G  57% /overlay
overlayfs:/overlay        7.3G      3.9G      3.0G  57% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/ubi0_1              24.3M      4.5M     18.6M  19% /rwm

jamesmacwhite, 2022-04-22:
So in your case as you've got USB storage configured on your `/overlay` partition, you can specify a path from root, so I'd do this.

Update workdir to /opt/adguardhome in the `/etc/config/adguardhome` config.

```
service adguardhome stop
mkdir /opt
mv /tmp/adguardhome /opt
service adguardhome start
```

You AdguardHome data is now on a persistent storage path.

sunindersingh, 2022-04-22:
> So in your case as you've got USB storage configured on your `/overlay` partition, you can specify a path from root, so I'd do this.
> 
> Update `workdir` to `/opt/adguardhome` in the `/etc/config/adguardhome` config.
> 
> ```
> service adguardhome stop
> mkdir /opt
> mv /tmp/adguardhome /opt
> service adguardhome start
> ```
> 
> You AdguardHome data is now on a persistent storage path. It's now being written to the USB, not /tmp or flash space.

Ok let me check. wait please.

sunindersingh, 2022-04-22:
> service adguardhome start

Nope its not saving 

[image]

jamesmacwhite, 2022-04-22:
Are your DNS logs and stats maintained across reboots? If not, I'd suggest the data isn't persistent.

sunindersingh, 2022-04-22:
> Are your DNS logs and stats maintained across reboots? If not, I'd suggest the data isn't persistent.

Did'nt get you. I am using AdGuardHome DHCP service instead of Openwrt DNS/DHCP packs.

The objective is to assign Dynamic IP and Static IP reserve for device. Here Dynamic IP is working fine. Only issue Static IP reserve [DHCP static lease] flush once reboot. Did i misunderstood the feature??

Here the steps which i follow :- 

Here the steps i use:-

Set static IP to desktop.
ssh openwrt
service adguard stop
service odhcpd stop / service odhcpd disable
service dnsmasq stop / service dnsmasq disable
service adguard start
https://x.x.x.x:8080/ > settings > dhcp > enable dhcp > interface br-lan
That openwrt pack startup script need to edit. After reboot issue is fixed.

service adguardhome disable
vi /etc/init.d/adguardhome. Change Start from 21 to 99 and Stop 89 to 120.
service adguardhome enable
reboot

service adguardhome stop
mkdir /opt
mv /tmp/adguardhome /opt
vi /etc/config/adguardhome and workdir to /opt/adguardhome
service adguardhome start

add DHCP static lease

Reboot and Check

GeorgeSapkin, 2025-10-30:
@sunindersingh is this still relevant?

GeorgeSapkin, 2026-02-02:
No response. Closing as stale.

closed 2026-02-02 by GeorgeSapkin

noguxx, 2026-11-16:
@GeorgeSapkin
I use adguardhome to replace -odhcp6c -odhcpd-ipv6only -dnsmasq. 
The way I build adguardhome, will include leases.json, excluding other adguardhome stats, when a backup or firmware upgrade is made.

```
--- a/feeds/packages/net/adguardhome/Makefile
+++ b/feeds/packages/net/adguardhome/Makefile
@@ -48,6 +48,7 @@
 
 define Package/adguardhome/conffiles
 /etc/adguardhome/adguardhome.yaml
+/etc/adguardhome/leases.json
 /etc/config/adguardhome
 endef
 
```
______________________
patches/001-dhcp_leases_in_config_dir.patch
```
--- a/internal/home/home.go
+++ b/internal/home/home.go
@@ -335,11 +335,13 @@
 	confModifier agh.ConfigModifier,
 	httpReg aghhttp.Registrar,
 	workDir string,
+	confPath string,
 	hc *aghnet.HostsContainer,
 ) (err error) {
 	//lint:ignore SA1019 Migration is not over.
 	config.DHCP.WorkDir = workDir
 	config.DHCP.DataDir = filepath.Join(workDir, dataDir)
+	config.DHCP.ConfDir = filepath.Dir(confPath)
 	config.DHCP.HTTPReg = httpReg
 	config.DHCP.CommandConstructor = executil.SystemCommandConstructor{}
 	config.DHCP.Logger = logger.With(slogutil.KeyPrefix, "dhcpd")
@@ -805,7 +807,7 @@
 		confPath,
 	)
 
-	err = initContextClients(ctx, baseLogger, sigHdlr, confModifier, httpReg, workDir, hc)
+	err = initContextClients(ctx, baseLogger, sigHdlr, confModifier, httpReg, workDir, confPath, hc)
 	fatalOnError(err)
 
 	tlsMgr, err := initTLS(ctx, baseLogger, sigHdlr, confModifier, httpReg)
--- a/internal/dhcpd/config.go
+++ b/internal/dhcpd/config.go
@@ -54,6 +54,10 @@
 
 	// DataDir is used to store DHCP leases.
 	DataDir string `yaml:"-"`
+
+	// ConfDir is the directory of the configuration file, in which the file
+	// with stored DHCP leases is kept.
+	ConfDir string `yaml:"-"`
 
 	// dbFilePath is the path to the file with stored DHCP leases.
 	dbFilePath string `yaml:"-"`
--- a/internal/dhcpd/dhcpd.go
+++ b/internal/dhcpd/dhcpd.go
@@ -119,7 +119,7 @@
 
 			LocalDomainName: conf.LocalDomainName,
 
-			dbFilePath: filepath.Join(conf.DataDir, dataFilename),
+			dbFilePath: filepath.Join(conf.ConfDir, dataFilename),
 		},
 	}

```