openwrt/packages #18349: grep does not filter with 2 brackets together

Jeronimo17, 2022-04-22:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
`echo Rocío | grep "Roc[ií][oó]" -E`

```
Formating code blocks by wrapping them with pairs of ```
```

micmac1, 2022-04-22:
Hi!

The grep you're using is from busybox I reckon. In that case the packages repo is the wrong place for this issue.

On Debian I see the same behavior.

```
deb@deb:~$ echo Rocío | busybox grep "Roc[ií][oó]" -E
deb@deb:~$ echo Rocío | busybox grep "R[oó]cí[oó]" -E
Rocío
deb@deb:~$
```
In my humble opinion this is not an OpenWrt issue and should be reported to [busybox upstream]([busybox.net](https://busybox.net/)).

Jeronimo17, 2022-04-23:
Yes it is busybox

Thank you

closed 2022-04-23 by Jeronimo17