openwrt/packages #18451: miniupnpd: Luci-app doesnt show active redirects

Damian-208, 2022-05-05:
Maintainer: common packages maintainers
Environment: (MediaTek MT7628AN ver:1 eco:2, TP-Link TL-WR840N v4, OpenWrt 22.03.0-rc1 r19302-df622768da)

Description:

```
   .At the moment Luci-app-UPnP doesn't show the active redirects opened by the package. I know the package works because I can see the redirects when I check firewall(nftables) in Luci but I think the problem is because the code for the miniupnpd package is tied to iptables. It's a very minor issue so please can this be resolved? 🙏🏽🙏🏽
```
[image]
[image]

kode54, 2022-06-13:
Here, try this diff/revision patch, or replacement, for `/usr/libexec/rpcd/luci.upnp` that correctly handles the display, but I'm not exactly sure if the deletion works properly.

https://gist.github.com/kode54/ae329a98df2ebdfd22f7a31ded20311d/revisions

kode54, 2022-06-13:
Okay, I've since fixed both description reading and deletion. It appears to be fully working with firewall4 now.

Ansuel, 2022-06-13:
Can you create a pr? You also tested the deletion right?

kode54, 2022-06-13:
I verified deletion works. Where do I file the PR?

closed 2022-06-22 by Damian-208

firewire10000, 2022-09-04:
I have tried the **luci.upnp** file under **/usr/libexec/rpcd/luci.upnp** but I'm still seeing a blank table.