openwrt/packages #18473: xtables-addons-3.19  ISO C90 forbids variable length array 'buf' [-Werror=vla]

suyuan168, 2022-05-08:
Maintainer: @jow- <jo@mein.io>
a53, ipq60xx, OpenWrt 22.02)

 CC [M]  /home/openmptcprouter/antrouter618/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.19/extensions/LUA/byte_array.o
/home/openmptcprouter/antrouter618/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.19/extensions/LUA/byte_array.c: In function 'byte_array_to_string':
/home/openmptcprouter/antrouter618/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.19/extensions/LUA/byte_array.c:110:2: error: ISO C90 forbids variable length array 'buf' [-Werror=vla]
  uint8_t buf[(array->length * 3) + 255];
  ^~~~~~~
/home/openmptcprouter/antrouter618/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.19/extensions/LUA/byte_array.c:112:2: error: ISO C90 forbids variable length array 'res' [-Werror=vla]
  char res[255 + (array->length * 3)]; /* make sure the buffer is big enough*/
  ^~~~
cc1: all warnings being treated as errors
make[9]: *** [scripts/Makefile.build:297: /home/openmptcprouter/antrouter618/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.19/extensions/LUA/byte_array.o] Error 1

In any case, I will get the following error.
gcc 8.4
Default openwrt.
Passed occasionally once, but didn't compile properly anyway.
This xtables-addons also bothered me back in 3.18, lots of lua bugs.
For example version 3.18 encounters:
make[6]: Entering directory '/home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/linux-5.4.164'
  CC [M] /home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.18/extensions/LUA/xt_LUA_target.o
/home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.18/extensions/LUA/xt_LUA_target.c: In function 'load_script_into_state':
/home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.18/extensions/LUA/xt_LUA_target.c:139:19: error: logical not is only applied to the left hand side of comparison [-Werror=logical-not-parentheses]
  if (!script_size > 0) {
                   ^
/home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.18/extensions/LUA/xt_LUA_target.c:139:6: note: add parentheses around left hand side expression to silence this warning
  if (!script_size > 0) {
      ^~~~~~~~~~~~~
      ( )
cc1: all warnings being treated as errors
scripts/Makefile.build:297: recipe for target '/home/openmptcprouter/3grouter/x8/source/build_dir/target-aarch64_cortex-a53_musl/linux-ipq60xx_generic/xtables-addons-3.18/extensions/LUA/xt_LUA_target.o' failed

pprindeville, 2022-05-09:
You can submit a patch upstream; Jan is fairly receptive and timely.

suyuan168, 2022-05-09:
> You can submit a patch upstream; Jan is fairly receptive and timely.

Thanks for your answer. Submit a patch upstream? where. Sorry I didn't get it.

closed 2022-05-11 by suyuan168