openwrt/packages #18476: node: build failure on buildbot

LGA1150, 2022-05-09:
Maintainer: @nxhack 
Environment: arm/mips, 21.02/22.03/snapshot

Description:
See https://downloads.openwrt.org/releases/faillogs-22.03/arm_cortex-a15_neon-vfpv4/packages/node/compile.txt
Looks like the buildbot lacks some dependencies.
```
In file included from /usr/include/c++/8/memory:62,
                 from ../deps/v8/src/libplatform/default-foreground-task-runner.h:8,
                 from ../deps/v8/src/libplatform/default-foreground-task-runner.cc:5:
/usr/include/c++/8/bits/stl_algobase.h:59:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [tools/v8_gypfiles/v8_libplatform.host.mk:145: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/node-v14.18.3/out/Release/obj.host/v8_libplatform/deps/v8/src/libplatform/default-foreground-task-runner.o] Error 1
rm a131f4691e1e91feaabca64379a2b37b8530f7d3.intermediate 6f2b8c87ebb9c01ea8d5d85184e36e994a0d994d.intermediate
make[4]: *** [Makefile:109: node] Error 2
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/node-v14.18.3'
make[3]: *** [Makefile:160: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/node-v14.18.3/.built] Error 2
time: package/feeds/packages/node/compile#369.19#25.90#402.66
```

nxhack, 2022-05-09:
https://github.com/openwrt/packages/pull/17076

closed 2022-05-09 by BKPepe

aloisklink, 2022-11-23:
> Is the buildbot fixed? `node` package is still missing in opkg source?

This still seems like it's an issue.

As of `Wed Nov 23 00:52:41 2022`, https://downloads.openwrt.org/releases/faillogs-22.03/arm_cortex-a15_neon-vfpv4/packages/node/compile.txt is still showing the above error.

See https://forum.openwrt.org/t/why-arent-the-node-and-node-npm-packages-available-on-arm-cortex-a9-vfpv3-d16-in-22-03-2/142722

I think the OpenWRT buildbot is missing an `apt install g++-multilib`, so I've made a merge request that should fix it, see https://gitlab.com/openwrt/buildbot/-/merge_requests/11 (at least I'm assuming that's the source-code for the buildbot)