openwrt/packages #18480: Can not remove Privoxy in Openwrt 22.03.0-rc1

Vitaliy86, 2022-05-09:
`< privoxy>: `
Openwrt 22.03.0-rc1

Can not remove
" opkg remove privoxy --force-remove
Removing package privoxy from root...
Not deleting modified conffile /etc/privoxy/user.action.
Not deleting modified conffile /etc/config/privoxy.
Collected errors:
 * xsystem: /bin/sh: Child killed by signal 9.
 * pkg_run_script: package "privoxy" prerm script returned status -1.
"

micmac1, 2022-05-09:
Hi there,

This is because the init script has a bug (or two). opkg calls `/etc/init.d/privoxy stop` and in `stop()` it first kills the pid in the pid file (which it never removes, so if you stop privoxy twice without restarting it in between it could end up killing a completely unrelated process (bug no 1)). End then, it uses `pgrep privoxy`. This is catching pids created by opkg itself, for example:

```
{privoxy.prerm} /bin/sh //usr/lib/opkg/info/privoxy.prerm remove
{privoxy} /bin/sh /etc/rc.common /etc/init.d/privoxy stop
```

And `stop()` then goes crazy and kills those too. That's a) not really funny, but it kind of made me laugh, and b) bug no 2, because it prevents opkg from removing the package.

To be able to uninstall simply comment out the pgrep line in the privoxy init fail:

`#local _tmp=$(pgrep privoxy | tr "\n" " ")`

Kind regards,
Seb

jefferyto, 2022-05-11:
I think #18293 fixes this.

jefferyto, 2022-05-17:
An update for 22.03 was just merged (#18543), it should take a day or two for the updated package to be available in opkg. (I'm not sure it would be safe to make changes to the pid file without know more about the luci app.)

closed 2022-05-17 by jefferyto