openwrt/packages #18529: gcc: does not build on Fedora 36

flyn-org, 2022-05-14:
Maintainer: @noblepepper 
Environment: x86_64 master

Description:

I received the following compile error while trying to build the gcc package on x86_64 master. The error there (as opposed to the warnings) reads `../.././gcc/config/i386/i386.c:9625:18: error: converting to 'bool' from 'std::nullptr_t'  requires direct-initialization [-fpermissive] 9625 |   bool error_p = NULL;`.

```
x86_64-openwrt-linux-musl-g++ -fno-PIE -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rttic
In file included from ../.././gcc/config/i386/i386.c:37:
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx) const':
../.././gcc/recog.h:300:51: warning: cast between incompatible function types from 'insn_]
  300 |   rtx_insn * operator () (rtx a0) const { return ((f1)func) (a0); }
      |                                                   ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx) cons:
../.././gcc/recog.h:301:59: warning: cast between incompatible function types from 'insn_]
  301 |   rtx_insn * operator () (rtx a0, rtx a1) const { return ((f2)func) (a0, a1); }
      |                                                           ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx):
../.././gcc/recog.h:302:67: warning: cast between incompatible function types from 'insn_]
  302 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2) const { return ((f3)func) (a0, }
      |                                                                   ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:303:75: warning: cast between incompatible function types from 'insn_]
  303 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3) const { return ((f4)fun}
      |                                                                           ^~~~~~~~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:304:83: warning: cast between incompatible function types from 'insn_]
  304 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4) const { return }
      |                                                                                  ~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:305:91: warning: cast between incompatible function types from 'insn_]
  305 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5) const {}
      |                                                                                  ~
In file included from ../.././gcc/config/i386/i386.c:37:
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:306:99: warning: cast between incompatible function types from 'insn_]
  306 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6)}
      |                                                                                  ~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:307:107: warning: cast between incompatible function types from 'insn]
  307 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6,}
      |                                                                                  ~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:308:115: warning: cast between incompatible function types from 'insn]
  308 |   rtx_insn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6,}
      |                                                                                  ~
../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:309:123: warning: cast between incompatible function types from 'insn]
  309 | nsn * operator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6, rtx a7}
      |                                                                                  ~

../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:310:132: warning: cast between incompatible function types from 'insn]
  310 | rator () (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6, rtx a7, rtx a8,}
      |                                                                                  ~

../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:311:141: warning: cast between incompatible function types from 'insn]
  311 | (rtx a0, rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6, rtx a7, rtx a8, rtx a9, }
      |                                                                                  ~

../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:312:150: warning: cast between incompatible function types from 'insn]
  312 | rtx a1, rtx a2, rtx a3, rtx a4, rtx a5, rtx a6, rtx a7, rtx a8, rtx a9, rtx a10, }
      |                                                                                  ~

../.././gcc/recog.h: In member function 'rtx_insn* insn_gen_fn::operator()(rtx, rtx, rtx,:
../.././gcc/recog.h:313:159: warning: cast between incompatible function types from 'insn]
  313 | tx a2, rtx a3, rtx a4, rtx a5, rtx a6, rtx a7, rtx a8, rtx a9, rtx a10, rtx a11, }
[... 34 lines cut ...]
../.././gcc/config/i386/i386.c: In function 'rtx_def* ix86_expand_special_args_builtin(co:
../.././gcc/config/i386/i386.c:37227:7: warning: this statement may fall through [-Wimpli]
37227 |       switch (icode)
      |       ^~~~~~
../.././gcc/config/i386/i386.c:37254:5: note: here
37254 |     case V64QI_FTYPE_PCCHAR_V64QI_UDI:
      |     ^~~~
make[7]: *** [Makefile:2217: i386.o] Error 1
make[7]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user/build_dir/targ'
make[6]: *** [Makefile:4236: all-gcc] Error 2
make[6]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user/build_dir/targ'
make[5]: *** [Makefile:901: all] Error 2
make[5]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user/build_dir/targ'
make[4]: *** [Makefile:228: /home/mike/Source/aquinas/openwrt-aquinas-user/build_dir/targ2
make[4]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user/feeds/packages'
time: package/feeds/packages/gcc/compile#578.66#91.23#751.90
    ERROR: package/feeds/packages/gcc failed to build.
make[3]: *** [package/Makefile:116: package/feeds/packages/gcc/compile] Error 1
make[3]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user'
make[2]: *** [package/Makefile:110: /home/mike/Source/aquinas/openwrt-aquinas-user/stagin2
make[2]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user'
make[1]: *** [/home/mike/Source/aquinas/openwrt-aquinas-user/include/toplevel.mk:230: wor2
make[1]: Leaving directory '/home/mike/Source/aquinas/openwrt-aquinas-user'
make: *** [Makefile:447: user-openwrt-x86-64-generic-ext4-combined.img] Error 2
```

neheb, 2022-05-14:
You.can fix with = {}

neheb, 2022-05-14:
The reasons this fails now is because of the new musl update.

noblepepper, 2022-05-16:
I obviously haven't had much time to work on gcc. I took over as maintainer when the previous maintainer was in a similar situation so it would make me happy to see this package get some much needed attention.

I did some work last year and had a update almost ready to allow building gcc 7, 8, 9 or 10 by selecting in menuconfig. You are welcome to use any of it that  you would like. (https://github.com/noblepepper/packages/tree/gccUpdate/devel/gcc)

@neheb If you would like anything from me to set a new maintainer please let me know

neheb, 2022-05-16:
There's a problem with this package and host clang. There's some bug that needs this fix applied to it:

https://github.com/openwrt/openwrt/commit/6e23813c1ed7494a9da3b5e59cfcafd71097c420

I have no idea how to properly pass it though.

flyn-org, 2022-05-19:
Fixed by https://github.com/openwrt/packages/pull/18532.

closed 2022-05-19 by flyn-org