openwrt/packages #18545: php8-mod-gd, possibly libgd: Error loading shared library libjpeg.so.62

M95D, 2022-05-15:
Maintainer: Michael Heimpold @mhei 
Environment: mvebu, Linksys WRT1900ACv1, git master branch

Description:
php8-mod-gd builds and installs without error, including all dependecies, but when i start php8-fpm i get this error:
```
PHP Startup: Unable to load dynamic library 'gd.so' (tried: /usr/lib/php8/gd.so (Error loading shared library libjpeg.so.62: No such file or directory (needed by /usr/lib/libgd.so.3))
```
Files that are installed are:
usr/lib/libjpeg.so.9 -> libjpeg.so.9.3.0
/usr/lib/libjpeg.so.9.3.0
There is no libjpeg.so.62.

Creating a symlink /usr/lib/libjpeg.so.62 -> libjpeg.so.9.3.0 fixes the error.

Thanks.

mhei, 2022-05-17:
Seems that you've still the older libjpeg installed. Try to remove this older package and install libjpeg-turbo instead.
BTW: libjpeg was remove from git master in 2020, php8 was introduced around one year later.

closed 2022-05-17 by mhei

M95D, 2022-05-18:
It was auto-selected when I added php-mod-gd.

mhei, 2022-05-19:
Does it means that you confirm, that you have libjpeg installed instead of libjpeg-turbo?
Hm... just to be sure: you build your image from scratch, installed it and then you encounter this problem? Can you append the output of the following command from your build dir: `scripts/diffconfig.sh  | grep jpeg`
And/or can you share your opkg config from the target? I'm wondering where you get the libjpeg from...

M95D, 2022-05-19:
I built the firmware image some time ago without php-mod-gd. When I needed it, I added php-mod-gd, <M> option, in menuconfig, and built it again. After the image and packages were (re)built, I moved all the built packages to my local repo and used "opkg update; opkg install php8-mod-gd" from the previously installed firmware. I'm sure it installed php8-mod-gd from my repo, but I didn't pay attention to what it installed as dependencies. It seems that libjpeg was installed from OpenWrt official repo, not libjpeg-turbo from my repo.

mhei, 2022-05-19:
Only the official repo 19.07.x contains libjpeg - the none-turbo variant. Starting with 21.02 only libjpeg-turbo was included in the builds and repos. The dependency of php8-mod-gd is pulling libjpeg not explicitely libjpeg-turbo. So if you still have a reference to a repo which contains libjpeg, I could imagine that this is installed.