openwrt/packages #18571: Netatalk 3.1.13 Segfault

marakoss, 2022-05-18:
Maintainer: @commodo
Environment: Qualcomm Atheros QCA9558 ver 1 rev 0, OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a

Description:
"Connection failed while trying to connect to APF EXT4 share"

cat /var/log/afpd.log outputs:

```
May 18 16:20:06.666398 cnid_metad[3843] {cnid_metad.c:509} (note:AFPDaemon): CNID Server listening on localhost:4700
May 18 16:20:06.707354 afpd[3842] {afp_config.c:174} (note:AFPDaemon): Netatalk AFP/TCP listening on interface br-lan with address 192.168.1.1:548
May 18 16:20:06.725438 afpd[3842] {afp_config.c:174} (note:AFPDaemon): Netatalk AFP/TCP listening on interface br-lan with address fde4:4ccb:9f02::1:548
May 18 16:20:06.726356 afpd[3842] {afp_config.c:174} (note:AFPDaemon): Netatalk AFP/TCP listening on interface br-lan with address fe80::a62b:b0ff:fede:c94:548
May 18 16:24:09.547578 afpd[4275] {auth.c:235} (note:AFPDaemon): Login by marc (AFP3.4)
May 18 16:24:09.865890 afpd[4275] {fault.c:123} (severe:Default): ===============================================================
May 18 16:24:09.866229 afpd[4275] {fault.c:124} (severe:Default): INTERNAL ERROR: Signal 11 in pid 4275 (3.1.13)
May 18 16:24:09.866536 afpd[4275] {fault.c:125} (severe:Default): ===============================================================
```

It worked previously on my setup, broken after my last sysupgrade (from OpenWRT v ~20). I dunno how to downgrade this specific package to pinpoint which update broke it. If somebody can show me, I'll try that.

BKPepe, 2022-05-18:
This is related to fix it: https://github.com/openwrt/packages/pull/18337

BKPepe, 2022-06-06:
It should be fixed. Closing. I see spammer somehow. I should avoid mention PRs and issues in the commit message as in the past.

closed 2022-06-06 by BKPepe

Neustradamus, 2024-11-25:
Linked to:
- https://github.com/openwrt/packages/pull/18337
- https://github.com/openwrt/packages/pull/18687
- https://github.com/openwrt/packages/pull/18696
- https://github.com/openwrt/packages/pull/24257