openwrt/packages #18578: podman: Cannot run pod deployment

gilfrade, 2022-05-19:
Maintainer: @oskarirauta 
Environment: OpenWRT 22.03-rc1

Description:

Cannot run pod deployment in podman. This may be a catatonic package issue.
Tested with this `example.kube.yaml` 
```
apiVersion: apps/v1
kind: Deployment
metadata:
  name: nginx-deployment
  labels:
    app: nginx
spec:
  replicas: 1
  selector:
    matchLabels:
      app: nginx
  template:
    metadata:
      labels:
        app: nginx
    spec:
      containers:
      - name: nginx
        image: nginx:1.14.2
        ports:
        - containerPort: 80

```

`podman play kube example.kube.yaml`

```
[  241.574698] IPv6: ADDRCONF(NETDEV_CHANGE): veth84a9040f: link becomes ready
[  241.583100] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  241.590926] cni-podman0: port 1(veth84a9040f) entered blocking state
[  241.597842] cni-podman0: port 1(veth84a9040f) entered disabled state
[  241.605608] device veth84a9040f entered promiscuous mode
[  241.612005] cni-podman0: port 1(veth84a9040f) entered blocking state
[  241.618983] cni-podman0: port 1(veth84a9040f) entered forwarding state
[  241.626428] IPv6: ADDRCONF(NETDEV_CHANGE): cni-podman0: link becomes ready
[error starting container 47a29ee45b66a256696cd102c0de4ed07f2e1aa3cd1ef29f5bac685d03e54907: cannot get namespace path unless container a70d77e1e870955d1133d7a7c3d9f8f839aff35f604826ac3a3b9d495521075c is running: container is stopped]
[  242.507859] cni-podman0: port 1(veth84a9040f) entered disabled state
[  242.517821] device veth84a9040f left promiscuous mode
```

`podman ps -a`

```
CONTAINER ID  IMAGE                           COMMAND               CREATED         STATUS                     PORTS       NAMES
a70d77e1e870  localhost/podman-pause:4.1.0-0                        19 seconds ago  Exited (1) 16 seconds ago              52d1eb4e8e22-infra
47a29ee45b66  docker.io/library/nginx:1.14.2  nginx -g daemon o...  16 seconds ago  Created                                nginx-deployment-pod-0-nginx
```

`podman logs a70d77e1e870`

```
{"msg":"exec container process (missing dynamic library?) `/catatonit`: No such file or directory","level":"error","time":"2022-05-19T11:12:45.000801865Z"}
```

If i uncomment `infra_image = "k8s.gcr.io/pause:3.4.1"` in `/etc/containers/containers.conf` then `podman play kube` works as expected.

More Info:
```
host:
  arch: amd64
  buildahVersion: 1.26.1
  cgroupControllers:
  - cpuset
  - cpu
  - io
  - memory
  - pids
  - rdma
  cgroupManager: cgroupfs
  cgroupVersion: v2
  conmon:
    package: Unknown
    path: /usr/bin/conmon
    version: 'conmon version 2.1.0, commit: '
  cpuUtilization:
    idlePercent: 93.57
    systemPercent: 3.76
    userPercent: 2.66
  cpus: 4
  distribution:
    distribution: '"openwrt"'
    version: "0"
  eventLogger: none
  hostname: STANDARD
  idMappings:
    gidmap: null
    uidmap: null
  kernel: 5.10.108
  linkmode: dynamic
  logDriver: k8s-file
  memFree: 1376780288
  memTotal: 1958440960
  networkBackend: cni
  ociRuntime:
    name: crun
    package: Unknown
    path: /usr/bin/crun
    version: |-
      crun version 1.4.5
      commit: 85649a8c561f7fbf018e5c5640a7e9370c80ce73
      spec: 1.0.0
      +SELINUX +APPARMOR +CAP +SECCOMP +EBPF +YAJL
  os: linux
  remoteSocket:
    exists: true
    path: /run/podman/podman.sock
  security:
    apparmorEnabled: false
    capabilities: CAP_CHOWN,CAP_DAC_OVERRIDE,CAP_FOWNER,CAP_FSETID,CAP_KILL,CAP_NET_BIND_SERVICE,CAP_SETFCAP,CAP_SETGID,CAP_SETPCAP,CAP_SETUID,CAP_SYS_CHROOT
    rootless: false
    seccompEnabled: true
    seccompProfilePath: /usr/share/containers/seccomp.json
    selinuxEnabled: false
  serviceIsRemote: false
  slirp4netns:
    executable: ""
    package: ""
    version: ""
  swapFree: 0
  swapTotal: 0
  uptime: 12m 44.98s
plugins:
  log:
  - k8s-file
  - none
  - passthrough
  network:
  - bridge
  - macvlan
  - ipvlan
  volume:
  - local
registries:
  search:
  - docker.io
  - registry.fedoraproject.org
  - registry.access.redhat.com
store:
  configFile: /etc/containers/storage.conf
  containerStore:
    number: 6
    paused: 0
    running: 0
    stopped: 6
  graphDriverName: overlay
  graphOptions:
    overlay.mountopt: nodev
  graphRoot: /var/lib/containers/storage
  graphRootAllocated: 979218432
  graphRootUsed: 124571648
  graphStatus:
    Backing Filesystem: tmpfs
    Native Overlay Diff: "true"
    Supports d_type: "true"
    Using metacopy: "false"
  imageCopyTmpDir: /var/tmp
  imageStore:
    number: 3
  runRoot: /run/containers/storage
  volumePath: /var/lib/containers/storage/volumes
version:
  APIVersion: 4.1.0
  Built: 0
  BuiltTime: Thu Jan  1 00:00:00 1970
  GitCommit: ""
  GoVersion: go1.18.1
  Os: linux
  OsArch: linux/amd64
  Version: 4.1.0

```

oskarirauta, 2022-05-19:
Remove port forwarding and firewalling and do it on router instead. Guidance is available on nftables issue thread..

That might just do the job as it seems like a networking issue..

Lähetetty iPhonesta

oskarirauta, 2022-05-19:
Also you could try creating pod manually, then container(S), if it works out, generate cube file with podman, try it out and if it works— make comparison - podman != kubernetes

Lähetetty iPhonesta

gilfrade, 2022-05-20:
@oskarirauta This does not seem related to network issues. POD containers do not start because infra image must be running first and its failing. 
If i use infra image from kubernetes works so it may be a build problem with catatonic since its used in infra podman-pause image.

oskarirauta, 2022-05-20:
I wonder if used pause version is newer than in previous podman release- seems that it is the problem. Although, your error Messages were heavily related to networking issues..

Lähetetty iPhonesta

oskarirauta, 2022-05-20:
I can’t seem to find a new commit from mainstream, or related issue so it’s likely openwrt specific. I also have been wondering if we still should use system we used before catatonit (can’t remember name now but nstiny or something similar) - but feel free to make a pr that would set either previous version of pause (recommended) as default, if there was a bump in version of pause, or what works for now, that kubernetes version…

Lähetetty iPhonesta

oskarirauta, 2022-05-20:
Do you have catatonit as installed? Your log also says that either it or a shared library is missing..

How about did you try what I asked? Create pod and containers manually and see if that works?

Lähetetty iPhonesta

oskarirauta, 2022-05-20:
I wonder what is the version of pause it tries, in my setup on production environment which uses still the previous version, pause’s version is 4.0.2-0.. how about you try with that version?

Lähetetty iPhonesta

gilfrade, 2022-05-20:
I only showed this error:  
```
{"msg":"exec container process (missing dynamic library?) `/catatonit`: No such file or directory","level":"error","time":"2022-05-19T11:12:45.000801865Z"}
```

And it seems to point to a missing dynamic library in catatonit.

@oskarirauta i tried your early release of podman 4.0.2 (21d2575a951980933d1d7519c0d17aa26e530278) and it fails the same way.

oskarirauta, 2022-05-20:
> @oskarirauta <https://github.com/oskarirauta> i tried your early release of podman 4.0.2 (21d2575 <https://github.com/openwrt/packages/commit/21d2575a951980933d1d7519c0d17aa26e530278>) and it fails the same way.

New version of pause seems to be 4.1.0 and seems to be working fine or at least I do not have that same error with nginx in pod on a virtual machine..

gilfrade, 2022-05-20:
Right now i have installed:

```
catatonit - 0.1.7-1
podman - 4.1.0-1
```

Fails using this POD:
```
apiVersion: apps/v1
kind: Deployment
metadata:
  name: nginx-deployment
  labels:
    app: nginx
spec:
  replicas: 1
  selector:
    matchLabels:
      app: nginx
  template:
    metadata:
      labels:
        app: nginx
    spec:
      containers:
      - name: nginx
        image: nginx:1.14.2
        ports:
        - containerPort: 80
```
OpenWRT target is x86_64 running on APU2 Board.

oskarirauta, 2022-05-20:
My cube:

apiVersion: v1
kind: Pod
metadata:
  annotations:
    bind-mount-options:/dev/log: Z
    bind-mount-options:/srv/caddy/conf: Z
    bind-mount-options:/srv/caddy/entrypoint.d: Z
    bind-mount-options:/srv/caddy/htdocs: z
    bind-mount-options:/srv/caddy/logs: z
    bind-mount-options:/srv/nginx/conf: Z
    bind-mount-options:/srv/nginx/crontabs: Z
    bind-mount-options:/srv/nginx/entrypoint.d: Z
    bind-mount-options:/srv/nginx/htdocs: z
    bind-mount-options:/srv/nginx/logs: Z
  creationTimestamp: "2022-05-20T14:33:17Z"
  labels:
    app: servers
  name: servers
spec:
  containers:
  - args:
    - caddy.sh
    - run
    - --config
    - /etc/caddy/Caddyfile
    image: docker.io/oskarirauta/caddy:latest
    name: caddy
    resources:
      limits:
        cpu: 1250m
        memory: 128Mi
    securityContext:
      capabilities:
        add:
        - CAP_SYS_NICE
        drop:
        - CAP_MKNOD
        - CAP_NET_RAW
        - CAP_AUDIT_WRITE
    volumeMounts:
    - mountPath: /scripts/entrypoint.d/
      name: srv-caddy-entrypoint.d-host-0
    - mountPath: /var/log/
      name: srv-caddy-logs-host-1
    - mountPath: /dev/log
      name: dev-log-host-2
    - mountPath: /etc/caddy/
      name: srv-caddy-conf-host-3
    - mountPath: /var/htdocs/
      name: srv-caddy-htdocs-host-4
  - args:
    - nginx
    - -g
    - daemon off;
    image: docker.io/oskarirauta/nginx:latest
    name: nginx
    resources:
      limits:
        cpu: 1240m
        memory: 128Mi
    securityContext:
      capabilities:
        add:
        - CAP_SYS_NICE
        drop:
        - CAP_MKNOD
        - CAP_NET_RAW
        - CAP_AUDIT_WRITE
    volumeMounts:
    - mountPath: /dev/log
      name: dev-log-host-0
    - mountPath: /etc/nginx/
      name: srv-nginx-conf-host-1
    - mountPath: /var/log/nginx/
      name: srv-nginx-logs-host-2
    - mountPath: /var/htdocs/
      name: srv-nginx-htdocs-host-3
    - mountPath: /scripts/entrypoint.d/
      name: srv-nginx-entrypoint.d-host-4
    - mountPath: /etc/crontabs/
      name: srv-nginx-crontabs-host-5
  restartPolicy: Never
  volumes:
  - hostPath:
      path: /srv/caddy/logs
      type: Directory
    name: srv-caddy-logs-host-1
  - hostPath:
      path: /dev/log
      type: File
    name: dev-log-host-2
  - hostPath:
      path: /srv/caddy/conf
      type: Directory
    name: srv-caddy-conf-host-3
  - hostPath:
      path: /dev/log
      type: File
    name: dev-log-host-0
  - hostPath:
      path: /srv/nginx/conf
      type: Directory
    name: srv-nginx-conf-host-1
  - hostPath:
      path: /srv/nginx/logs
      type: Directory
    name: srv-nginx-logs-host-2
  - hostPath:
      path: /srv/caddy/entrypoint.d
      type: Directory
    name: srv-caddy-entrypoint.d-host-0
  - hostPath:
      path: /srv/caddy/htdocs
      type: Directory
    name: srv-caddy-htdocs-host-4
  - hostPath:
      path: /srv/nginx/htdocs
      type: Directory
    name: srv-nginx-htdocs-host-3
  - hostPath:
      path: /srv/nginx/entrypoint.d
      type: Directory
    name: srv-nginx-entrypoint.d-host-4
  - hostPath:
      path: /srv/nginx/crontabs
      type: Directory
    name: srv-nginx-crontabs-host-5
status: {}

Maybe change kind to Pod, add creation stamp, remove apps/ from apiversion..
Maybe also replicas and selector.. Try again then.. Possible even add status: {}

Maybe that’ll work out.. No idea, how podman kube files have evolved but when they came, it was a fact that it had only some compatibility, not full..
That’s why I said that you should try creating your pod and containers manually, then use command:
podman generate kube nginx-deployment

And that will output a cube file compliant to podman perfectly..

Recommendation: possible remove container port….
Do port forwarding and firewalling on openwrt side to avoid problems which are related to firewall restart etc..
That way you can also isolate container(s) from accessing the host, but letting host access containers..

gilfrade, 2022-05-20:
@oskarirauta i will try your suggestions.

Another question, how do you port forward from openwrt side to container without knowing its IP address?

oskarirauta, 2022-05-20:
Just set a static ip for your pod…

Lähetetty iPhonesta

oskarirauta, 2022-05-20:
Read my guide I proposed earlier.. all the answers are there.

Lähetetty iPhonesta

gilfrade, 2022-05-22:
@oskarirauta tried your suggestion and manually created a POD, still failed.
```
root# podman pod create -n nginx
017558376afbb29e1a5b1e119c2bd36ebb4a8d3645a80afba54330f0633ff836

root# podman run -d --pod nginx docker.io/library/nginx:1.14.2
[ 1035.127738] IPv6: ADDRCONF(NETDEV_CHANGE): veth858f36f4: link becomes ready
[ 1035.136040] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 1035.144085] cni-podman0: port 1(veth858f36f4) entered blocking state
[ 1035.151555] cni-podman0: port 1(veth858f36f4) entered disabled state
[ 1035.159435] device veth858f36f4 entered promiscuous mode
[ 1035.165851] cni-podman0: port 1(veth858f36f4) entered blocking state
[ 1035.172949] cni-podman0: port 1(veth858f36f4) entered forwarding state
Error: cannot get namespace path unless container b42e6ef1d1970de241e1a029c218e57e1284a33ea7c273502f3e5596688af9f0 is running: container is stopped

root# podman ps -a --pod
CONTAINER ID  IMAGE                           COMMAND               CREATED             STATUS                    PORTS       NAMES               POD ID        PODNAME
b42e6ef1d197  localhost/podman-pause:4.1.0-0                        About a minute ago  Exited (1) 8 seconds ago              017558376afb-infra  017558376afb  nginx
532a24964702  docker.io/library/nginx:1.14.2  nginx -g daemon o...  9 seconds ago       Created                               kind_pascal         017558376afb  nginx

root# podman generate kube nginx
# Save the output of this file and use kubectl create -f to import
# it into Kubernetes.
#
# Created with podman-4.1.0
apiVersion: v1
kind: Pod
metadata:
  annotations:
    io.kubernetes.cri-o.ContainerType/kindpascal: container
    io.kubernetes.cri-o.SandboxID/kindpascal: nginx
    io.kubernetes.cri-o.TTY/kindpascal: "true"
    io.podman.annotations.autoremove/kindpascal: "FALSE"
    io.podman.annotations.init/kindpascal: "FALSE"
    io.podman.annotations.privileged/kindpascal: "FALSE"
    io.podman.annotations.publish-all/kindpascal: "FALSE"
  creationTimestamp: "2022-05-22T17:21:39Z"
  labels:
    app: nginx
  name: nginx
spec:
  containers:
  - image: docker.io/library/nginx:1.14.2
    name: kindpascal
    resources: {}
    securityContext:
      capabilities:
        drop:
        - CAP_MKNOD
        - CAP_NET_RAW
        - CAP_AUDIT_WRITE
  restartPolicy: Never
status: {}

```

Your suggestion of assigning a static IP to POD works and it really is easier to manage firewall. Thanks.
```
root# podman play kube nginx-example.kube.yaml --network=podman --ip=10.88.0.10
root# curl -I http://10.88.0.10:80
HTTP/1.1 200 OK
Server: nginx/1.14.2
Date: Sun, 22 May 2022 18:06:25 GMT
Content-Type: text/html
Content-Length: 612
Last-Modified: Tue, 04 Dec 2018 14:44:49 GMT
Connection: keep-alive
ETag: "5c0692e1-264"
Accept-Ranges: bytes
```
Using `infra_image = "k8s.gcr.io/pause:3.4.1"`

oskarirauta, 2022-10-11:
Although my nginx is a bit outdated and uses alpine as base..