openwrt/packages #18585: adguardhome	0.107.6-1 doesn't start automatically after reboot

d4f0, 2022-05-20:
Maintainer: @\alaviss
Environment: ARMv7 Processor rev 0 (v7l) / Netgear Nighthawk X4S R7800 / OpenWrt 21.02.3

Description:
After reboot adguardhome service fails to start.
In System Log it says Fri May 20 11:54:30 2022 daemon.err AdGuardHome[1472]: 2022/05/20 11:54:30.254075 [fatal] listen tcp 192.168.1.1:8080: bind: cannot assign requested address

It starts fine after changing /etc/init.d/adguardhome:
START=21 --> START=95
STOP=89  --> STOP=01

I used those values as they are the ones in https://github.com/AdguardTeam/AdGuardHome/blob/21905d98693a58a24560ceac5461a470beb70af8/internal/home/service.go

jamesmacwhite, 2022-05-29:
@d4f0 This is a known issue, it is dependant on certain network factors per setup, so it's not always the case. It is technically what you call a race condition.

The issue with the start values the AdGuardHome dev team have used is if you use AdGuardHome as the primary DNS resolver, you are leaving your router without DNS for much longer than you really should during startup.

START=95 is basically right at the end when /etc/rc.local is called, which isn't ideal.

Instead, if you have a specific network condition, you can create your own init.d script that initialises or prepares whatever network related condition isn't present before AdGuardHome starts with a lower START value than 21.

We could increase the START value a bit, but again it's down to specific network setups.

jamesmacwhite, 2022-05-29:
In your case it looks like AdGuardHome couldn't bind to 192.168.1.1. The start value of 21 is just after networking, but perhaps it hadn't fully completed, we could bump it a bit, but increasing it to 95 isn't the right move for the rationale above.

Could to try bumping it to 25/30 and see if that provides enough time for networking related conditions to be available?

d4f0, 2022-05-30:
With 25 it works fine.

Thanks.

closed 2022-05-30 by d4f0

jamesmacwhite, 2022-05-30:
@d4f0 Thanks for testing, I will make a pull request to increase the start value by default. It possibly won't fix all network race conditions but better than the current issue.