openwrt/packages #18587: openvpn build failed

ptpt52, 2022-05-20:
build on bcm2711 with kernel 5.15

```
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../include  -I../../include -I../../src/compat -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/usr/include -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/include/fortify -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/include   -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include  -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/wolfssl     -DPLUGIN_LIBDIR=\"/usr/lib/openvpn/plugins\"  -Wall -Wno-stringop-truncation -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6=openvpn-2.5.6 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -std=c99 -MT ssl.o -MD -MP -MF .deps/ssl.Tpo -c -o ssl.o ssl.c
mv -f .deps/ssl.Tpo .deps/ssl.Po
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../include  -I../../include -I../../src/compat -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/usr/include -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/include/fortify -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl/include   -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include  -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/wolfssl     -DPLUGIN_LIBDIR=\"/usr/lib/openvpn/plugins\"  -Wall -Wno-stringop-truncation -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6=openvpn-2.5.6 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -std=c99 -MT ssl_openssl.o -MD -MP -MF .deps/ssl_openssl.Tpo -c -o ssl_openssl.o ssl_openssl.c
ssl_openssl.c: In function 'openvpn_extkey_ec_finish':
ssl_openssl.c:1356:11: error: unknown type name 'EC_KEY_METHOD'
 1356 |     const EC_KEY_METHOD *ec_meth = EC_KEY_get_method(ec);
      |           ^~~~~~~~~~~~~
ssl_openssl.c:1356:36: warning: implicit declaration of function 'EC_KEY_get_method'; did you mean 'RSA_get_method'? [-Wimplicit-function-declaration]
 1356 |     const EC_KEY_METHOD *ec_meth = EC_KEY_get_method(ec);
      |                                    ^~~~~~~~~~~~~~~~~
      |                                    RSA_get_method
ssl_openssl.c:1356:36: warning: initialization of 'const int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
ssl_openssl.c:1357:5: warning: implicit declaration of function 'EC_KEY_METHOD_free' [-Wimplicit-function-declaration]
 1357 |     EC_KEY_METHOD_free((EC_KEY_METHOD *) ec_meth);
      |     ^~~~~~~~~~~~~~~~~~
ssl_openssl.c:1357:25: error: 'EC_KEY_METHOD' undeclared (first use in this function); did you mean 'EC_METHOD'?
 1357 |     EC_KEY_METHOD_free((EC_KEY_METHOD *) ec_meth);
      |                         ^~~~~~~~~~~~~
      |                         EC_METHOD
ssl_openssl.c:1357:25: note: each undeclared identifier is reported only once for each function it appears in
ssl_openssl.c:1357:40: error: expected expression before ')' token
 1357 |     EC_KEY_METHOD_free((EC_KEY_METHOD *) ec_meth);
      |                                        ^
ssl_openssl.c:1356:26: warning: unused variable 'ec_meth' [-Wunused-variable]
 1356 |     const EC_KEY_METHOD *ec_meth = EC_KEY_get_method(ec);
      |                          ^~~~~~~
ssl_openssl.c: In function 'tls_ctx_use_external_ec_key':
ssl_openssl.c:1421:5: error: unknown type name 'EC_KEY_METHOD'; did you mean 'EC_METHOD'?
 1421 |     EC_KEY_METHOD *ec_method;
      |     ^~~~~~~~~~~~~
      |     EC_METHOD
ssl_openssl.c:1425:17: warning: implicit declaration of function 'EC_KEY_METHOD_new' [-Wimplicit-function-declaration]
 1425 |     ec_method = EC_KEY_METHOD_new(EC_KEY_OpenSSL());
      |                 ^~~~~~~~~~~~~~~~~
ssl_openssl.c:1425:35: warning: implicit declaration of function 'EC_KEY_OpenSSL' [-Wimplicit-function-declaration]
 1425 |     ec_method = EC_KEY_METHOD_new(EC_KEY_OpenSSL());
      |                                   ^~~~~~~~~~~~~~
ssl_openssl.c:1425:15: warning: assignment to 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 1425 |     ec_method = EC_KEY_METHOD_new(EC_KEY_OpenSSL());
      |               ^
ssl_openssl.c:1432:5: warning: implicit declaration of function 'EC_KEY_METHOD_set_init' [-Wimplicit-function-declaration]
 1432 |     EC_KEY_METHOD_set_init(ec_method, NULL, openvpn_extkey_ec_finish, NULL, NULL, NULL, NULL);
      |     ^~~~~~~~~~~~~~~~~~~~~~
ssl_openssl.c:1433:5: warning: implicit declaration of function 'EC_KEY_METHOD_set_sign' [-Wimplicit-function-declaration]
 1433 |     EC_KEY_METHOD_set_sign(ec_method, ecdsa_sign, ecdsa_sign_setup, ecdsa_sign_sig);
      |     ^~~~~~~~~~~~~~~~~~~~~~
ssl_openssl.c:1441:10: warning: implicit declaration of function 'EC_KEY_set_method'; did you mean 'RSA_set_method'? [-Wimplicit-function-declaration]
 1441 |     if (!EC_KEY_set_method(ec, ec_method))
      |          ^~~~~~~~~~~~~~~~~
      |          RSA_set_method
ssl_openssl.c: In function 'tls_ctx_load_ca':
ssl_openssl.c:1605:55: warning: passing argument 1 of 'wolfSSL_sk_X509_NAME_new' from incompatible pointer type [-Wincompatible-pointer-types]
 1605 |                         cert_names = sk_X509_NAME_new(sk_x509_name_cmp);
      |                                                       ^~~~~~~~~~~~~~~~
      |                                                       |
      |                                                       int (*)(const X509_NAME * const*, const X509_NAME * const*) {aka int (*)(const WOLFSSL_X509_NAME * const*, const WOLFSSL_X509_NAME * const*)}
In file included from /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/wolfssl/openssl/ssl.h:35,
                 from ssl_openssl.h:32,
                 from ssl_backend.h:36,
                 from ssl_openssl.c:44:
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/wolfssl/ssl.h:4453:24: note: expected 'wolf_sk_compare_cb' {aka 'int (*)(const void *, const void *)'} but argument is of type 'int (*)(const X509_NAME * const*, const X509_NAME * const*)' {aka 'int (*)(const WOLFSSL_X509_NAME * const*, const WOLFSSL_X509_NAME * const*)'}
 4453 |     wolf_sk_compare_cb cb);
      |     ~~~~~~~~~~~~~~~~~~~^~
Makefile:694: recipe for target 'ssl_openssl.o' failed
make[6]: *** [ssl_openssl.o] Error 1
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6/src/openvpn'
Makefile:433: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6/src'
Makefile:611: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6'
Makefile:499: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6'
Makefile:149: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6/.built' failed
make[2]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/openvpn-wolfssl/openvpn-2.5.6/.built] Error 2
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/net/openvpn'
time: package/feeds/packages/openvpn/wolfssl/compile#30.78#5.32#35.87
    ERROR: package/feeds/packages/openvpn failed to build (build variant: wolfssl).
package/Makefile:114: recipe for target 'package/feeds/packages/openvpn/compile' failed
make[1]: *** [package/feeds/packages/openvpn/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:228: recipe for target 'package/feeds/packages/openvpn/compile' failed
make: *** [package/feeds/packages/openvpn/compile] Error 2

```

ptpt52, 2022-10-17:
@ynezz @AuthorReflex  ping 
any help on the build issue for openvpn-wolfssl ?

ynezz, 2022-10-17:
>  any help on the build issue for openvpn-wolfssl ?

It should work as it seems to build fine on [buildbots](https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/packages/openvpn-wolfssl_2.5.7-3_aarch64_cortex-a72.ipk), there is no [such build failure](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a72/) being registered either.

How could one reproduce that issue?

ptpt52, 2022-10-17:
enable all build options:
```
-# CONFIG_PACKAGE_openvpn-wolfssl is not set
+CONFIG_PACKAGE_openvpn-wolfssl=m
+CONFIG_OPENVPN_wolfssl=y
+CONFIG_OPENVPN_wolfssl_ENABLE_LZO=y
+CONFIG_OPENVPN_wolfssl_ENABLE_LZ4=y
+CONFIG_OPENVPN_wolfssl_ENABLE_X509_ALT_USERNAME=y
+CONFIG_OPENVPN_wolfssl_ENABLE_MANAGEMENT=y
+CONFIG_OPENVPN_wolfssl_ENABLE_FRAGMENT=y
+CONFIG_OPENVPN_wolfssl_ENABLE_MULTIHOME=y
+CONFIG_OPENVPN_wolfssl_ENABLE_PORT_SHARE=y
+CONFIG_OPENVPN_wolfssl_ENABLE_DEF_AUTH=y
+CONFIG_OPENVPN_wolfssl_ENABLE_PF=y
+CONFIG_OPENVPN_wolfssl_ENABLE_IPROUTE2=y
+CONFIG_OPENVPN_wolfssl_ENABLE_SMALL=y
```

ynezz, 2022-10-17:
> enable all build options:

Ok, can you try to narrow the problem domain and for example find out which option specifically triggers that issue?

ptpt52, 2022-10-17:
@ynezz 
looks like dropping `CONFIG_OPENVPN_wolfssl_ENABLE_MANAGEMENT` option build ok for me.

ptpt52, 2022-10-17:
> > looks like dropping CONFIG_OPENVPN_wolfssl_ENABLE_MANAGEMENT option build ok for me.
> 
> Could you please try to reproduce now the issue with OpenSSL variant? I would like to find out, if it's related to wolfSSL or if it's broken generally.

it is sure that openvpn-openssl build good, with all options enabled

ynezz, 2022-10-17:
> it is sure that openvpn-openssl build good, with all options enabled

Ok, do you've any rough idea when `openvpn-wolfssl` did approx. broke? Ideally bisect it to working/bad commits in OpenWrt master tree.