openwrt/packages #18589: openconnect: sed: bad option in substitution expression

mmokrejs, 2022-05-21:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-22.119.37126-a993714

Description: `sed` should use a different separator instead of a forward slash to delineate network number. It uses `#` for other arguments. Here is the problematic part:

```
"-e", "s/|IPV4ADDR|/192.168.2.0/24/g"
```

Here is remained of the commandline as reported by `strace -s 1024 -v -f /etc/init.d/ocserv start`
```
, "-e", "s/|NETMASK|/255.255.255.0/g", "-e", "s#|IPV6ADDR|##g", "-e", "s/|ENABLE_IPV6|/#/g", "/etc/ocserv/ocserv.conf.template"]
```

Due to the above the daemon does not start:

`/etc/init.d/ocserv start`

nmav, 2022-05-22:
Thanks, would you like to propose a fix with a PR?

mmokrejs, 2022-05-22:
No, sorry, I do not want t study structure of the project. I just have no time.

commit 246f678c (2024-02-19): ocserv: use better separator for sed

This prevents clashes with network addresses that
contain '/'.

Resolves: #18589
commit 8e9ad0e3 (2024-02-19): ocserv: use better separator for sed

This prevents clashes with network addresses that
contain '/'.

Resolves: #18589
closed 2023-09-10 by nmav