openwrt/packages #18615: Netdata Cloud on OpenWrt won’t work

vitorgrohs, 2022-05-25:
Hello friends,

I tried to enable the cloud connection on a Xiaomi 4A with openwrt 21.02 (All the routers I tested do not connect to the cloud anyway...)

I linked the topic that already existed in the openwrt forum with the netdata one:

https://community.netdata.cloud/t/netdata-cloud-on-openwrt-wont-work/1408/8

https://forum.openwrt.org/t/netdata-cloud-on-openwrt-won-t-work/127869

It looks like the package no longer allows cloud integration. Locally it works normally. Could this functionality be restored?

Like this:
https://www.youtube.com/watch?v=c8nViaoGZaU&t=269s

Thanks!!!

diizzyy, 2022-06-05:
https://github.com/openwrt/packages/blob/master/admin/netdata/Makefile#L68

vorwd, 2024-07-01:
> https://github.com/openwrt/packages/blob/master/admin/netdata/Makefile#L68

So, are you saying that if I remove the --disable-cloud arg, then cloud integration will work as expected?