openwrt/packages #18629: mwan3: adding new rules via luci does not work

TheB1gG, 2022-05-29:
Maintainer: @feckert 
Environment: (put here arch, model, OpenWrt version)
x86
OpenWrt SNAPSHOT r19679-810eac8c7f

Description:

I currently have 15 mwan3 rules in place.
I open https://openwrt/cgi-bin/luci/admin/network/mwan3/rule
add a new rule and click add
then I enter my rule definition for IPv4. I click save after I'm done with it
I see my rule at the bottm of the page
I click Save & Apply and my rule is gone, nothing was added to /etc/config/mwan3

If I add the rule manually to /etc/config/mwan3 and do a mwan3 restart it does work. I'm not sure if this is caused by luci or mwan3. But since everything else in luci works my guess is currently that there is something mwan3 is not happy about.

vspili, 2022-07-01:
I have noticed the same problem.
But if I use the "Save & Apply" button in the "Unsaved Changes" menu in the top bar, the changes are committed.

feckert, 2022-07-12:
The creation of rules works for me. I think this is related to the issue https://github.com/openwrt/luci/issues/5760. I also have a problem with that. I cannot create an interface via the LuCI. Since it is probably the same problem, I will close the ticket here. It also seems to have something to do with the LuCI and with mwan3 in the package feed, so we should maintain the issue there.

closed 2022-07-12 by feckert