openwrt/packages #18635: aria2: sending SIGSEGV to aria2c for invalid read access from 000XXXX

AdorableJiang, 2022-05-30:
Maintainer: @kaloz 
Environment: 
```
arch:MediaTek MT7621 ver:1 eco:3
model:HiWiFi HC5962
OpenWrt version: OpenWrt SNAPSHOT r19734-89562924b0 / LuCI Master git-22.137.71281-d6dbedd
```
Description:
P2P downloads (magnets&torrents) leads to Aria2 crash, while http links works fine.
```
Mon May 30 20:50:56 2022 kern.info kernel: [  997.832446] do_page_fault(): sending SIGSEGV to aria2c for invalid read access from 0008c79b
Mon May 30 20:50:56 2022 kern.info kernel: [  997.840925] epc = 77e10e08 in libc.so[77d92000+a9000]
Mon May 30 20:50:56 2022 kern.info kernel: [  997.845985] ra  = 77e12b5c in libc.so[77d92000+a9000]
```
[system.log](https://github.com/openwrt/packages/files/8798254/system.log)
[kernel.log](https://github.com/openwrt/packages/files/8798256/kernel.log)

neheb, 2022-06-04:
I bet this has to do with 64-bit time_t...

AdorableJiang, 2022-06-05:
> I bet this has to do with 64-bit time_t...

May I ask you to explain more?

neheb, 2022-06-05:
I've seen a lot of PRs fixing problems with bad format strings that don't give enough space for 64-bit time_t. I would compile aria2 with -Werror=format=2 and see what issues pop up.

AdorableJiang, 2022-06-18:
I downgrade openwrt to `OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a`. The aria2 works fine and stable.

AdorableJiang, 2022-07-29:
This is solved by clean (`rm /etc/config/aria2`) and re-configure aria2.

closed 2022-07-29 by AdorableJiang