openwrt/packages #18682: transmission-daemon: 22.03 - jail: failed to load dependencies

Juniorek81, 2022-06-04:
# Issue template (remove lines from top till here)

Maintainer: @dangowrt
Environment: (mvebu/cortexa9, Linksys WRT32X , OpenWrt 22.03-SNAPSHOT r19385-f765f2f114)

Description: after migrating from OpenWrt 21.02-SNAPSHOT r16577-75cbd8de00 to the latest 22.03 build, the transmision gets errors. I can't open the web interface and login to the panel. Moving back to 21.02 fixes this problem. 

```
Sat Jun  4 21:44:45 2022 user.err : jail: failed to load dependencies
Sat Jun  4 21:44:45 2022 daemon.err transmission-daemon[5322]: jail: failed to load dependencies ```

olo81, 2022-08-04:
I have the same

trippleflux, 2022-08-05:
Stop transmission and try to edit the following into your /etc/init.d/transmission : 
```
#	seccomp_path="/etc/seccomp/transmission-daemon.json"
#	if [ -f "$seccomp_path" ]; then
#		procd_set_param seccomp "$seccomp_path"
#	fi
...
#	procd_add_jail transmission log
#	procd_add_jail_mount "$config_file"
#	procd_add_jail_mount_rw "$config_dir/resume"
#	procd_add_jail_mount_rw "$config_dir/torrents"
#	procd_add_jail_mount_rw "$config_dir/blocklists"
#	procd_add_jail_mount_rw "$config_dir/stats.json"
#	procd_add_jail_mount_rw "$download_dir"
#	[ "$incomplete_dir_enabled" = "1" ] && procd_add_jail_mount_rw "$incomplete_dir"
#	[ "$watch_dir_enabled" = "1" ] && procd_add_jail_mount_rw "$watch_dir"
	web_home="${web_home:-/usr/share/transmission/web}"
	[ -d "$web_home" ]
	[ -f "$ca_bundle_file" ]
```

Probably seccomp for transmission has changed in the newer kernel, Tried to make it by my own but cannot do it properly with utrace #19067 .