openwrt/packages #18701: xray: too many open files error when large traffic

terryding77, 2022-06-07:
# xray: too many open files error when large traffic
Maintainer: @1715173329 
Environment:  openwrt-21.02 xray-core 1.5.4

Description:
After I use xray to do git clone or some other large traffic works, there is a error log  **too many open files** in xray and new request to xray would be lost. After google for answer, I find it can be resolved by adding open files settings in **/etc/init.d/xray** file
```
start_service() {
......
        procd_set_param limits nofile="100000 100000"
......
        procd_close_instance
}

```
without this setting line, the default nofile is 1024, I think it is not enough for some xray usage scenarios.
**100000** is a suitable number for my router, can we add a new config param in **/etc/config/xray** to setup this param
or try insensitive way like dockerd package？
https://github.com/openwrt/packages/blob/ea567e3c7cfb759c0273204e47251bdc01806f4b/utils/dockerd/files/dockerd.init#L214
https://github.com/openwrt/packages/blob/ea567e3c7cfb759c0273204e47251bdc01806f4b/utils/dockerd/files/dockerd.init#L225

1715173329, 2022-06-07:
Thanks for catching this. Could you please add `procd_set_param limits core="unlimited"` to init script and check if it helps?

terryding77, 2022-06-07:
> Thanks for catching this. Could you please add `procd_set_param limits core="unlimited"` to init script and check if it helps?

I try to only add `procd_set_param limits core="unlimited` into **/etc/init.d/xray** file, and do some job by xray.
And use `ls -1 /proc/{xray_pid}/fd/ | wc -l` to check open files, after the open files reach to 1024, my xray log file output errors
```
2022/06/07 09:51:53 [Warning] transport/internet/tcp: failed to accepted raw connections > accept tcp [::]:12345: accept4: too many open files
```
and network not response for a while until some open file closed.

1715173329, 2022-06-07:
Thanks for clarification. I will create a PR for it later.

PR #18705 (2022-06-08): xray-core: increase nofile limitation
Maintainer: me

Description:
This fixes "too many open files" error caused by max-file limitation
when xray processes large traffic.

Fixes: #18701