openwrt/packages #18736: curl: cannot be built with IDN support

anyuta1166, 2022-06-10:
Maintainer: @stangri
Environment: x86-64, OpenWrt master

Description:

If I enable **CONFIG_LIBCURL_LIBIDN2**, libidn2 builds fine, but libcurl is built without IDN support.

Relevant build log lines:

```
checking whether to build with libidn2... yes (/home/user/openwrt/staging_dir/target-x86_64_musl/usr)
checking for libidn2 options with pkg-config... found
configure: pkg-config: IDN_LIBS: "-lidn2 "
configure: pkg-config: IDN_LDFLAGS: "-L/home/user/openwrt/staging_dir/target-x86_64_musl/usr/lib "
configure: pkg-config: IDN_CPPFLAGS: "-I/home/user/openwrt/staging_dir/target-x86_64_musl/usr/include "
configure: pkg-config: IDN_DIR: "/home/user/openwrt/staging_dir/target-x86_64_musl/usr/lib"
checking if idn2_lookup_ul can be linked... no
checking idn2.h usability... yes
checking idn2.h presence... yes
checking for idn2.h... yes
configure: WARNING: Cannot find libraries for IDN support: IDN disabled
```

During the build process this check is failed: `checking if idn2_lookup_ul can be linked... no` and IDN support is disabled.

micmac1, 2022-06-10:
Check config.log in curl build dir. Does it say libiconv.so.2 and libintl.so.8 can't be found? In that case your libidn2 got built with BUILD_NLS=y. Probably curl Makefile should include nls.mk to fix this.

stangri, 2023-01-16:
@anyuta1166 could you please test with the current version of curl in OpenWrt, which Makefile now includes nls.mk?

stangri, 2023-03-10:
Closing due to lack of replies.

closed 2023-03-10 by stangri