openwrt/packages #18741: `irqbalance: ` segfaults for `aarch64_cortex-a53`

aidvu, 2022-06-11:
Maintainer: @hnyman 
Environment: aarch64_cortex-a53, Xiaomi AX6S, OpenWrt 21.*.* 

Description:

It seems the recent `irqbalance` bump to 1.9.0 made it segfault on above architecture. Judging by a recent patch https://github.com/Irqbalance/irqbalance/pull/210, it's probably broken for any AARCH64 CPU. 

I can confirm it was working previously on both Belkin RT3200 and Xiaomi AX6S.

aidvu, 2022-06-11:
Some debug info:

Belkin RT3200, irqbalance 1.8.0
```
root@Banshee:~# cat /tmp/sysinfo/model
Linksys E8450 (UBI)
root@Banshee:~# irqbalance --version
irqbalance version 1.8.0
root@Banshee:~# service irqbalance info
{
	"irqbalance": {
		"instances": {
			"irqbalance": {
				"running": true,
				"pid": 1871,
				"command": [
					"/usr/sbin/irqbalance",
					"-f",
					"-t",
					"10",
					""
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": 5
				}
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"irqbalance"
					],
					[
						"run_script",
						"/etc/init.d/irqbalance",
						"reload"
					]
				],
				1000
			]
		]
	}
}
```

Xiaomi AX6S
```
root@Wendigo:~# cat /tmp/sysinfo/model 
Xiaomi Redmi Router AX6S
root@Wendigo:~# irqbalance --version
irqbalance version 1.9.0
root@Wendigo:~# service irqbalance info
{
	"irqbalance": {
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"irqbalance"
					],
					[
						"run_script",
						"/etc/init.d/irqbalance",
						"reload"
					]
				],
				1000
			]
		]
	}
}
```

hnyman, 2022-06-12:
Thanks for spotting the bug and also the upstream fix.

closed 2022-06-12 by hnyman

aidvu, 2022-06-12:
Had a plan to make a patch and try to build it, since I haven't built OpenWRT myself yet, but that sure was quick. Thank you. :)

commit ab7a2310 (2022-06-12): irqbalance: upstream patch to fix segfault

Apply post-1.9.0 patch from upstream to fix a segfault.
savedptr can be NULL, so guard for that.

Fixes #18741
commit 4d09e063 (2022-06-12): irqbalance: upstream patch to fix segfault

Apply post-1.9.0 patch from upstream to fix a segfault.
savedptr can be NULL, so guard for that.

Fixes #18741

(cherry picked from commit ab7a23102e85aaab1e0ff2ba6874e7d6b09f45d5)
commit a47793ef (2022-06-12): irqbalance: upstream patch to fix segfault

Apply post-1.9.0 patch from upstream to fix a segfault.
savedptr can be NULL, so guard for that.

Fixes #18741

(cherry picked from commit ab7a23102e85aaab1e0ff2ba6874e7d6b09f45d5)