openwrt/packages #18748: prometheus-node-exporter-lua: daemon does not start after deferring itself to wait for interface to become ready

Hurricos, 2022-06-13:
Maintainer: @champtar 
Environment: Ctera C200 V1, on:
```txt
root@krakow:/etc/rc.d# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19621-d1b824650f'
DISTRIB_TARGET='kirkwood/generic'
DISTRIB_ARCH='arm_xscale'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19621-d1b824650f'
```

Description: daemon does not start after deferring itself to wait for interface to become ready.
It reports: `prometheus-node-exporter-lua: defering start until listen interface lan becomes ready`, but then `procd: Not starting instance prometheus-node-exporter-lua::data, command not set`, and when `br-lan` comes up, `prometheus-node-exporter-lua` is not running.

```txt
root@krakow:/etc/rc.d# logread | grep prometheus -C 1
Sun Jun 12 23:50:35 2022 daemon.notice netifd: Interface 'loopback' has link connectivity
Sun Jun 12 23:50:35 2022 daemon.info prometheus-node-exporter-lua: defering start until listen interface lan becomes ready
Sun Jun 12 23:50:35 2022 daemon.info procd: Not starting instance prometheus-node-exporter-lua::data, command not set
Sun Jun 12 23:50:37 2022 kern.info kernel: [   32.847774] mv643xx_eth_port mv643xx_eth_port.0 eth0: link up, 1000 Mb/s, full duplex, flow control enabled
root@krakow:/etc/rc.d# /etc/init.d/prometheus-node-exporter-lua status
running
root@krakow:/etc/rc.d# /etc/init.d/prometheus-node-exporter-lua info
{
	"prometheus-node-exporter-lua": {
		"instances": {
			"data": {
				"running": false,
				"term_timeout": 5,
				"exit_code": 0
			},
			"instance1": {
				"running": false,
				"command": [
					"/usr/sbin/uhttpd",
					"-f",
					"-c",
					"/dev/null",
					"-l",
					"/",
					"-L",
					"/usr/bin/prometheus-node-exporter-lua",
					"-k",
					"70"
				],
				"term_timeout": 5,
				"exit_code": 1
			}
		}
	}
}
root@krakow:/etc/rc.d# ps www | grep prometheus-node
 5618 root      1204 S    grep prometheus-node
```

I suspect the [reload trigger addition is failing](https://github.com/openwrt/packages/blob/master/utils/prometheus-node-exporter-lua/files/etc/init.d/prometheus-node-exporter-lua#L55).

champtar, 2022-06-13:
Can you show your config ? (prometheus and relevant part of network)
You don't have `-p` in the command, so for me `network_is_up` returns true but `network_get_ipaddr6` and `network_get_ipaddr` are empty

Hurricos, 2022-06-13:
```txt
# /etc/config/network
                                                                                                                                                       
config interface 'loopback'                                                                                                                            
        option device 'lo'                                                                                                                             
        option proto 'static'                                                                                                                          
        option ipaddr '127.0.0.1'                                                                                                                      
        option netmask '255.0.0.0'                                                                                                                     
                                                                                                                                                       
config globals 'globals'                                                                                                                               
        option ula_prefix 'fd54:7cfb:ebc8::/48'                                                                                                        
                                                                                                                                                       
config device                                                                                                                                          
        option name 'br-lan'                                                                                                                           
        option type 'bridge'                                                                                                                           
        list ports 'eth0'                                                                                                                              
                                                                                                                                                       
config interface 'lan'                                                                                                                                 
        option device 'br-lan'                                                                                                                         
        option proto 'dhcp'                                                                                                                            
                                                                                                                                                       
config interface 'lan6'                                                                                                                                
        option device 'br-lan'                                                                                                                         
        option proto 'dhcpv6'    
```

```txt
#  /etc/config/prometheus-node-exporter-lua 

config prometheus-node-exporter-lua 'main'
        option listen_interface 'lan'
        option listen_port '9100'
        option listen_ipv6 '0'
```

champtar, 2022-06-13:
Can you run
```
. /lib/functions/network.sh
network_get_ipaddr bind lan
echo $bind
```

Hurricos, 2022-06-13:
After quite a while of uptime:

```txt
***@***.***:~# . /lib/functions/network.sh
***@***.***:~# network_get_ipaddr bind lan
***@***.***:~# echo $bind
192.168.0.192
***@***.***:~#
```

I don't think the network was up at the original prometheus init time.

Martin

champtar, 2022-06-13:
can you try https://github.com/openwrt/packages/pull/18750 ?

Hurricos, 2022-06-16:
> can you try #18750 ?

Thank you, #18750 works!

```console
root@krakow:~# uptime
 16:09:01 up 3 min,  load average: 0.25, 0.37, 0.17
root@krakow:~# logread | grep prometheus -C 1
Thu Jun 16 16:05:47 2022 daemon.notice netifd: Interface 'loopback' has link connectivity
Thu Jun 16 16:05:47 2022 daemon.info prometheus-node-exporter-lua: defering start until listen interface lan becomes ready
Thu Jun 16 16:05:50 2022 kern.info kernel: [   32.975115] mv643xx_eth_port mv643xx_eth_port.0 eth0: link up, 1000 Mb/s, full duplex, flow control enabled
root@krakow:~# /etc/init.d/prometheus-node-exporter-lua status
running
root@krakow:~# /etc/init.d/prometheus-node-exporter-lua info
{
	"prometheus-node-exporter-lua": {
		"instances": {
			"instance1": {
				"running": true,
				"pid": 2589,
				"command": [
					"/usr/sbin/uhttpd",
					"-f",
					"-c",
					"/dev/null",
					"-l",
					"/",
					"-L",
					"/usr/bin/prometheus-node-exporter-lua",
					"-k",
					"70",
					"-p",
					"192.168.0.192:9100"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 3600,
					"timeout": 5,
					"retry": 5
				}
			}
		},
		"triggers": [
			[
				"config.change",
				[
					"if",
					[
						"eq",
						"package",
						"prometheus-node-exporter-lua"
					],
					[
						"run_script",
						"/etc/init.d/prometheus-node-exporter-lua",
						"reload"
					]
				],
				1000
			],
			[
				"interface.*",
				[
					"if",
					[
						"eq",
						"interface",
						"lan"
					],
					[
						"run_script",
						"/etc/init.d/prometheus-node-exporter-lua",
						"reload"
					]
				],
				1000
			]
		]
	}
}
```

champtar, 2022-06-17:
Fix in master an 22.03, thanks for testing

closed 2022-06-17 by champtar