openwrt/packages #18780: podman no runtime(s) available

lejeczek, 2022-06-19:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)
Linksys MR8300 (Dallas)
arm_cortex-a7_neon-vfpv4
21.02.3

Description:
-> $ podman version
Error: default OCI runtime "crun" not found: invalid argument

```
Format code blocks by wrapping them with pairs of ```
```

paper42, 2022-06-19:
see #15060

lejeczek, 2022-06-19:
What does that translate into? - image(s) for my Linksys model are behind what is "generally" available, still a year later?

paper42, 2022-06-19:
You will need to install either `crun` or `runc` to use podman.