openwrt/packages #18806: <f2fs-tools>:  fails to build for BTHome Hub 5 in OpenWrt 22.03 branch

ezplanet, 2022-06-23:
Please make sure that the issue subject starts with `<package-name>: `

Maintainer: PKG_MAINTAINER:=Felix Fietkau <nbd@nbd.name>i (find it by checking history of the package Makefile)
Environment: (Lantiq XRX200, BT Home Hub 5, OpenWrt 22.03 branch)

Description: <f2fs-tools>: fails to build
Way to reproduce: get openwrt 22.03 branch from git, delete .config.

make menuconfig
select lantiq, XRX200, BTHOMEHUB5

Then in Utilitities-->Filesystem select f2fs-tools

run make and the build will fail as follows:

rm -f /home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0/.built
touch /home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0=f2fs-tools-1.14.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/include -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include/fortify -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0=f2fs-tools-1.14.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/include -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include/fortify -I/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include " LDFLAGS="-L/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/lib -L/home/mauro/router/openwrt-22.03/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib -znow -zrelro " make  -C /home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0=f2fs-tools-1.14.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0'
make[4]: *** No targets specified and no makefile found.  Stop.
make[4]: Leaving directory '/home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0'
make[3]: *** [Makefile:149: /home/mauro/router/openwrt-22.03/build_dir/target-mips_24kc_musl/f2fs-tools-default/f2fs-tools-1.14.0/.built] Error 2
make[3]: Leaving directory '/home/mauro/router/openwrt-22.03/package/utils/f2fs-tools'
time: package/utils/f2fs-tools/default/compile#0.10#0.07#0.16
    ERROR: package/utils/f2fs-tools failed to build (build variant: default).
make[2]: *** [package/Makefile:116: package/utils/f2fs-tools/compile] Error 1
make[2]: Leaving directory '/home/mauro/router/openwrt-22.03'
make[1]: *** [package/Makefile:110: /home/mauro/router/openwrt-22.03/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mauro/router/openwrt-22.03'
make: *** [/home/mauro/router/openwrt-22.03/include/toplevel.mk:230: world] Error 2

ezplanet, 2022-06-24:
Since I use external USB flash drives for stats, logs, etc. this is kind of blocking me do adopt this new 22.03 release.

BKPepe, 2022-07-23:
The package f2fs-tools can be found in [the different repository](https://github.com/openwrt/openwrt/blob/0f301b0b1d7ca4b5fe290a72f0434525405f5a26/package/utils/f2fs-tools/Makefile) as your issue is opened in the wrong repository, I'm closing this.

closed 2022-07-23 by BKPepe