openwrt/packages #18807: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'

Drb-sz, 2022-06-24:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
root@comnect:~# logread |grep upnp
Fri Jun 24 10:38:37 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:44 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:38:45 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:05 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
Fri Jun 24 10:39:06 2022 daemon.err uhttpd[2649]: /etc/init.d/miniupnpd: .: line 15: can't open '/lib/lsb/init-functions'
root@comnect:~# 
```
Format code blocks by wrapping them with pairs of ```
```

Drb-sz, 2022-07-01:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:I can't find my device in the network neighbor, can someone tell me what's wrong?
Tips: I can't access tpohttp://192.168.1.1:5000/rootDesc.xml

```
Format code blocks by wrapping them with pairs of ```
root@comnect:~# /usr/sbin/miniupnpd -i br-lan  -a 192.168.1.1 -p 5000 -d
miniupnpd[21842]: system uptime is 4441 seconds
/usr/sbin/miniupnpd: unrecognized option: i
WPSInit() ok !
miniupnpd[21842]: WPS listening on port 8888
<?xml version="1.0" encoding="utf-8"?>
<root xmlns="urn:schemas-upnp-org:device-1-0">
 <specVersion>
  <major>1</major>
  <minor>0</minor>
 </specVersion>
 <device>
  <deviceType>urn:schemas-wifialliance-org:device:WFADevice:1</deviceType>
  <friendlyName>RalinkAPS</friendlyName>
  <manufacturer>Ralink Technology, Corp.</manufacturer>
  <manufacturerURL>http://www.ralinktech.com.tw</manufacturerURL>
  <modelDescription>Ralink AP WPS device</modelDescription>
  <modelName>Ralink Wireless Access Point</modelName>
  <modelNumber>RT3062</modelNumber>
  <modelURL>http://www.ralinktech.com.tw</modelURL>
  <serialNumber>87654321</serialNumber>
  <UDN>uuid:bc329e00-1dd8-11b2-8601-4c6e6e6a25ff</UDN>
  <serviceList>
   <service>
    <serviceType>urn:schemas-wifialliance-org:service:WFAWLANConfig:1</serviceType>
    <serviceId>urn:wifialliance-org:serviceId:WFAWLANConfig1</serviceId>
    <SCPDURL>WFAWLANConfigSCPD.xml</SCPDURL>
    <controlURL>control</controlURL>
    <eventSubURL>event</eventSubURL>
   </service>
  </serviceList>
 </device>
</root>
-------------
<?xml version="1.0"?>

<scpd xmlns="urn:schemas-upnp-org:service-1-0">
 <specVersion>
  <major>1</major>
  <minor>0</minor>
 </specVersion>
 <actionList>
  <action>
   <name>DelAPSettings</name>
   <argumentList>
    <argument>
     <name>NewAPSettings</name>
     <direction>in</direction>
     <relatedStateVariable>APSettings</relatedStateVariable>
    </argument>
   </argumentList>
  </action>
  <action>
   <name>DelSTASettings</name>
[... 849 lines cut ...]

miniupnpd[21842]: HTTP connection from 192.168.1.232:62540
miniupnpd[21842]: HTTP REQUEST : UNSUBSCRIBE /event (HTTP/1.1)
miniupnpd[21842]: ProcessHTTPUnSubscribe /event
miniupnpd[21842]: SID 'uuid:bc329e00-1dd8-11b2-8601-4c6e6e6af3e9'
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61662 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61663 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61663 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61662 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61662 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61663 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61662 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:61663 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57692 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57694 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57694 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57692 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57692 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57694 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57692 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:57694 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:63340 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:63343 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:63340 ST: urn:dial-multiscreen-org:service:dial:1
miniupnpd[21842]: SSDP M-SEARCH from 192.168.1.232:63343 ST: urn:dial-multiscreen-org:service:dial:1

Neustradamus, 2022-08-01:
@neheb: I think you are the good person for this problem.

hnyman, 2023-01-14:
Apparently related to old OEM fork. mtk-lede...

closed 2023-01-14 by hnyman