openwrt/packages #18812: strongswan: missing x448 with wolfssl

lowjoel, 2022-06-24:
# Issue template (remove lines from top till here)

Maintainer: @pprindeville @Thermi
Environment: OpenWrt 22.03-rc4, Linksys E8450 (aarch64 mt7622)

Description: Using WolfSSL for Strongswan does not enable the Curve448 option. Using OpenSSL does.

According to the [Strongswan docs](https://docs.strongswan.org/docs/5.9/config/IKEv2CipherSuites.html#_modern_elliptic_curve_groups), Curve448 should work with WolfSSL. WolfSSL in OpenWrt 22.03 is 5.9.0, which does support it (since wolfssl 4.8.0: https://github.com/wolfSSL/wolfssl/blob/master/ChangeLog.md#wolfssl-release-480-july-09-2021)

My suspicion is that we aren't passing in a feature enable flag to WolfSSL's configure script like we are Ed25519.

lowjoel, 2022-06-24:
I have a proposed change openwrt/openwrt#10116 that will enable Curve448 that I've opened in the main openwrt repository:

```
# swanctl --list-algs
encryption:
  AES_CBC[aes]
  AES_ECB[aes]
  3DES_CBC[des]
  DES_CBC[des]
  DES_ECB[des]
  RC2_CBC[rc2]
  AES_CTR[wolfssl]
  NULL[wolfssl]
integrity:
  HMAC_MD5_96[wolfssl]
  HMAC_MD5_128[wolfssl]
  HMAC_SHA1_96[wolfssl]
  HMAC_SHA1_128[wolfssl]
  HMAC_SHA1_160[wolfssl]
  HMAC_SHA2_256_128[wolfssl]
  HMAC_SHA2_256_256[wolfssl]
  HMAC_SHA2_384_192[wolfssl]
  HMAC_SHA2_384_384[wolfssl]
  HMAC_SHA2_512_256[wolfssl]
  HMAC_SHA2_512_512[wolfssl]
  AES_XCBC_96[xcbc]
aead:
  AES_GCM_16[wolfssl]
  AES_GCM_12[wolfssl]
  AES_CCM_16[wolfssl]
  AES_CCM_12[wolfssl]
  AES_CCM_8[wolfssl]
  CHACHA20_POLY1305[wolfssl]
hasher:
  HASH_SHA1[sha1]
  HASH_SHA2_224[sha2]
  HASH_SHA2_256[sha2]
  HASH_SHA2_384[sha2]
  HASH_SHA2_512[sha2]
  HASH_MD5[md5]
  HASH_SHA3_224[wolfssl]
  HASH_SHA3_256[wolfssl]
  HASH_SHA3_384[wolfssl]
  HASH_SHA3_512[wolfssl]
prf:
  PRF_KEYED_SHA1[sha1]
  PRF_HMAC_MD5[wolfssl]
  PRF_HMAC_SHA1[wolfssl]
  PRF_HMAC_SHA2_256[wolfssl]
  PRF_HMAC_SHA2_384[wolfssl]
  PRF_HMAC_SHA2_512[wolfssl]
  PRF_FIPS_SHA1_160[fips-prf]
  PRF_AES128_XCBC[xcbc]
xof:
drbg:
dh:
  ECP_256[wolfssl]
  ECP_384[wolfssl]
  ECP_521[wolfssl]
  ECP_224[wolfssl]
  MODP_3072[wolfssl]
  MODP_4096[wolfssl]
  MODP_2048[wolfssl]
  MODP_2048_224[wolfssl]
  MODP_2048_256[wolfssl]
  MODP_1536[wolfssl]
  MODP_1024[wolfssl]
  MODP_1024_160[wolfssl]
  MODP_768[wolfssl]
  MODP_CUSTOM[wolfssl]
  CURVE_25519[wolfssl]
  CURVE_448[wolfssl]
  MODP_6144[gmp]
  MODP_8192[gmp]
rng:
  RNG_WEAK[wolfssl]
  RNG_STRONG[random]
  RNG_TRUE[random]
nonce-gen:
  NONCE_GEN[nonce]
```

Thermi, 2022-06-24:
Yes, exactly. The strongswan test scripts in their repo build wolfssl for testing purposes and there's a flag to enable curve448. Check here: https://github.com/strongswan/strongswan/blob/f62b942d6a832aee134934dc4b44d01231e46186/scripts/test.sh#L49

That's an issue for the wolfssl maintainer.

@cotequeiroz Can you take this over, please?

closed 2022-07-03 by jow-

lowjoel, 2022-07-03:
For later reference: to clarify, Using Curve448 with Strongswan still requires compiling wolfssl with the `WOLFSSL_HAS_ECC448` set to `y` (defaults to `n`)