openwrt/packages #18816: python3-pip: Cannot install uvloop with pip

sgpublic, 2022-06-25:
Maintainer: @commodo
Environment: OpenWrt-21.02.3

Description:

I have python3-pip v22.1.2 and gcc v7.4.0-5 installed, While installing uvloop using `pip install uvloop`, I get the following error:

pip-log.txt
```
2022-06-25T15:32:11,482   config.status: error: in `/tmp/pip-install-6a7fy3cp/uvloop_84e37e587cf64d1f946f708775a37cca/build/libuv-x86_64':
2022-06-25T15:32:11,483   config.status: error: Something went wrong bootstrapping makefile fragments
2022-06-25T15:32:11,484       for automatic dependency tracking.  Try re-running configure with the
2022-06-25T15:32:11,485       '--disable-dependency-tracking' option to at least be able to build
2022-06-25T15:32:11,486       the package (albeit without support for automatic dependency tracking).
2022-06-25T15:32:11,487   See `config.log' for more details
```

config.log
```
conftest.c:58:10: fatal error: sys/ahafs_evProds.h: No such file or directory
 #include <sys/ahafs_evProds.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

How to install uvloop? Thank a lot!

commodo, 2022-06-25:
is this on the target?

we don't support building stuff (via pip) on the target;
even if you get gcc on the target, you need to install a various set of development libs;
openwrt is not yet in the are of behaving like a Debian/Fedora distro; it's more light-weight;

pip (on the target) works well if you install packages that only have Python source-code;
for other packages that need C/C++ compilation (or something more complicated) it's a bit better to have it packaged here: https://github.com/openwrt/packages/tree/master/lang/python

sgpublic, 2022-06-26:
I see, it means I need to compile the package directly at build time, thank you very much for your reply.

closed 2022-06-26 by sgpublic