openwrt/packages #18822: Can we have v2ray package

garysdevil, 2022-06-26:
[image]

Mitsuhaxy, 2022-06-29:
You can use xray instead of v2ray, the configuration is fully compatible
`opkg install xray-core`

1715173329, 2022-06-30:
your device is even not officially supported, so how can "we" do that?

garysdevil, 2022-07-03:
> your device is even not officially supported, so how can "we" do that?

I bought the device "GL_SFT1200", I can use "glinet" app to operate my router. Are you kidding me, it is not official?

1715173329, 2022-07-18:
> Are you kidding me, it is not official?

That depends on how you define "official". For OpenWrt, it's unofficial and unsupported.

BKPepe, 2022-07-23:
You might want to reach [the OpenWrt community on the forum](https://forum.openwrt.org/) to request it. There is at least a GUI: https://github.com/openwrt/packages/blob/6544b67f1f2cb8fe806c67ef7e1a69a3ed138546/net/v2raya/Makefile

closed 2022-07-23 by BKPepe