openwrt/packages #18841: <mwan3>: Interface detected as up while actually down

PsychoGame, 2022-06-30:
Maintainer: @feckert 
Environment: 
- Architecture: Qualcomm Atheros QCA956X ver 1 rev 0 (ath79)
- Model: TP-Link Archer C7 v5
- OpenWRT version: OpenWrt 22.03.0-rc4 r19426-2b1941e47d / LuCI openwrt-22.03 branch git-22.167.28394-8a4486a

Description:
On OpenWRT 22.03.0-rc4, I noticed some strange behaviour with the package mwan3, which I didn't encounter before in previous versions of OpenWRT.
We use the following set-up:
- There are 3 mobile broadband routers, which got addressed to respectively WAN (metric 10), WANB (metric 20) and WANC (metric 30).
- 2 of these mobile broadband routers are set-up as load-balanced (WAN and WANB)
- 1 mobile broadband router is fail-over (WANC)
So basically the policy is WANAB -> WANC
The OpenWRT router gets rebooted every night at 03:30 UK time.
The set-up is situated on-board of a ship which sails between 2 ports (resp. Dublin & Liverpool), that's the reason for the fail-over set-up, because 2 of the broadband routers use Irish providers & 1 is using UK (provider).
Now during sailing sometimes the switching over seems to work flawlessly, but sometimes the mwan tracker seems to fail in tracking if a connection is up/down, so the whole fail-over set-up fails resulting in network outage.
In my case it's usually wanb which gets locked-up in "online" state.
Disabling the wanb interface works, but as soon as it gets restarted, it will mark the interface as "online" again and so the route fail-over to wanc simply doesn't get set.
I made sure that wanb is actually down, by pinging the tracking IP's over wanb, which all result in 100% packet loss.
The problem is solvable by restarting the mwan3 service and then the tracking works properly again, but this isn't really a solution.
How can I help to try and figure out what might be the issue? 
What logs can I provide or which configuration files are important to post here?

feckert, 2022-07-05:
It is difficult to say what the problem is.
Maybe we start like this:
What tracking IPs are you using?
Is the mobile modem set up in a special way so that it autoconnect or does have roaming enabled?

PsychoGame, 2022-07-05:
Thank you for your response Florian,<br>
The mobile modems haven't been set-up in a specific way to be honest.<br>
These are just 3 Huawei Mobile Broadband modemrouters, which are always powered, and continuously scanning for a mobile broadband network, until it connects to a possible celltower.<br>
The WAN/WANB and WANC interface on the OpenWRT router (which contains the Mwan3 package), is connected to these 3 modemrouters, and have been assigned a static IP-Address which have a infinite leasetime.<br>
The tracking IP's which I use on all interfaces are:<br>
- 8.8.8.8<br>
- 8.8.4.4<br>
- 208.67.220.220<br<
- 208.67.222.222<br>
For your convenience I've uploaded the mwan3, and my network config files:<br>
[/etc/config/mwan3](https://termbin.com/3swt)
[/etc/config/network](https://termbin.com/rsd5)

feckert, 2022-07-12:
This means that the modems are connected via Ethernet and there is no cellular modem in the router.
From my point of view the `network` and `mwan3` configuration looks valid.

Since you are using the 22.03-rc4, I suspect that this is the problem.
There was a switch from iptables to nftables, can you verify whether the rules are created via iptables-legacy or iptables-nft?

feckert, 2023-05-03:
No reply from the ticket creator

closed 2023-05-03 by feckert