openwrt/packages #18850: shadowsocks-libev udp tproxy can't work

time-river, 2022-07-02:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

## Description:

The udp relay can't work,  **`ss-redir` can't receive any udp packets.** But the tproxy rule works.

## Environment

### Hardware:
NanoPi R5S

### Software:
kernel - 5.10.116-1-46ad65b4062101b6af7b599cdea01149
openwrt 22.03 rc3
overture 1.7-1
shadowsocks-libev 3.3.5-5

### Configuration
shadowsocks config:
```
config ss_local 'socks_tcp'
        option local_port '1080'
        option timeout '30'
        option fast_open '1'
        option no_delay '1'
        option reuse_port '1'
        option local_address '192.168.86.1'
        option mode 'tcp_only'
        option server 'tcp'

config ss_local 'socks_udp'
        option local_port '1080'
        option timeout '30'
        option fast_open '1'
        option no_delay '1'
        option reuse_port '1'
        option local_address '192.168.86.1'
        option mode 'udp_only'
        option server 'udp'
        option verbose '1'

config ss_redir 'redir_tcp'
        option local_port '1100'
        option timeout '60'
        option fast_open '1'
        option reuse_port '1'
        option no_delay '1'
        option local_address '::'
        option mode 'tcp_only'
        option server 'tcp'
        option disabled '0'

config ss_redir 'redir_udp_v6'
        option local_port '1100'
        option timeout '60'
        option fast_open '1'
        option reuse_port '1'
        option no_delay '1'
        option local_address '::'
        option mode 'udp_only'
        option verbose '1'
        option server 'udp'
        option disabled '0'

config ss_redir 'redir_udp_v4'
        option local_port '1100'
        option timeout '60'
        option fast_open '1'
        option reuse_port '1'
        option no_delay '1'
        option local_address '0.0.0.0'
        option mode 'udp_only'
        option verbose '1'
        option server 'udp'
```

Both ipv4 and ipv6 rules are configured, nftables shows:

```
table inet fw4 {
        chain ss_rules_pre_udp {
                type filter hook prerouting priority mangle; policy accept;
                meta l4proto udp goto ss_rules_pre_src_udp
        }

        chain ss_rules_pre_src_udp {
                ip daddr @ss_rules_dst_bypass_ accept
                ip6 daddr @ss_rules6_dst_bypass_ accept
                goto ss_rules_src_udp
        }

        chain ss_rules_src_udp {
                ip saddr @ss_rules_src_bypass accept
                ip saddr @ss_rules_src_forward goto ss_rules_forward_udp
                ip saddr @ss_rules_src_checkdst goto ss_rules_dst_udp
                ip6 saddr @ss_rules6_src_bypass accept
                ip6 saddr @ss_rules6_src_forward goto ss_rules_forward_udp
                ip6 saddr @ss_rules6_src_checkdst goto ss_rules_dst_udp
                goto ss_rules_dst_udp
        }

        chain ss_rules_dst_udp {
                ip daddr @ss_rules_dst_bypass accept
                ip daddr @ss_rules_dst_forward goto ss_rules_forward_udp
                ip6 daddr @ss_rules6_dst_bypass accept
                ip6 daddr @ss_rules6_dst_forward goto ss_rules_forward_udp
                goto ss_rules_forward_udp
        }

        chain ss_rules_forward_udp {
                udp dport != { 53, 67, 137, 138, 546, 547, 1080, 3702, 5353, 5355, 8125, 10000 } meta mark set 0x00000001 tproxy to :1100
        }
}
```

route rules:

```
root@FriendlyWrt ~# ip -4 rule list table 100
32765:  from all fwmark 0x1 lookup 100
root@FriendlyWrt ~# ip -4 route list table 100
local default dev lo scope host

root@FriendlyWrt ~# ip -6 route list table 100
local default dev lo metric 1024 pref medium
root@FriendlyWrt ~# ip -6 rule list table 100
32765:  from all fwmark 0x1 lookup 100
```

Then check the tproxy:

```
## in NanoPi R5S
# echo 'file net/ipv4/netfilter/nf_tproxy_ipv4.c +p' > /sys/kernel/debug/dynamic_debug/control
# echo 'file net/ipv6/netfilter/nf_tproxy_ipv6.c +p' > /sys/kernel/debug/dynamic_debug/control

## in client
#  nc -4 -u google.com 443
#  nc -6 -u google.com 443

## in NanoPi R5S
# dmesg | tail
[ 4246.309766] tproxy socket lookup: proto 17  [ipv6 address]:32917 -> 2607:f8b0:4007:0803:0000:0000:0000:200e:443, lookup type: 1, sock 0000000000000000
[ 4246.309955] tproxy socket lookup: proto 17 [ipv6 address]:32917 -> 240e:0390:0cea:f7b1:0000:0000:0000:0001:1100, lookup type: 0, sock 000000001ea2de2f
[ 4250.597554] tproxy socket lookup: proto 17 c0a85628:49569 -> 8efa442e:443, lookup type: 1, sock 0000000000000000
[ 4250.597741] tproxy socket lookup: proto 17 c0a85628:49569 -> c0a85601:1100, lookup type: 0, sock 000000002f59ac4e
```

The tproxy works truely, but `ss-redir` can't receive anything. Any idea can be provided?

@yousong #17937

time-river, 2022-07-02:
I find the similar issue [vernesong/OpenClash#374 (comment)](https://github.com/vernesong/OpenClash/issues/374#issuecomment-623879365), then I `rmmod br_netfilter`, it works.

It seems the kernel problem, I think it's bug, but it looks like it isn't eazy to fix ;-( .
At least, give warning for that.

And I find another bug, here is fix #18852

yousong, 2022-07-03:
It's caused by skb_orphan() in ip_rcv().  That will undo tproxy action triggered previously in br_nf_pre_routing().

 - With br_netfilter and `net.bridge.bridge-nf-call-iptables=1`, tproxy action will be triggered first by `br_nf_pre_routing()`
 - Then skb_orphan() in ip_rcv() (or ip_rcv_core()) will undo that
 - And br_netfilter ip_sabotage_in() will prevent those hookfn from being called again

http://patchwork.ozlabs.org/project/netfilter-devel/patch/1518715545-2188-1-git-send-email-gregory.vanderschueren@tessares.net/

Turning off `net.bridge.bridge-nf-call-iptables=1` should also work as with `rmmod br_netfilter`.  Otherwise it seems that kernel code has to be changed if we want both br_netfilter and tproxy.

closed 2022-08-11 by yousong

zhanhb, 2023-07-03:
Is there any way to let ss-redir listen both ipv4 and ipv6?

When `local_address` of `ss-redir`  is set to `0.0.0.0`, the udp relay to ipv4 succeed, while to ipv6 failed.
When `local_address` of `ss-redir`  is set to `::`, the udp relay to ipv6 succeed, while to ipv4 failed.

Here's the test commands on my PC:
```bash
# communications error to 8.8.8.8#53: timed out
dig -4 google.com @dns.google

# communications error to 2001:4860:4860::8844#53: timed out
dig -6 google.com @dns.google
```

yousong, 2023-07-03:
Hi @zhanhb It's worth checking whether the netfilter rules only handling one af family.

I am going to lock this old conversation.  Please open a new issue for new topic.  Thank you.