openwrt/packages #18851: lighttpd authentication modules depends on libnettle

stokito, 2022-07-02:
Maintainer: @flyn-org
Environment: OpenWrt 21.02.1   lighttpd/1.4.64 (ssl)

Description:

OpenWrt now uses wolfssl by default. But when I installed the lighttpd with mod-authn_file I found that now I have [libnettle8](https://openwrt.org/packages/pkgdata/libnettle8) installed:

```
# opkg whatdepends libnettle8
Root set:
  libnettle8
What depends on root set
        lighttpd-mod-authn_file 1.4.64-1        depends on libnettle8
        lighttpd 1.4.64-1       depends on libnettle8
        lighttpd-mod-webdav 1.4.64-1    depends on lighttpd
        lighttpd-mod-auth 1.4.64-1      depends on lighttpd
        lighttpd-mod-rewrite 1.4.64-1   depends on lighttpd
```

As far I understood the libnettle8 is just another crypto library.
From the [mod_authn_file source](https://github.com/lighttpd/lighttpd1.4/blob/master/src/mod_authn_file.c) it looks like it just includes [sys-crypto.h](https://github.com/lighttpd/lighttpd1.4/blob/master/src/sys-crypto.h) which may be configured any crypto lib. I.e. it can use the already installed wolfssl.

In the [lighttpd makefile](https://github.com/openwrt/packages/blob/openwrt-21.02/net/lighttpd/Makefile#L176) I see that the libnettle8 is the only needed dependency.

Is it possible to avoid installing the libnettle8?

I also have Turris Omnia which is a flavor of OpenWrt and there they are compiled the Lighttpd only with OpenSSL but the lighttpd-mod-authn_file doesn't have a dependency even to it:
```
# opkg depends lighttpd-mod-authn_file
lighttpd-mod-authn_file depends on:
        libc
        lighttpd
```

neheb, 2022-07-02:
ping @gstrauss

gstrauss, 2022-07-02:
tl;dr: The OpenWRT lighttpd base package **could be modified** to depend on WolfSSL instead of on Nettle, but that would not be my preference or recommendation for OpenWRT.

lighttpd is modular and provides multiple options for crypto and TLS.
However, there is no one-size-fits all to packaging.  More details:

Nettle provides some cryptographic functions, but (intentionally) does not provide TLS on top of the cryptographic functions.  As the lighttpd base can be *built* without TLS support, the default for lighttpd builds with libnettle so that mod_auth is able to use stronger crypto (without the overhead of a TLS library) when lighttpd is *built* without TLS support.

Nettle is about 175k .text on my 64-bit Linux, which is not tiny, but also not very large.  Compare that to libwolfssl.so which is about 700k .text on my 64-bit Linux (built with --enable-lighty --enable-opensslextra).   For mbedTLS: all three libmbed*.so add up to about 400k .text on my 64-bit Linux, and libmbedcrypto.so is about 265k of that.  TLS libraries OpenSSL and GnuTLS are much larger, but also generally much faster.

lighttpd provides multiple TLS modules, and the user can pick one: mod_openssl, mod_gnutls, mod_mbedtls, mod_nss, mod_wolfssl.  These are **standalone modules** and the **base lighttpd package** does not require them.  Of course, you need to pick one if you want TLS.  This makes Nettle a good choice for minimizing the dependencies of the lighttpd base, and allowing users to install the TLS module that they desire.

It is possible to build lighttpd *without* Nettle, and then if lighttpd is built *with* one of the TLS libraries, then lighttpd base and lighttpd mod_auth will use the cryptographic functions from that TLS library.  **Note: doing so changes the dependency of the *base* lighttpd package.**  lighttpd supports this.  However, for simplicity, I chose the lighttpd default to use Nettle.

If you are looking to truly minimize the memory use of lighttpd shared libs, you probably want to compile and build lighttpd yourself, and might also wish to make a static build of lighttpd (https://wiki.lighttpd.net/InstallFromSource).
Remember shared libs are file-backed pages in memory, so this is not necessary resident in memory, which is why lighttpd base dependency on Nettle is a reasonable choice and is not a big deal, IMHO.

My personal opinion: I was disappointed that OpenWRT switched to WolfSSL.  In my personal opinion (intentionally repeating that), the mbedTLS code is much higher quality, though mbedTLS is admittedly behind the competition in not currently supporting TLS 1.3 and OCSP stapling.  On the other hand, the WolfSSL code has -- trying to be polite -- sufficient technical debt and multiple really, really, really invasive and concerning code smells and inconsistencies, all of which (personally) bother me immensely and wasted many, many hours of my time in testing lighttpd with WolfSSL.  Well over a year ago, I shared these personal opinions privately in an email exchange with the CEO and Founder of WolfSSL, so I do not think that I am not saying anything here that has not also been communicated to the WolfSSL team.  I also wish to be clear that the WolfSSL team has been very polite, very professional, and sympathetic to the concerns I shared.

Aside: there is a bug in lighttpd 1.4.60 - 1.4.65 base64 code which makes lighttpd mod_wolfssl unusable.  This will be fixed in lighttpd 1.4.66 and I may submit a patch to OpenWRT sooner.  As I happened across this bug myself and added test cases to avoid future regressions, it is unlikely that anyone is currently using OpenWRT 21.02 lighttpd with mod_wolfssl.
https://git.lighttpd.net/lighttpd/lighttpd1.4/commit/5fef5e22089e26adc2d72f27b5b7fb9128c50eff

FYI: in the future, Turris OS (future TOS 6.1.0?) plans to use the OpenWRT lighttpd packages rather than maintaining their own:
https://gitlab.nic.cz/turris/os/packages/-/issues/474

gstrauss, 2022-07-02:
@stokito Nettle is a (small) crypto library, but is not a (larger) TLS library (which depends on crypto libs).  Do you have specific concerns about lighttpd using Nettle beyond trying to minimize the *number* of package dependencies?  Is there a disk space issue?  A memory use issue?  Other issues that concerns you that inspired you to file this github issue?

neheb, 2022-07-02:
@gstrauss OT but the reason OpenWrt switched to wolfssl is b cause hostapd has no support for mbedTLS and nobody has contributed support. Would be great to switch back.

stokito, 2022-07-03:
@neheb you probably talking about [wpad-basic-openssl](https://openwrt.org/packages/pkgdata/wpad-basic-wolfssl) and [wpad-basic-openssl](https://openwrt.org/packages/pkgdata/wpad-basic-wolfssl). There is no mbedtls support 
 https://github.com/openwrt/openwrt/blob/openwrt-21.02/package/network/services/hostapd/Makefile

neheb, 2022-07-03:
oh. seems someone's working on this: https://www.spinics.net/lists/hostap/msg09799.html

stokito, 2022-07-03:
out of curiosity I checked it's sources:
  
  crypto_gnutls.c
  crypto_internal.c
  crypto_libtomcrypt.c
  crypto_linux.c
  crypto_nettle.c
  crypto_openssl.c
  crypto_wolfssl.c

Looks like the hostap may use the nettle and the libtomcrypt which internally is used by Dropbear (but is statically compiled).
Anyway, hope the mbedtls will back.

gstrauss, 2022-07-03:
@stokito wrote:
> Is it possible to avoid installing the libnettle8?

Possible?  Yes.  Recommended for OpenWRT?  No, not by the lighttpd developers (me; @gstrauss)

@stokito  Did I answer your question?  Is there any further action item in this github issue?

@gstrauss wrote:
> @stokito Nettle is a (small) crypto library, but is not a (larger) TLS library (which depends on crypto libs). Do you have specific concerns about lighttpd using Nettle beyond trying to minimize the number of package dependencies? Is there a disk space issue? A memory use issue? Other issues which concern you and inspired you to file this github issue?

stokito, 2022-07-03:
Yes, thank you @gstrauss for the explanation, you answered literally all my questions that I didn't even asked yet :) Looks like I even faced the bug that you mentioned when tried to use mod_wolfssl with Let'sEncrypt :+1:

I'll close the issue.
Just want to add few things to the discussion to anyone else who will read and interested on the topic.
As far I know the Lighttp in OpenWrt used by
1. Gl.Inet and Turris routers and they both use mod OpenSSL.
2. Anyone else who serves a website for the internet and they need full TLS.

The case when you need mod_authn_file but not mod_ssl looks very rare for me.

Also in my case for WebDAV I just use plain passwords instead of hashed. So it looks like I even don't need the nettle. Similarly the mod_dav has a dependency to sqlite for some locks that is not needed for a basic functionality.

On my MIPS router The unpacked size is libnettle.so.8.0 243,148 + libhogweed.so.6.0 263,264 = 494Kb.

closed 2022-07-03 by stokito

gstrauss, 2022-07-19:
@stokito FYI: I have reconsidered my position and am planning to add a build option to lighttpd to permit forcing a crypto lib option for use by the base lighttpd objects.  If wolfssl is not removable from the system without breaking lots of things, then it is okay for lighttpd to depend on wolfssl, too, for crypto algorithms like SHA-256.  (In the future that might be mbedtls instead of wolfssl.)  lighttpd will still be able to offer the option of choosing a different lighttpd TLS module for TLS.  (Also, lighttpd use of Nettle does not require libhogweed if you're counting bytes, though libnettle.so and libhogweed.so do come in the same package.  One more thing: there is the option to build lighttpd without depending on Nettle or on TLS libraries, but that would need to be a separate package like a lighttpd-minimal, and minimal might not include lighttpd mod_webdav.)

---

> @gstrauss OT but the reason OpenWrt switched to wolfssl is b cause hostapd has no support for mbedTLS and nobody has contributed support. Would be great to switch back.

@neheb if you are serious about this and committed, then please create a new issue and subscribe me.

The cut-n-paste from stack overflow in the discussion in https://www.spinics.net/lists/hostap/msg09799.html did not sit well with me, so I took a crack at porting hostap to use mbedtls from scratch, and leveraging what I wrote for lighttpd mod_mbedtls.  I just spent a solid two weeks and have the basics working.  However, hostap and wpa_supplicant are very advanced and full of features, implementing many standards, so aiming for feature-completeness with mbedtls is not my goal.  I could use some guidance on what the requirements are for openwrt use of hostap.  For example, I don't have SAE (WPA3-Personal) working yet.

I pushed my work in progress development branch to https://github.com/gstrauss/hostap/tree/mbedtls

hostap has plenty of application tests under tests/hwsim, and those tests are great to have.  Unfortunately, hostap does not have much in the way of unit tests beyond smoke testing basic crypto algorithms and my branch passes those basic crypto tests 100%.  While surveying the code and prototyping to make sure that mbedtls has interfaces to implement what is needed by hostap, I have filled in hostap src/crypto/tls.h interfaces with mbedtls calls, but I am sure there are bugs in my fresh code and there is quite a bit of code to test up the stack to get SAE working.

neheb, 2022-07-19:
Done.

I personally use OpenSSL as size is not a concern of mine, but mbedtls would be great given that most have that concern.

stokito, 2022-07-23:
Follow up https://github.com/openwrt/openwrt/issues/10303

gstrauss, 2022-08-08:
FYI: #19131 for lighttpd 1.4.66

lighttpd 1.4.66 has been released with a fix for base64 handling which affected lighttpd mod_wolfssl.

lighttpd 1.4.66 also includes a patch to allow `-DFORCE_WOLFSSL_CRYPTO` which you can use for testing.  There has been no integration into the lighttpd builds, or modifications made to the OpenWRT lighttpd Makefile to set that define.

gstrauss, 2022-10-06:
Follow-up in https://github.com/openwrt/packages/pull/19537