openwrt/packages #18855: nut: package outdated, new upstream v2.8.0 includes many fixes & new drivers.

biergaizi, 2022-07-03:
Maintainer: @neheb
Environment: OpenWrt Development Snapshot

Description:

Dear maintainers,

Network UPS Tools (NUT) is a package for Uninterruptible Power Supply management. Recently, the upstream has released a new  stable version v2.8.0 in Apirl 26, 2022. Previously, due to some project management difficulties, NUT didn't release any stable version in the past 6 years since 2016, in spite of the fact that there was a steady stream of new fixes and drivers that got merged into the upstream. Thus, this new v2.8.0 included a huge number of fixes, changes, and new device support, and solved many outstanding problems in the old version.

Main changes include but not limited to: several fixes and new  drivers, including new i2c and modbus drivers. libusb-1.0 support. Python 3. Revised startup scripts. New systemd service files. Updated protocol, ABI and API. Please read the full resease announcement for details:

https://github.com/networkupstools/nut/releases/tag/v2.8.0

Without updating NUT on OpenWrt, users of several new UPS models cannot manage them on OpenWrt via NUT. At the very least, a new unstable package for v2.8.0 is needed.

It's worth noting that the addition of Modbus-based drivers requires an external library libmodbus and the build option `--with-modbus` switched on when building NUT, otherwise these drivers would be unavailable.

neheb, 2022-07-10:
I don't maintain this. Neither is anyone else unfortunately.

EricLuehrsen, 2022-09-15:
resolved #19340

closed 2022-09-15 by neheb