openwrt/packages #18869: Wi-Fi Protected Setup upnp

Drb-sz, 2022-07-05:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
RalinkWPSMiniUPnPDaemonHowTo.doc
Does anyone have this document? I want to know what his role is
```
Format code blocks by wrapping them with pairs of ```
```

brada4, 2022-07-08:
mediatek holds ralink documents since acquiring it 10 years ago. You must look for Mediatek AP SDK.

Drb-sz, 2022-07-08:
ok thanks!

jow-, 2022-07-08:
Completely unrelated to this repository.

closed 2022-07-08 by jow-