openwrt/packages #18876: python-cryptography failed to build on master

trippleflux, 2022-07-06:
Failed to build with the following log : 
```
make[3]: Entering directory '/home/username/works/openwrt/feeds/packages/lang/python/python-cryptography'
rm -f /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/.built
touch /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/.built_check
SHELL= flock /home/username/works/openwrt/tmp/.pip.flock -c ' ARCH="x86_64" CC="gcc" CCSHARED="gcc -DPIC -fPIC" CXX="g++" LD="gcc" LDSHARED="gcc -shared" CFLAGS="-O2 -I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include" CPPFLAGS="-I/home/username/works/openwrt/staging_dir/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/include -I/home/username/works/openwrt/staging_dir/hostpkg/include/python3.10" LDFLAGS="-L/home/username/works/openwrt/staging_dir/host/lib -L/home/username/works/openwrt/staging_dir/hostpkg/lib -L/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib -lpython3.10 -Wl,-rpath,/home/username/works/openwrt/staging_dir/hostpkg/lib" PYTHONPATH="/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10:/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=0 PYTHONOPTIMIZE="" /home/username/works/openwrt/staging_dir/hostpkg/bin/pip3.10 --cache-dir "/home/username/works/openwrt/dl/pip-cache" --disable-pip-version-check install --no-binary :all: --require-hashes  --requirement ../host-pip-requirements/cffi.txt  '
Requirement already satisfied: cffi==1.15.0 in /home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages (from -r ../host-pip-requirements/cffi.txt (line 1)) (1.15.0)
Requirement already satisfied: pycparser==2.21 in /home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages (from -r ../host-pip-requirements/cffi.txt (line 2)) (2.21)
install -d -m0755 /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/ipkg-install//usr/lib/python3.10/site-packages
cd "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/" && CC="x86_64-openwrt-linux-gnu-gcc" CCSHARED="x86_64-openwrt-linux-gnu-gcc -DPIC -fpic" CXX="x86_64-openwrt-linux-gnu-g++" LD="x86_64-openwrt-linux-gnu-gcc" LDSHARED="x86_64-openwrt-linux-gnu-gcc -shared" CFLAGS="-mtune=skylake -march=skylake -O3 -pipe -fno-caller-saves -fno-plt -mtune=skylake -march=skylake -O3 -pipe -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8=cryptography-3.4.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/include/python3.10" LDFLAGS="-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/lib -znow -zrelro -lpython3.10" _PYTHON_HOST_PLATFORM="linux-x86_64" __PYVENV_LAUNCHER__="/usr/bin/python3.10" PYTHONPATH="/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/python3.10:/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom//usr/lib/python3.10/site-packages:/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/ipkg-install//usr/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=1 PYTHONOPTIMIZE="" _python_sysroot="/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom" _python_prefix="/usr" _python_exec_prefix="/usr"     /home/username/works/openwrt/staging_dir/hostpkg/bin/python3.10  setup.py   install --prefix="/usr" --root="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/ipkg-install" --single-version-externally-managed
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.

    =============================DEBUG ASSISTANCE=============================
    If you are seeing a compilation error please try the following steps to
    successfully install cryptography:
    1) Upgrade to the latest pip and try again. This will fix errors for most
       users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
    2) Read https://cryptography.io/en/latest/installation.html for specific
       instructions for your platform.
    3) Check our frequently asked questions for more information:
       https://cryptography.io/en/latest/faq.html
    4) Ensure you have a recent Rust toolchain installed:
       https://cryptography.io/en/latest/installation.html#rust
    5) If you are experiencing issues with Rust for *this release only* you may
       set the environment variable `CRYPTOGRAPHY_DONT_BUILD_RUST=1`.
    =============================DEBUG ASSISTANCE=============================

Traceback (most recent call last):
  File "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/setup.py", line 71, in <module>
    setup(
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/python3.10/distutils/core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/dist.py", line 453, in __init__
    _Distribution.__init__(
  File "/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/python3.10/distutils/dist.py", line 292, in __init__
    self.finalize_options()
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/dist.py", line 831, in finalize_options
    ep(self)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/dist.py", line 852, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/.eggs/setuptools_rust-1.4.1-py3.10.egg/setuptools_rust/__init__.py", line 1, in <module>
    from .build import build_rust
  File "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/.eggs/setuptools_rust-1.4.1-py3.10.egg/setuptools_rust/build.py", line 20, in <module>
    from setuptools.command.build import build as CommandBuild  # type: ignore[import]
ModuleNotFoundError: No module named 'setuptools.command.build'
make[3]: *** [Makefile:57: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/cryptography-3.4.8/.built] Error 1
make[3]: Leaving directory '/home/username/works/openwrt/feeds/packages/lang/python/python-cryptography'
time: package/feeds/packages/python-cryptography/compile#1.29#0.26#2.50
    ERROR: package/feeds/packages/python-cryptography failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python-cryptography/compile] Error 1
make[2]: Leaving directory '/home/username/works/openwrt'
make[1]: *** [package/Makefile:110: /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:230: world] Error 2
```

yuklin, 2022-07-06:
I get this error too

it seems to relate with python version of compile host, i dont know

yuklin, 2022-07-06:
In the directory /home/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/command 

there is no such file as build.py, only
build_clib.py
build_ext.py
build_py.py

version of setuptools in packages is 58.1.0

maybe just upgrade setuptools aside ?

trippleflux, 2022-07-06:
@yuklin 
Now on re-compiling my OpenWRT world, trying to change the ./feeds/packages/lang/python/python3-version.mk with the following : 
```
PYTHON3_SETUPTOOLS_VERSION:=63.1.0
PYTHON3_PIP_VERSION:=22.1.2
```

Based on my findings that somehow it might relate with pip version(upgrading pip) and setuptools.

Will post back once I have finished my compilation.

yuklin, 2022-07-06:
> @yuklin Now on re-compiling my OpenWRT world, trying to change the ./feeds/packages/lang/python/python3-version.mk with the following :
>
> ```
> PYTHON3_SETUPTOOLS_VERSION:=63.1.0 [...]

cool, thanks alot

trippleflux, 2022-07-06:
Hmm, failed since it looks like using bundled pip and setuptools from python 3.10.5 release : 
```
# Use host pip to install python-setuptools
/home/username/works/openwrt/staging_dir/hostpkg/bin/pip3.10 --disable-pip-version-check --cache-dir "/home/username/works/openwrt/dl/pip-cache" install --ignore-installed --root=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/Python-3.10.5/install-setuptools --prefix=/usr /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/Python-3.10.5/Lib/ensurepip/_bundled/setuptools-63.1.0-py3-none-any.whl
WARNING: The directory '/home/username/works/openwrt/dl/pip-cache' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
WARNING: Requirement '/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/Python-3.10.5/Lib/ensurepip/_bundled/setuptools-63.1.0-py3-none-any.whl' looks like a filename, but the file does not exist
Processing /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/Python-3.10.5/Lib/ensurepip/_bundled/setuptools-63.1.0-py3-none-any.whl
ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/Python-3.10.5/Lib/ensurepip/_bundled/setuptools-63.1.0-py3-none-any.whl'
```

yuklin, 2022-07-06:
> mannually download setuptools and pip whl to ....../Lib/ensurepip/_bundled/ directory
> 
> in 63.1.0 and 22.1.2

Forget that, use "/home/{your dir}/staging_dir/hostpkg/bin/pip install pip setuptools --upgrade"

shankarunni, 2022-07-06:
I updated the "0004-disable-rust.patch" to the following.  A more thorough de-rustification (de-oxidation? :)), if you please:
```
--- a/setup.py
+++ b/setup.py
@@ -11,7 +11,7 @@
 from setuptools import find_packages, setup

 try:
-    from setuptools_rust import RustExtension
+    pass
 except ImportError:
     print(
         """
@@ -41,13 +41,15 @@

 # `install_requirements` and `setup_requirements` must be kept in sync with
 # `pyproject.toml`
-setuptools_rust = "setuptools-rust>=0.11.4"
 install_requirements = ["cffi>=1.12"]
-setup_requirements = install_requirements + [setuptools_rust]

-if os.environ.get("CRYPTOGRAPHY_DONT_BUILD_RUST"):
+if True:
+    setuptools_rust = ""
+    setup_requirements = install_requirements
     rust_extensions = []
 else:
+    setuptools_rust = "setuptools-rust>=0.11.4"
+    setup_requirements = install_requirements + [setuptools_rust]
     rust_extensions = [
         RustExtension(
             "_rust",
```
It seems to build cleanly after this.   (This is on the openwrt-21.02 branch; the same change should apply in all the current branches).

closed 2022-07-07 by commodo

scottmckenzie, 2022-07-15:
This is still failing to build for me on target-x86_64. I'm trying to build v21.02.3.

I believe I have the latest version of 0004-disable-rust.patch :
```
--- a/setup.py
+++ b/setup.py
@@ -11,7 +11,7 @@ import sys
 from setuptools import find_packages, setup
 
 try:
-    from setuptools_rust import RustExtension
+    pass
 except ImportError:
     print(
         """
@@ -45,7 +45,7 @@ setuptools_rust = "setuptools-rust>=0.11
 install_requirements = ["cffi>=1.12"]
 setup_requirements = install_requirements + [setuptools_rust]
 
-if os.environ.get("CRYPTOGRAPHY_DONT_BUILD_RUST"):
+if True:
     rust_extensions = []
 else:
     rust_extensions = [
```

Build output is nearly identical to trippleflux's original comment.

jefferyto, 2022-07-18:
@scottmckenzie You do not have the [latest patch](https://github.com/openwrt/packages/commit/580926cb6ca2021190dca9918285853be4d7d4b2). Please update your local packages feed repo.