openwrt/packages #18877: `strace`: Fails to build with `libdw` with `configure: error: failed to find dwfl_linux_proc_attach in libdw`.

felixbecker2, 2022-07-06:
**Maintainer:** Felix Fietkau <nbd@nbd.name>  
**Environment:**
* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .confi`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWRT version: Git branch `openwrt-22.03` `from git://git.openwrt.org/openwrt/openwrt.git`, checked out on 2022-07-05, latest commit: Hash `ebfe66e494e57f4b421f1190d6bff1d361db1b3d` at 2022-07-05T10:16:08+02:00:  
  `git log -1 origin --format='%H | %aI | %an'`:  
  `ebfe66e494e57f4b421f1190d6bff1d361db1b3d | 2022-07-05T10:16:08+02:00 | Sander Vanheule`
* Building on GNU/Linux on x86_64-architecture (Distribution: Artix Linux)

**Description:**

Building `package/devel/strace` from source with `libdw` fails for me as follows:

`make -j1 V=s`:  
```
[...]
checking for dwfl_linux_proc_attach in -ldw... no
configure: error: in `/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/strace-5.16':
configure: error: failed to find dwfl_linux_proc_attach in libdw
[...]
```

Configuration details regarding `strace`:  
`grep -i STRACE .config`:  
```
CONFIG_PACKAGE_strace=y
# CONFIG_STRACE_NONE is not set
CONFIG_STRACE_LIBDW=y
# CONFIG_STRACE_LIBUNWIND is not set
```

`grep -i libdw .config`:  
```
CONFIG_PACKAGE_libdw=y
CONFIG_STRACE_LIBDW=y
```

Full `.config` is attached:  
[`.config.txt`](https://github.com/openwrt/packages/files/9055368/default.config.txt)

felixbecker2, 2022-07-07:
*<small>Mentioning @nbd168, since I did not know in the first place that a mention is necessary and I don't know if an edit of the initial post where @nbd168 is added triggers a notitication.</small>*

neheb, 2022-07-11:
@felixbecker2 locally, config.log shows:

```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.1.0_musl/lib/gcc/mipsel-openwrt-linux-musl/12.1.0/../../../../mipsel-openwrt-linux-musl/bin/ld: warning: libintl.so.8, needed by /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libdw.so, not found (try using -rpath or -rpath-link)
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.1.0_musl/lib/gcc/mipsel-openwrt-linux-musl/12.1.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libdw.so: undefined reference to `libintl_gettext'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.1.0_musl/lib/gcc/mipsel-openwrt-linux-musl/12.1.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-mipsel_24kc_musl/usr/lib/libdw.so: undefined reference to `libintl_dgettext'
```

which means missing nls.mk.

BKPepe, 2022-07-11:
Package ``strace`` is not in the packages repository, but it is located in [the main repository](https://github.com/openwrt/openwrt/issues). Thus the issue should be moved there, but I don't have permission to do it, so closing it here.

closed 2022-07-11 by BKPepe

stintel, 2022-11-07:
FYI: https://github.com/openwrt/openwrt/commit/8650f77314f624a3aae358a6fef36eda8e06c833